arendst/Tasmota #19983: berry -> 'autoexec.be' does not load/run after Restart

ABaumruck, 2023-11-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Seeed Studio XIAO ESP32C3
- [x] Tasmota binary firmware version number used: 13.2.0.
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: web serial
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:09:50.238 CMD: Backlog Template; Module; GPIO 255
22:09:50.252 RSL: RESULT = {"NAME":"Tiefkühler","GPIO":[1,0,1,160,224,0,1,1,1,1,1312,0,0,0,0,0,0,0,1,1,1,1],"FLAG":0,"BASE":1}
22:09:50.486 RSL: RESULT = {"Module":{"0":"Tiefkühler"}}
22:09:50.688 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"160":"Switch1"},"GPIO4":{"224":"Relay1"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"1312":"DS18x201"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
22:12:15.731 CMD: Backlog Rule1; Rule2; Rule3
22:12:15.744 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
22:12:15.979 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
22:12:16.180 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:10:57.669 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Fridge","FriendlyName":["Fridge"],"Topic":"tasmota_F20854","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:10:57.672 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T00:19:27","StartupUTC":"2023-11-11T20:51:30","Sleep":50,"CfgHolder":4617,"BootCount":252,"BCResetTime":"2023-10-24T18:05:22","SaveCount":446}}
22:10:57.673 RSL: STATUS2 = {"StatusFWR":{"Version":"13.2.0(tasmota32)","BuildDateTime":"2023-10-19T09:05:17","Core":"2_0_14","SDK":"4.4.6.231011","CpuFrequency":160,"Hardware":"ESP32-C3 v0.3","CR":"400/699"}}
22:10:57.675 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box7530UP",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
22:10:57.684 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1852,"Free":1027,"Heap":193,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","080840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
22:10:57.693 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-F20854-2132","IPAddress":"192.168.2.191","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"2003:e0:cf03:5d00:464e:6dff:fea2:ab4a","Mac":"58:CF:79:F2:08:54","IP6Global":"2003:e0:cf03:5d00:5acf:79ff:fef2:854","IP6Local":"fe80::5acf:79ff:fef2:854","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":19.0}}
22:10:57.698 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-11-11T21:10:57","Local":"2023-11-11T22:10:57","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:52","Sunset":"17:15"}}
22:10:57.714 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-11-11T22:10:57","Switch1":"OFF","DS18B20":{"Id":"011438EF97AA","Temperature":24.6},"TempUnit":"C"}}
22:10:57.717 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-11-11T22:10:57","Uptime":"0T00:19:27","UptimeSec":1167,"Heap":193,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box7530UP","BSSId":"44:4E:6D:A2:AB:4C","Channel":3,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:03"}}}
22:10:57.722 RSL: STATUS12 = {"StatusSTK":{"Exception":5,"Reason":"Load access fault","EPC":"420cb280","EXCVADDR":"00000000","CallChain":["420baf50","420cd4e4","420cd4e4","420baf50","420cd4e4","420cd4e4","420cb4a4","420cd4e4","420cc47a","420cc4e6","4038368c","420ba76c","42023132","42023a94","42067ef2","42067e00","42023e02","4202e85c","42024036","4202c196","42117c3a","42034340","420325d2","40388086","4202c3da","4202d19e","4202e6f2","4202d18e","4202e6e8","40383738","4207e6e2","4038bfae","42030f78","4203176e","420317f0","42030f78","4203176e","420125c6","4206b214","4206b210","40389208","4203538a","4205cde0","4205f518","42040204","420901f4","4209fe14","42019376"]}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP32-C3 v0.3 
00:00:00.021 UFS: FlashFS mounted with 296 kB free
00:00:00.034 CFG: Loaded from File, Count 445
00:00:00.036 FRC: Some settings have been reset (4)
00:00:00.050 TFS: File '.drvset003' not found
00:00:00.053 Project tasmota - Fridge Version 13.2.0(tasmota32)-2_0_14(2023-10-19T09:05:17)
00:00:00.563 WIF: Connecting to AP1 FRITZ!Box7530UP Channel 3 BSSId 44:4E:6D:A2:AB:4C in mode 11n as tasmota-F20854-2132...
00:00:02.257 WIF: Connected
00:00:02.510 HTP: Web server active on tasmota-F20854-2132 with IP address 192.168.2.191
00:00:02.512 RSL: INFO1 = {"Info1":{"Module":"Tiefkühler","Version":"13.2.0(tasmota32)","FallbackTopic":"cmnd/DVES_F20854_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:02.524 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-F20854-2132","IPAddress":"192.168.2.191","IP6Global":"","IP6Local":"fe80::5acf:79ff:fef2:854"}}
00:00:02.550 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":5,"Reason":"Load access fault","EPC":"420cb280","EXCVADDR":"00000000","CallChain":["420baf50","420cd4e4","420cd4e4","420baf50","420cd4e4","420cd4e4","420cb4a4","420cd4e4","420cc47a","420cc4e6","4038368c","420ba76c","42023132","42023a94","42067ef2","42067e00","42023e02","4202e85c","42024036","4202c196","42117c3a","42034340","420325d2","40388086","4202c3da","4202d19e","4202e6f2","4202d18e","4202e6e8","40383738","4207e6e2","4038bfae","42030f78","4203176e","420317f0","42030f78","4203176e","420125c6","4206b214","4206b210","40389208","4203538a","4205cde0","4205f518","42040204","420901f4","4209fe14","42019376"]},"BootCount":252}}
00:00:02.603 RSL: RESULT = {"POWER":"OFF"}
00:00:02.614 RSL: POWER = OFF
21:51:37.944 QPC: Reset
21:51:37.947 RSL: STATE = {"Time":"2023-11-11T21:51:37","Uptime":"0T00:00:07","UptimeSec":7,"Heap":198,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box7530UP","BSSId":"44:4E:6D:A2:AB:4C","Channel":3,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:51:37.982 RSL: SENSOR = {"Time":"2023-11-11T21:51:37","Switch1":"OFF","DS18B20":{"Id":"011438EF97AA","Temperature":24.6},"TempUnit":"C"}

[... 66 lines cut ...]
end

[... cut ...]
tasmota.cmd("switchmode 1")                                 # make sure the switch follows the state of Power0 output (LED) and it is OFF when closed with the magnet 
tasmota.add_rule("Switch1#State", udp_switch_msg)           # rule to send an udp msg when "Switch1#State" has changed 

if u.begin("", 0)
  if send_udp_msg("first")                                   # send first udp message, and if successful
    tasmota.add_cron(crontab,udp_cron_msg,"repeat_msg")      # add a cron to call "every ......"
  end
end

##### END OF MAIN PROG 
```

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-11-11:
00:00:00.036 FRC: Some settings have been reset (4)

Something made a crash loop, probably your berry script
So berry has been disabled to recover stability

Working as designed, not a Tasmota issue

barbudor, 2023-11-11:
https://tasmota.github.io/docs/FAQ/#cannot-connect-to-my-mqtt-broker

barbudor, 2023-11-11:
https://tasmota.github.io/docs/FAQ/#frequent-rebootsbootloops

barbudor, 2023-11-11:
Your problem is likely you are trying to send UDP before wifi is up
Either forget about the initial message or use the trigger 'wifi#connected' to send the initial message

ABaumruck, 2023-11-11:
Strange!!
Why should the problem be the berry script, when I can manually (brrestart -> load('autoexec.be') run the script flawlessly??

Additionally: after a "brrestart" the autoexec.be is NOT executed

ABaumruck, 2023-11-11:
OK, the wifi sounds like this could be the reason for the initial problem
Does not explain the brrestart -> no autoexec.be problem