arendst/Tasmota #20030: ESP32-C3 retains gpio state for non-relay gpios due to gpio_hold_en

smhc, 2023-11-15:
### PROBLEM DESCRIPTION

For ESP32-C3 boards, any pin configured as a relay will retain its state even when configured as something else, e.g output_hi or output_lo. The use of gpio.digital_write() is also ignored for these pins. This can make diagnosing issues quite difficult when temporarily using 'relay' to test a pin and having the state "stuck" thereafter.

The "bug" is due to this code which uses gpio_hold_en for REL1: https://github.com/arendst/Tasmota/blob/a01bed3928d74306f5f138c0b9a341ae5ddf981e/tasmota/tasmota_support/support.ino#L1466

Unfortunately this 'gpio_hold_en' flag is left on the pin even after reconfiguring the gpio. A hard power reset is required to clear this flag. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x ] Device used (e.g., Sonoff Basic): ESP32-C3-12F
- [ x ] Tasmota binary firmware version number used: 13.2.0
  - [ x ] Pre-compiled
  - [ ] Self-compiled
- [ x ] Flashing tools used: Web
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

- Configure a GPIO as "Relay 1"
- Toggle the relay
- Configure the same GPIO as either OUTPUT_LO or OUTPUT_HI
- Note that the setting is ignored
- Modify the gpio via 'gpio.digital_write()
- Note that the API call is ignored

### EXPECTED BEHAVIOUR

- The GPIO should have gpio_hold_dis set when configured as something other than a relay.
- The use of gpio_hold_en() should only be used if PowerRetain is on?
- The documentation might want to describe this behaviour?
- Do nothing, and just keep this issue around to help others searching for the problem?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-11-16:
Some background info.

All esp8266 and esp32 derivatives are supposed to keep their relay state after a restart. Measures have been taken to make this possible as default arduino code did not support this. The esp32c3 needed additional steps to have this functionality. Alas does the solution need a power cycle and not just a restart to remove this functionality. 

I will add a SetOption to enable debugging in your case but the default behaviour will be as it is now.

arendst, 2023-11-16:
Ah I see you're using berry. In that case the dis should be called first like in function DigitalWrite(). It would have been better if berry allowed you to use that function but then it seems berry doesn't mind template config anyway.....

smhc, 2023-11-16:
I thought at a minimum it could at least call gpio_hold_dis on all *non*relay pins when saving configuration (or if it happens to know the previous setting for the pin, only for pins that were previously set as relay 1). That would solve 99% of the issues without needing any new option or special logic in the berry callback.

arendst, 2023-11-16:
I'll have to dive in and find a way to divert edge cases (like redefining relays by something different).

arendst, 2023-11-16:
Try latest dev. At least the "normal" functionality should now work like changing RELAY to OUTPUT_HI/LOW.

smhc, 2023-11-16:
Thanks @arendst - I'm not in a position to try it right now but I assume it would work ok. I'm guessing that the digital write occurs when configuring an output_lo/hi, triggering the reset.
I'm just not sure how the performance/power may be impacted making this call on every value change.. but I suppose it already was getting called for relay1s anyway.

arendst, 2023-11-17:
Latest change should fix both "default" Tasmota outputs AND Berry configured outputs.

smhc, 2023-11-17:
Thanks @arendst that looks like it should work. I can try it out later this week but happy to close the issue for now.

Jason2866, 2023-11-21:
Closing since fixed. You can reopen when the issue is unexpectedly still there.

closed 2023-11-21 by Jason2866