arendst/Tasmota #20048: Reset loop of NSPanel on latest Tasmota version after factory flash

stleusc, 2023-11-17:
### PROBLEM DESCRIPTION
I just flashed a NSPanel that had ESPHome running via esptool.
I used the factory bin thinking I should wipe everything out.
Flashing worked fine.
However AP is not letting me connect.

Hooked up serial console to see why:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4

00:00:00.003-215/51 HDW: ESP32-D0WD-V3 v3.0
00:00:00.008-214/50 UFS: FlashFS mounted with 308 kB free
00:00:00.013 CFG: Loaded from File, Count 11
00:00:00.023 QPC: Count 1
00:00:00.069 BRY: Berry initialized, RAM used=4438 bytes
00:00:00.078 TFS: File '.drvset003' not found
00:00:00.086 Project tasmota - Tasmota Version 13.2.0(nspanel)-2_0_14(2023-10-19T09:05:12)
00:00:00.242 WIF: WifiManager active for 3 minutes
00:00:00.983 HTP: Web server active on tasmota-C359AC-6572 with IP address 192.168.4.1
00:00:06.986 QPC: Reset
```

Clearly, the RESET is not what I want.

Any advice?

**(Please, remember to close the issue when the problem has been addressed)**

stleusc, 2023-11-18:
Btw, when I use the serial port to flash a basic version of esphome it shows me the AP and would allow me to configure it. Did not continue from there since I want/need Tasmota. But I guess it shows the HW is fine.
I do NOT get the reset. 
Tried Tasmota 12.0.x and got the same QPC reset almost exactly 6s after AP starts. Every time!

Can't believe I am the first/only one...

sfromis, 2023-11-18:
`00:00:06.986 QPC: Reset` is perfect. I means that it survived long enough to consider the boot successful.

Many devices cannot get on Wifi with the serial flashing setup connected. Try removing that, and powering via mains instead.

stleusc, 2023-11-18:
well that is what I tried INITIALLY, but the screen stayed blank which was odd..... since the Nextion still should show whatever it had before, right?
I was not able to connect so I decided to test with serial connected.

Odd thing is, flashing ESPHome allows me to connect to the ESPHome AP, while using serial power!
I will try again now.
After going back to Tasmota, I now see the QPC: Reset after 15s.

stleusc, 2023-11-18:
nope, it will not stay active, even with mains power.
Also did another test. I have a second panel running Tasmota. Swapped that board in the one I am trying to flash and boom, screen comes on. So at least, the Nextion also seems fine.
That leaves the question, what is QPC reset and why am I stuck in that loop....

I THOUGHT using the factory image and esptool would really allow me to go from esphome to tasmota. I actually did that a few times before, no trouble. Definitely also did that on the currently working panel.

stleusc, 2023-11-18:
Oh, and just for fun, I can connect the 'working' panel and power it via serial and watch it talk to MQTT. 
So it really comes down to this QPC: Reset, and why that happens?!

sfromis, 2023-11-18:
Again, the QPC means *success*, it is **not** an error message. The backstory is that Tasmota has a feature "Quick Power Cycle" recovery, where 7 power cycles will trigger a "factory reset". Can be very handy sometimes, as long as you do not disable it. Therefore `QPC: Reset` means that it got far enough that it reset the counter for this feature, as the boot is ok.

And since the boot is successful, your flashing was ok. The Tasmota boot process may take a bit more power, and thus have higher risk of the USB/serial adapter not delivering enough juice.

I have no hands-on with the device, but until you've completed the configuration, I'd not expect any useful info by watching display or leds.

stleusc, 2023-11-18:
I have 2 IDENTICAL panels. both flashed with the exact same SW.
I have no problem running one off the serial connector. But I am only trying that because I simply can NOT connect to the AP of the 'bad' device. Seem like, even with mains power, the AP does not stay.

I am using putty and it somehow does not allow me to type... at least there is no feedback. But I also get the device reset after 6s.

I understand you say QPC: Reset is good.... but the moment I get the message is the moment the device reboots. Coincidence maybe?

stleusc, 2023-11-18:
well, I do NOT have the answer for what the issue was....
But I have probably found a way out.

I used esptool and took a 4MB dump of the working panel and flashed it onto the bad one. It turns on (screen shows still epshome 'waiting' screen as expected) and no resets as I can tell.
I seemingly took the WiFi config too?

In any case, still curious what was going on.... I also saw it connect to Wifi WITH serial attached. So it definitely was an issue with the code that was running. What it was? NO CLUE!

stleusc, 2023-11-18:
@sfromis while you did not provide any real answers to why I was having this problem I honestly appreciate the moral support and it helped knowing that the message did not mean the device was dead! Thank you for that!