arendst/Tasmota #20070: bcm280 lost after enabling mqtt and creating rule from console

mesiu84, 2023-11-20:
### PROBLEM DESCRIPTION
BCM280 is lost from Tasmota after doing following:
1. clear setup
2. configure bcm280 in UI + restart <- all works OK
3. configure MQTT + restart <- all works OK
4. add internal rule (rule1 on bcm280#humidity>70 do power1 1 endon on bcm280#humidity<=70 do power1 0 endon) <- all works OK
5. enable rule1 (rule1 1) <- all works OK
6. restart tasmota

After doing above steps bcm280 is lost and I don't see any readings from it. Doing restart doesn't fix the issue, but if I do:
7. rule1 0
8. restart tasmota

Then bcm280 is back again. Enabling rule1 (2 or 3 doesn't matter) and doing a restart I can't see bcm280 again, if I don't do a restart all works until next restart or there is no power and tasmota boots up again

### REQUESTED INFORMATION
  Configuration output here:
Sonoff Basic R2
Program Version	13.2.0(sensors) <- from official URL
Core/SDK Version	2_7_4_9/2.2.2-dev(38a443e)
Uptime	2T18:12:35
Flash Write Count	833 at 0xF5000
Boot Count	14

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
```17:56:33.937 MQT: stat/lazienka_parter/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":229,"Free":282,"Rules":"ON System#Boot DO Var1 tmp ENDON ON System#Boot DO RuleTimer1 60 ENDON ON Rules#Timer=1 DO Var1 BME280 ENDON ON Switch1#state=1 do publish2 %topic%/stat/%Var1% ON ENDON ON Switch1#state=0 do Publish2 %topic%/stat/%Var1% OFF ENDON"}}```

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
```17:53:48.843 CMD: status 0
17:53:48.852 MQT: stat/tasmota/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:53:48.868 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"2T18:15:08","StartupUTC":"2023-11-17T22:38:40","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2023-11-17T17:51:45","SaveCount":833,"SaveAddress":"F5000"}}
17:53:48.880 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"13.2.0(sensors)","BuildDateTime":"2023-10-19T09:02:00","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"380/699"}}
17:53:48.892 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["doma",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
17:53:48.920 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":701,"Free":300,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66,68","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
17:53:48.939 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-2493","IPAddress":"192.168.1.188","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"48:55:19:77:C9:BD","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:53:48.954 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.104","MqttPort":1883,"MqttClientMask":"DVES_77C9BD","MqttClient":"DVES_77C9BD","MqttUser":"mqtt-user","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:53:48.971 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2023-11-20T16:53:48","Local":"2023-11-20T17:53:48","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:06","Sunset":"17:04"}}
17:53:48.989 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2023-11-20T17:53:48","BME280":{"Temperature":21.6,"Humidity":64.3,"DewPoint":14.5,"Pressure":982.0},"PressureUnit":"hPa","TempUnit":"C"}}
17:53:49.002 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2023-11-20T17:53:48","Uptime":"2T18:15:08","UptimeSec":238508,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WiFiNet","BSSId":"9C:3D:CF:CA:7B:C0","Channel":13,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}}```

### EXPECTED BEHAVIOUR
BCM280 should not disappear

sfromis, 2023-11-20:
You did not post much of the requested info, specifically detailed logs from when it happens.

Anyway, using Sonoff Basic R2 is just about the worst hardware you can find, not a good platform for adding stuff to, and you are not the first to have trouble running the sensors build on that thing. You may have a boot loop causing Tasmota to "scale back" features until able to boot without crashing quickly again. It tends to work better using decent hardware instead.

Maybe you'd have luck picking up a very old Tasmota version with less features. Or maybe do a custom build to only include features you want.

mesiu84, 2023-11-21:
That's actually my plan to test older version (8.1.0 is my target) and if that doesn't work to try my custom build. But what would be a better hardware to work with? Sonoff was the easiest to get for me :-) but now I can try something else too

sfromis, 2023-11-21:
Since you got the worst device, the Sonoff Basic R2, "better" means "something else", which does not exclude Sonoff devices.

mesiu84, 2023-11-21:
I don't need anything fancy, it's just a small thing that I'm doing, that's why I don't need any extra devices. The question is more for a future, what is better in your opinion? I'm mostly using zigbee protocol and devices that work with zigbee, that sonoff was something that I just didn't found anywhere with zigbee.