arendst/Tasmota #20082: Matter device association fails

SilentResonance, 2023-11-21:
### PROBLEM DESCRIPTION
Thanks for this great software even with Matter support.
Added virtual occupancy endpoint in 13.1.0.3 and it worked fine.
Planning to use it on ESP32-C3, therefore updated Tasmota from 13.1.0.3 to 13.2.0.2 on ESP32-WROOM-32 devkit (still used for now). Google Home is used with a Nest Mini Gen2.
Building on platform.io in vscode.
Switched between several commits and it seems broken with 0ee62e597bcb877f3c4b2c4a2e8d0197d0ee51cf
Latest development commit also does not work.
With commit 73f219e2bb63de8024574b37ef0c31b7da5750e3 adding the device to Google Home via Matter

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32-WROOM-32 devkit
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled 13.1.0.3 and 13.2.0.2, development head (and commits, some listed above)
- [X] Flashing tools used: pio run -t nobuild -t factory_flash -e tasmota32 (from Platform.io in vscode: esptool.py v4.7-dev)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:52:40.203 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
21:52:40.440 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
21:52:40.652 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"288":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"160":"Switch1"},"GPIO19":{"161":"Switch2"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
21:53:33.087 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":250,"Free":261,"Rules":"on Switch1#state do Publish stat/Mailbox/Occupied %value% endon on (Switch1#state=0 AND Switch2#state=1) do MtrUpdate {\"name\":\"Mailbox\", \"Occupancy\":1} endon on (Switch1#state=1 AND Switch2#state=0 do MtrUpdate {\"name\":\"Mailbox\", \"Occupancy\":0} endon"}}
21:53:35.974 RSL: STATE = {"Time":"2023-11-21T21:53:35","Uptime":"0T00:20:10","UptimeSec":1210,"Heap":120,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":19,"Objects":223},"Wifi":{"AP":1,"SSId":"Homeland","BSSId":"74:42:7F:DF:89:9F","Channel":11,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:53:35.993 RSL: SENSOR = {"Time":"2023-11-21T21:53:35","Switch1":"ON","Switch2":"ON"}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
21:54:39.168 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_987D88","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:54:39.189 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:21:14","StartupUTC":"2023-11-21T20:33:25","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2023-11-20T00:27:10","SaveCount":31}}
21:54:39.208 RSL: STATUS2 = {"StatusFWR":{"Version":"13.2.0.2(tasmota32)","BuildDateTime":"2023-11-21T21:31:05","Core":"2_0_14","SDK":"4.4.6.231024","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"405/699"}}
21:54:39.223 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Homeland",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006002","00004000","00000020"]}}
21:54:39.253 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1880,"Free":999,"Heap":120,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20202000","D4BC482D","810A80B1","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,80,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
21:54:39.273 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-987D88-xxxx","IPAddress":"192.x.x.x","Gateway":"192.x.x.x","Subnetmask":"255.255.255.0","DNSServer1":"192.x.x.x","DNSServer2":"x:x:56ff:8c00:7642:x:x:899c","Mac":"48:E7:x:x:x:88","IP6Global":"x:x:56ff:8c00:4ae7:x:x:7d88","IP6Local":"fe80::4ae7:x:x:7d88","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.x.x.x","DNSServer2":"xxxx:xxxx:56ff:8c00:7642:x:x:899c","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:54:39.295 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_987D88","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:54:39.309 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-11-21T20:54:39","Local":"2023-11-21T21:54:39","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:08","Sunset":"17:03"}}
21:54:39.327 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-11-21T21:54:39","Switch1":"ON","Switch2":"ON"}}
21:54:39.347 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-11-21T21:54:39","Uptime":"0T00:21:14","UptimeSec":1274,"Heap":117,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":19,"Objects":229},"Wifi":{"AP":1,"SSId":"xxxx","BSSId":"74:42:x:x:89:9F","Channel":11,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
22:16:56.814 CMD: weblog 4
22:16:56.817 RSL: RESULT = {"WebLog":4}
22:16:57.879 CFG: Saved, Count 38, Bytes 4096
22:17:07.268 BRY: GC from 24116 to 12515 bytes, objects freed 270/161 (in 3 ms) - slots from 606/623 to 163/456
22:17:14.973 WIF: Checking connection...
22:17:17.973 HTP: Main Menu
22:17:18.226 BRY: GC from 22798 to 13761 bytes, objects freed 166/187 (in 2 ms) - slots from 454/471 to 166/410
[...]
22:17:30.460 BRY: GC from 24059 to 13743 bytes, objects freed 89/187 (in 2 ms) - slots from 182/395 to 166/395
22:17:32.507 MTR: UDP received from [0.0.0.0]:49540
22:17:32.520 MTR: >Received  (     0) PBKDFParamRequest rid=210165239 exch=32400 from [0.0.0.0]:49540
22:17:32.535 BRY: GC from 24045 to 13873 bytes, objects freed 109/172 (in 3 ms) - slots from 281/395 to 164/395
22:17:32.538 MTR: +Session   ( 35521) from '[0.0.0.0]:49540'
22:17:32.560 MTR: <Replied   (     0) PBKDFParamResponse
22:17:32.568 WIF: DNS failed for 0.0.0.0 after 1 ms
22:17:32.571 MTR: error sending packet to '[0.0.0.0]:49540'
22:17:32.908 MTR: .          Resending packet id=150715563
[... cut ...]
10. Open Google Home App and add new device to a room, after that Tasmota Mailbox (do you want to add? pops up) answer yes, scan QR code.
11. Device is found, Matter data generated and sucessfully added to Home (bridge and sensor).

### EXPECTED BEHAVIOUR
See above second part

### SCREENSHOTS
Using german Google Home App, therefore not adding screenshots here.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-11-22:
I will look at it. There has been some changes lately. Thanks for pointing out the failing commits, this helps a lot

Jason2866, 2023-11-22:
@s-hadinger The Reg SPI Base code changed since the one in `fixes.h` does not compile in IDF. Commit for C3https://github.com/tasmota/esp-idf/commit/5d613e9b7ab74c9181bfa1e2c30d56d85710c068
and for S3 https://github.com/tasmota/esp-idf/commit/df54dbf0ffb64215971be1641e69b2ccee360ada
The same changes are in IDF5.1 But idk why this should have an impact on general Matter function.
@TD-er fyi

Jason2866, 2023-11-22:
@SilentResonance Can you try without the SPI part in your `user_config_override.h`
So removing this
```
 USE_SPI                         // Hardware SPI using GPIO12(MISO), GPIO13(MOSI) and GPIO14(CLK) in addition to two user selectable GPIOs(CS and DC)
#define USE_RC522                       // Add support for MFRC522 13.56Mhz Rfid reader (+6k code)
#define USE_RC522_DATA_FUNCTION         // Add support for reading data block content (+0k4 code)
#define USE_RC522_TYPE_INFORMATION      // Add support for showing card type (+0k4 code)
```

s-hadinger, 2023-11-22:
The problem is clearly here: `MTR: UDP received from [0.0.0.0]:49540`

It means that getting the remote address fails, hence Tasmota is unable to respond.

It looks similar to the problem mentioned by @TD-er

s-hadinger, 2023-11-22:
@SilentResonance It's not a SPI problem, but an IPv6 problem.

SilentResonance, 2023-11-22:
> @SilentResonance It's not a SPI problem, but an IPv6 problem.
Should I still try without #define USE_SPI ? Probably without RC522 which needs SPI?
IPv6 problem in my network? Why is it working with the commit before the problematic one?

Jason2866, 2023-11-22:
@SilentResonance Currently no. The real problem is identified. It is "somewhere" in the IPv6 Arduino or IDF layer. With the new platform a lot changed "under the hood" in Arduino/IDF. We will come back to you if we have something to test.

s-hadinger, 2023-11-22:
The fix is on its way

Jason2866, 2023-11-22:
Just merged.

SilentResonance, 2023-11-23:
> Just merged.

Short question. I am not firm with PR and stuff Should I try a693c28d07f3374c2cc7e05dc8df9c75f8f2a96c ?

s-hadinger, 2023-11-23:
Just use the latest version from `development`, it should be fixed now

SilentResonance, 2023-11-23:
I tried last development commit a693c28d07f3374c2cc7e05dc8df9c75f8f2a96c
Works like a charm.
Thanks for this extraordinary fast fix.

closed 2023-11-23 by SilentResonance