arendst/Tasmota #20102: WebQuery get right answer on 8266 but erratic on ESP32

ygelt, 2023-11-25:
### PROBLEM DESCRIPTION
I use the same script on 4 tasmota devices : 3 with 8266, 1 with ESP32 which are all querying the same web service every 3 seconds using WebQuery GET. This works perfectly on the 8266 devices, I mean they receives the web service answer on each call; while the ESP32 does not receive the data most of the time, quite erratically.
The web service is correctly called by the device (correct headers and parameter).
It "receives the answer" means it logs they like here :
23:49:00.760 RSL: RESULT = {"WebQuery":{"cmd":0, "con":0, "alv":172297, "cf1":2, "cf2":3, "cf3":4, "cf4":4, "cf5":20}}
23:49:00.771 RSL: RESULT = {"WebQuery":"Done"}

It "does not receive the answer" means it logs like :
23:46:34.111 RSL: RESULT = {"WebQuery":{"WebQuery":}
23:46:34.120 RSL: RESULT = {"WebQuery":"Done"}

The calling WebQuery is:
=>WebQuery http://www.xxxx.xxxx/connect/v1/device?al=%0nWS% GET [id:%topic%|ip:%lip%]
(The issue is the same in http as https)

Resulting in log:
00:09:07.260 CMD: Grp 0, Cmd 'WEBQUERY', Idx 1, Len 107, Pld -99, Data 'http://www.xxxx.xxxx/connect/v1/device?al=29236 GET [id:01234567890123456789012345678901|ip:192.168.0.38]'

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Rocket Controller ESP32R4
- [x ] Tasmota binary firmware version number used: 13.0.0
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [ x] Flashing tools used: OTA
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:06:47.367 CMD: Backlog Template; Module; GPIO 255
00:06:47.396 RSL: RESULT = {"NAME":"ESP32R4","GPIO":[1,1,576,1,480,1,1,1,1,1,1,1,1,1,1,1,0,640,608,1,0,224,225,1152,0,0,0,0,227,226,32,33,34,0,0,35],"FLAG":0,"BASE":1}
00:06:47.620 RSL: RESULT = {"Module":{"0":"ESP32R4"}}
00:06:47.876 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA"},"GPIO22":{"608":"I2C SCL"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"224":"Relay1"},"GPIO26":{"225":"Relay2"},"GPIO27":{"1152":"RFrecv"},"GPIO32":{"227":"Relay4"},"GPIO33":{"226":"Relay3"},"GPIO34":{"32":"Button1"},"GPIO35":{"33":"Button2"},"GPIO36":{"34":"Button3"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"35":"Button4"}}

```
- [N/A ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:08:00.585 CMD: Status 0
00:08:00.590 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"TeGOrLS52scMK3fQUN5kolbKfdRVoIul","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:08:00.598 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-safeboot.bin","RestartReason":"Software reset CPU","Uptime":"1T10:58:25","StartupUTC":"2023-11-24T12:09:35","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2023-11-15T08:02:08","SaveCount":104}}
00:08:00.606 RSL: STATUS2 = {"StatusFWR":{"Version":"13.0.0(TasmoCompiler-esp32generic)","BuildDateTime":"2023-11-24T08:33:51","Core":"2_0_10","SDK":"4.4.5.230614","CpuFrequency":160,"Hardware":"ESP32-D0WD rev.1","CR":"427/699"}}
00:08:00.610 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VOO-014092",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00206000","00004000","00000000"]}}
00:08:00.619 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1253,"Free":1856,"Heap":121,"StackLowMark":1,"PsrMax":4093,"PsrFree":4085,"ProgramFlashSize":4096,"FlashSize":16384,"FlashChipId":"1840A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","078045CA","000C0001","00000003","00000000","C0000881","00000080","00200000","54000008","00000080"],"Drivers":"1,2,4,8,9,10,12,24,26,29,35,38,50,52,62,68","Sensors":"1,2,127"}}
00:08:00.627 RSL: STATUS5 = {"StatusNET":{"Hostname":"TeGOrLS52scMK3fQUN5kolbKfdRVoIu","IPAddress":"192.168.0.38","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"62.197.111.140","DNSServer2":"109.88.203.3","Mac":"3C:E9:0E:B0:8C:24","IP6Global":"2a02:2788:1b4:cbd:3ee9:eff:feb0:8c24","IP6Local":"fe80::3ee9:eff:feb0:8c24","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:08:00.634 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B08C24","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:08:00.639 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-11-25T23:08:00","Local":"2023-11-26T00:08:00","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:15","Sunset":"16:59"}}
00:08:00.648 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-11-26T00:08:00"}}
00:08:00.655 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-11-26T00:08:00","Uptime":"1T10:58:25","UptimeSec":125905,"Heap":120,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":32},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"VOO-014092","BSSId":"50:09:59:F3:80:01","Channel":6,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:09:07.260 CMD: Grp 0, Cmd 'WEBQUERY', Idx 1, Len 107, Pld -99, Data 'http://www.xxxx.xxxx/connect/v1/device?al=29236 GET [id:01234567890123456789012345678901|ip:192.168.0.38]'
00:09:07.461 RSL: RESULT = {"WebQuery":{"WebQuery":}
00:09:07.470 RSL: RESULT = {"WebQuery":"Done"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I can send my script in mp for a real-time test

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect this answer as on 8266 devices:
23:49:00.760 RSL: RESULT = {"WebQuery":{"cmd":0, "con":0, "alv":172297, "cf1":2, "cf2":3, "cf3":4, "cf4":4, "cf5":20}}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I compiled Tasmota 13.0.0 using TasmoCompiler including Script and TLS MQTT support but not Berry.
Here are my user_config_override.h options:
#ifndef ESP32
#define ESP32
#endif

#ifdef USE_BERRY
#undef USE_BERRY
#endif

#ifdef SCRIPT_MAXSSIZE
#undef SCRIPT_MAXSSIZE
#endif
#define SCRIPT_MAXSSIZE 64

#ifndef USE_SCRIPT
#define USE_SCRIPT
#endif

#ifndef USE_WEBSEND_RESPONSE
#define USE_WEBSEND_RESPONSE
#endif

#ifdef USE_RULES
#undef USE_RULES
#endif

#define PRECONFIGURED_SCRIPT ">D 128\n;--- start setup ---\n;......"

#ifndef START_SCRIPT_FROM_BOOT
#define START_SCRIPT_FROM_BOOT
#endif

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-11-26:
I've seen cases of corrupted response buffer. I can't remember if it was exclusively on ESP esp32 but, yes it might be the case. 
It has been in my to do list for months
I need to put it back on top of the list

sfromis, 2023-11-26:
May not be related, by recently another case of broken `WebQuery` responses was mentioned. IDK if the investigation got further.
https://discord.com/channels/479389167382691863/790182324939128832/1175492686833266739

barbudor, 2023-11-26:
I get some funny corruptions where part of the previous log message got it's way in the WebQuery response
```
18:04:00.948 CMD: backlog rule1 1 ; rule2 1
18:04:00.982 MQT: stat/esp32dev3/RULE = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":66,"Free":445,"Rules":"on tele-Time do webquery http://192.168.168.244/test.php GET endon"}}
18:04:01.210 MQT: stat/esp32dev3/RULE = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":38,"Free":473,"Rules":"on WebQuery#Time do var1 %value% endon"}}
18:04:03.964 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:04:03+01:00","Uptime":"0T00:01:35","UptimeSec":95,"Heap":169,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:04:03.996 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:04:04.031 {"Time":"2023-11-26T18:04:04+01:00"}
18:04:04.032 AssumeJSON=1
18:04:04.039 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{ssumeJSON=1}
18:04:04.053 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
18:04:13.927 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:04:13+01:00","Uptime":"0T00:01:45","UptimeSec":105,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:04:13.960 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:04:14.002 {"Time":"2023-11-26T18:04:14+01:00"}
18:04:14.004 AssumeJSON=1
18:04:14.010 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{ssumeJSON=1}
18:04:14.021 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
18:04:23.946 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:04:23+01:00","Uptime":"0T00:01:55","UptimeSec":115,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:04:23.983 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:04:24.029 {"Time":"2023-11-26T18:04:24+01:00"}
18:04:24.031 AssumeJSON=1
18:04:24.038 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{ssumeJSON=1}
18:04:24.049 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
18:04:33.925 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:04:33+01:00","Uptime":"0T00:02:05","UptimeSec":125,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:04:33.958 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:04:34.022 {"Time":"2023-11-26T18:04:34+01:00"}
18:04:34.024 AssumeJSON=1
18:04:34.033 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{"Time":"2023-11-26T18:04:34+01:00"}}
18:04:34.049 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
18:04:43.921 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:04:43+01:00","Uptime":"0T00:02:15","UptimeSec":135,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:04:43.954 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:04:43.993 {"Time":"2023-11-26T18:04:44+01:00"}
18:04:43.994 AssumeJSON=1
18:04:44.001 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{ssumeJSON=1}
18:04:44.014 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
18:04:53.937 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:04:53+01:00","Uptime":"0T00:02:25","UptimeSec":145,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:04:53.970 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:04:54.007 {"Time":"2023-11-26T18:04:54+01:00"}
18:04:54.008 AssumeJSON=1
18:04:54.016 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{ssumeJSON=1}
18:04:54.032 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
18:05:03.954 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:05:03+01:00","Uptime":"0T00:02:35","UptimeSec":155,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:05:03.988 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:05:04.229 {"Time":"2023-11-26T18:05:04+01:00"}
18:05:04.231 AssumeJSON=1
18:05:04.238 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{}
18:05:04.251 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
18:05:13.923 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:05:13+01:00","Uptime":"0T00:02:45","UptimeSec":165,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:05:13.957 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:05:14.035 {"Time":"2023-11-26T18:05:14+01:00"}
18:05:14.036 AssumeJSON=1
[... cut ...]
18:05:53.940 MQT: tele/esp32dev3/STATE = {"Time":"2023-11-26T18:05:53+01:00","Uptime":"0T00:03:25","UptimeSec":205,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:05:53.976 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
18:05:54.009 {"Time":"2023-11-26T18:05:54+01:00"}
18:05:54.010 AssumeJSON=1
18:05:54.017 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{ssumeJSON=xV���?�d}
18:05:54.031 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
```

barbudor, 2023-11-26:
@arendst I believe we've found a memory coruption problem around `Response_P()` and `AddLog()` functions in WebQuery Reponse. Probably pointing to `ext_vsnprintf_P(...)`.
The problem occurs only on ESP32, not on ESP8266.
I've seen such behavior for quite some time, months is not years but didn't had the time to investigate further before.

I have added some serial log here: https://github.com/barbudor/Tasmota/blob/5d2a8d1c792305e18e6b7e74be3a40ac284396f3/tasmota/tasmota_xdrv_driver/xdrv_01_9_webserver.ino#L3357-L3383

When it works fine and there is no corruption, it looks like:
```
19:03:52.946 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
a={"Time":"2023-11-26T19:03:53+01:00"}
b={"Time":"2023-11-26T19:03:53+01:00"}
c={"Time":"2023-11-26T19:03:53+01:00"}
y={"Time":"2023-11-26T19:03:53+01:00"}
z={"Time":"2023-11-26T19:03:53+01:00"}
19:03:52.000 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{"Time":"2023-11-26T19:03:53+01:00"}}
19:03:53.015 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
```

But randomly (often) it shows things like:
```
18:58:42.951 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
a={"Time":"2023-11-26T18:58:43+01:00"}
b={"Time":"2023-11-26T18:58:43+01:00"}
c={"WebQuery":
y={"WebQuery":
z={"WebQuery":
18:58:42.993 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{"WebQuery":}
18:58:43.017 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
```
or like:
```
19:02:02.983 RUL: TELE-TIME performs "webquery http://192.168.168.244/test.php GET"
a={"Time":"2023-11-26T19:02:03+01:00"}
b={"Time":"2023-11-26T19:02:03+01:00"}
c={"Time":"2023-11-26T19:02:03+01:00"}
y=
z=
19:02:03.022 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{"Time":"2<}
19:02:03.041 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":"Done"}
```

It's not related to the length of the HTTP response which here is quite short (less than 40 bytes).

IMHO, this makes obvious that the buffer holding the WebQuery response (`read0` in the code linked) gets corrupted when calling `Response_P(...)`. The same problems also affect `AddLog(...)` because before using `Serial.print` I was using AddLog and the log message was written in the WebQuery MQTT response.

arendst, 2023-11-26:
What happens if you copy the result from http.getstring() into a new variable? It might well be the http result is freed by garbage collection before the rest of the code is finished....

barbudor, 2023-11-26:
It looks better. No corruption so far
I need to find a way to get a proper test running for the night

barbudor, 2023-11-26:
PR sent

PR #20111 (2023-11-27): Fix ESP32 webquery response
## Description:

**Related issue (if applicable):** fixes #20102 

Creating a copy of the response seems to solve the problem

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [X] The code change is tested and works with Tasmota core ESP32 V.2.0.14
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/dev [... cut]
closed 2023-11-27 by arendst