arendst/Tasmota #20109: PWM on ESP32 causes high LoadAvg with default setting of SetOption15

smhc, 2023-11-26:
### PROBLEM DESCRIPTION

The use of PWM on an ESP32 appears to use software PWM instead of hardware, based purely on observations of LoadAvg. I questioned this in this thread https://github.com/arendst/Tasmota/discussions/20104 , but on more investigation it does seem to suggest a possible bug.

The documentation suggests that ESP32 should be using hardware PWM for light based PWM: https://tasmota.github.io/docs/Lights/#pwm6

However the LoadAvg suggests otherwise. Using "SetOption15 0" and performing "PWM1 1000" instead of the equivalent power/dimmer commands results in a more reasonable "LoadAvg".

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

21:38:29.486 MQT: stat/tasmota_EC01C4/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"416":"PWM1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
n/a

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:35:18.976 CMD: status 0
21:35:18.978 SRC: WebConsole from 0.0.0.0
21:35:18.979 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
21:35:18.986 MQT: stat/tasmota_EC01C4/STATUS = {"Status":{"Module":1,"DeviceName":"DevBoard32","FriendlyName":["DevBoard32"],"Topic":"tasmota_EC01C4","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[15,15,15,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:35:19.022 MQT: stat/tasmota_EC01C4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset digital core","Uptime":"0T00:01:38","StartupUTC":"2023-11-26T11:33:41","Sleep":220,"CfgHolder":4617,"BootCount":359,"BCResetTime":"2023-09-12T05:56:38","SaveCount":2544}}
21:35:19.058 MQT: stat/tasmota_EC01C4/STATUS2 = {"StatusFWR":{"Version":"13.2.0.2(tasmota32)","BuildDateTime":"2023-11-24T16:25:17","Core":"2_0_14","SDK":"4.4.6.231122","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"386/699"}}
21:35:19.083 MQT: stat/tasmota_EC01C4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xx",""],"TelePeriod":600,"Resolution":"558180C0","SetOption":["0000A00B","2805C80001000600003C5A0A192800000000","00008080","00006000","00004000","00000000"]}}
21:35:19.128 MQT: stat/tasmota_EC01C4/STATUS4 = {"StatusMEM":{"ProgramSize":1878,"Free":1001,"Heap":146,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
21:35:19.205 MQT: stat/tasmota_EC01C4/STATUS5 = {"StatusNET":{"Hostname":"tasmota-EC01C4-0452","IPAddress":"192.168.1.172","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"C8:F0:9E:EC:01:C4","IP6Global":"fd06:8df3:e47f:940:caf0:9eff:feec:1c4","IP6Local":"fe80::caf0:9eff:feec:1c4%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:35:19.258 MQT: stat/tasmota_EC01C4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.118","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EC01C4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:35:19.287 MQT: stat/tasmota_EC01C4/STATUS7 = {"StatusTIM":{"UTC":"2023-11-26T11:35:19","Local":"2023-11-26T21:35:19","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+10:00","Sunrise":"17:15","Sunset":"01:59"}}
21:35:19.309 MQT: stat/tasmota_EC01C4/STATUS10 = {"StatusSNS":{"Time":"2023-11-26T21:35:19"}}
21:35:19.315 BRY: GC from 7645 to 4196 bytes, objects freed 9/45 (in 0 ms) - slots from 50/122 to 50/122
21:35:19.320 MQT: stat/tasmota_EC01C4/STATUS11 = {"StatusSTS":{"Time":"2023-11-26T21:35:19","Uptime":"0T00:01:38","UptimeSec":98,"Heap":145,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":45},"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xx","BSSId":"D4:DA:21:5B:61:1A","Channel":11,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

Using either a "ESP32-S3" or "ESP32-C3-12F" (tried both)

1. Configure a pin as PWM
2. cmd: Power1 1
3. Wait 20 seconds...
4. cmd: Status 11
```
stat/tasmota_EC01C4/STATUS11 = {"StatusSTS":{"Time":"2023-11-26T21:18:34","Uptime":"0T00:02:48","UptimeSec":168,"Heap":151,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99
```
5. cmd: SetOption15 0
6. cmd: PWM1 1000
7. Wait 20 seconds...
8. cmd: Status 11
```
tat/tasmota_EC01C4/STATUS11 = {"StatusSTS":{"Time":"2023-11-26T21:20:06","Uptime":"0T00:00:31","UptimeSec":31,"Heap":151,"SleepMode":"Dynamic","Sleep":220,"LoadAvg":6,
```

### EXPECTED BEHAVIOUR

I would expect the "LoadAvg" to be similar based on the fact hardware based PWM should be used in both scenarios.

### SCREENSHOTS

n/a

### ADDITIONAL CONTEXT

https://tasmota.github.io/docs/Lights/#pwm6
https://github.com/arendst/Tasmota/discussions/20104 

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-11-26:
There is no software PWM for ESP32x The difference in LoadAvg is caused by different code parts used when so15 is set or not. Not a bug.

closed 2023-11-26 by Jason2866

smhc, 2023-11-26:
Ok thanks. It does seem unusual that the load would be so high given that the same can be achieved with a fraction of the load by simply setting the PWM values manually. Once set I wouldn't think any more processing is required.

It suggests that writing a bunch of rules that enables the appropriate PWM values for different colours is more efficient than using SetOption15?

Or maybe the load is nothing to be concerned about? The doco does state "For optimal operation of the Tasmota firmware, it is recommended to keep your device running at a LoadAvg value of 75 or lower."

Jason2866, 2023-11-26:
As long the device works as expected nothing to worry about LoadAvg. If it does not what it should LoadAvg is an indicator. The mentioned value 75 is for an device which has no workload.

sfromis, 2023-11-26:
LoadAvg does not mean CPU load.

smhc, 2023-11-27:
Ok, however the guides on monitoring performance/power suggest setting "Sleep" high and ensuring "LoadAvg" is low, especially when it is effectively idle. And indeed, sleep is high and loadavg is low when the module isn't doing anything. 

When using a PWM pin and "Power 1", the Sleep is forced to 10 and LoadAvg set to 99, which from the docs suggest is poor with respect to monitoring performance/power.

It is just surprising that these values go in the "wrong" direction when the module still shouldn't be doing anything other than enabling a mode on a pin.

PR #20146 (2023-12-26): Only reduce sleep for lights if necessary
## Description:

**Related issue (if applicable):** fixes #20109

There may be other cases where a reduced sleep is necessary, these should be added to the or clause of the if condition. There is no need to reduce sleep for the basic "on" case, it only increases load unnecessarily for a very common use case.

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is te [... cut]