arendst/Tasmota #20118: The PWM Buzzer does not buzz

thomas374b, 2023-11-28:
### PROBLEM DESCRIPTION
When using a PWM buzzer (SetOption111 1) the buzzer make no sound, only silent clicks are to hear

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff DualR3
- [x] Tasmota binary firmware version number used: 13.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- connect a passive buzzer to a pin
- assign this pin to Buzzer function
- type in console  Buzzer 1,10

### EXPECTED BEHAVIOUR
 one second of noise with approx. 977Hz

### ACTUAL BEHAVIOUR
 two silent clicks with approx. 1 second distance

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
as I already mentioned in issue #8794

It seems to be that the function
analogWrite(pin, duty);
in method
BuzzerSet()
in file
xrdv_24_buzzer.ino
is wrapping the duty to 8 bits.

I verified it with a scope and customized self-compiled firmware
with constant
duty := 254 I get sound with a 8µs long L-Pulse
duty := 255 I get nothing, but one click
duty := 256 I get nothing, but one click
duty := 257 I get sound with a 4µs long H-pulse
duty := 128 I get sound with equal length L- and H- pulse

The argument in the original code id pwm_range/2. But pwm_range is 1023 and 511 folded to 8bit is 255
Thats why, the buzzer is only silent clicking instead of buzzing

**(Please, remember to close the issue when the problem has been addressed)**

thomas374b, 2023-11-28:
BTW: It would be nice if frequency and duty could be configured. Depending on the used hardware (there might be capacitors) the buzzer could be at loudest on a certain (frequency,duty) setting

Staars, 2023-12-13:
There is some work done and I already have a driver for testing, that should be able to do all of that via Berry.
As I could write a prototype driver on top of it in Berry to play arbitrary RTTTL ringtones with user defined volume levels, I suppose this should cover a variety of needs.

The rest of the development is more about how (which API will be „invented“ or if it will be included in an existing module ) this can be integrated, not if it is technical possible.

arendst, 2023-12-13:
@staars I think the OP has issues with the standard driver which supports both ESP8266 ans ESP32. As berry only supports ESP32 your solution won't fix the ESP8266 but might fix the OP's issue ;-)

I'll have a look at the standard driver while you are free to add your ESP32 berry solution.

arendst, 2023-12-13:
As a workaround on ESP32 set pwmrange to 255 with command ``PwmRange 255``. This seems to work for me.

Still need to find out why the other pwmranges (511 and 1023) do not work ...

thomas374b, 2023-12-16:
The "buzzer" is a standard PC speaker connected via a Transistor and works fine with up to 8kHz when the duty is set to 127. It's not a problem of the hardware.

When using the pin as PWM, different code comes into play as when the pin is used as "Buzzer".  
-  The "Buzzer"-functionality uses a "general system timer" with fixed frequency to 977 Hz via ESP32 "ledc_xxx" API. This timer is already initialized somewhere else and just used.    
-  The "PWM"-functionality assign and initializes its own timer per PWM channel.

closed 2023-12-21 by arendst

thomas374b, 2023-12-23:
Yes, I can confirm that the correction from https://github.com/arendst/Tasmota/commit/9e2ae391fc33bca88acdeb3a468912840936bb59 works.

With PwmFrequency command one can change the pitch of the buzzer and it is load and clear to hear