arendst/Tasmota #20120: Turning off RGBCWWW bulbs crashes Tasmota

jsprenkle, 2023-11-28:
### PROBLEM DESCRIPTION
Hitting "Toggle" button on web admin page causes device to reboot.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): "Kauf Bulb"  template from Kauf:  {"NAME":"Kauf Bulb","GPIO":[0,0,0,0,416,419,0,0,417,420,418,0,0,0],"FLAG":0,"BASE":18}
- [X] Tasmota binary firmware version number used:    Program Version	13.2.0(tasmota)
Build Date & Time	2023-10-19T09:02:07
Core/SDK Version	2_7_4_9/2.2.2-dev(38a443e)
  - [X] Pre-compiled   Stock build from the website.
  - [ ] Self-compiled
- [X] Flashing tools used: _ Kauf comes with ESPHOME installed. Used the main menu option of ESPHOME to upload .bin.gz file
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

21:55:54.683 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
21:55:54.700 SRC: Backlog
21:55:54.703 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
21:55:54.710 MQT: stat/tasmota_DF322A/RESULT = {"NAME":"Kauf Bulb","GPIO":[0,0,0,0,416,419,0,0,417,420,418,0,0,0],"FLAG":0,"BASE":18}
21:55:54.911 SRC: Backlog
21:55:54.914 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
21:55:54.918 MQT: stat/tasmota_DF322A/RESULT = {"Module":{"0":"Kauf Bulb"}}
21:55:55.116 SRC: Backlog
21:55:55.119 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
21:55:55.128 MQT: stat/tasmota_DF322A/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"419":"PWM4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"420":"PWM5"},"GPIO14":{"418":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: N/A

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

21:57:45.684 CMD: status 0
21:57:45.686 SRC: WebConsole from 192.168.1.124
21:57:45.689 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
21:57:45.696 MQT: stat/tasmota_DF322A/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota-DF322A","FriendlyName":["tasmota-DF322A"],"Topic":"tasmota_DF322A","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:57:45.705 MQT: stat/tasmota_DF322A/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T00:34:38","StartupUTC":"2023-11-28T20:23:07","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2023-11-27T23:13:13","SaveCount":80,"SaveAddress":"F4000"}}
21:57:45.715 MQT: stat/tasmota_DF322A/STATUS2 = {"StatusFWR":{"Version":"13.2.0(tasmota)","BuildDateTime":"2023-10-19T09:02:07","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"373/699"}}
21:57:45.725 MQT: stat/tasmota_DF322A/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Sprenkle",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C80001000600003C5A0A192800000000","00000080","00806000","00004000","00000000"]}}
21:57:45.740 MQT: stat/tasmota_DF322A/STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
21:57:45.747 MQT: stat/tasmota_DF322A/STATUS5 = {"StatusNET":{"Hostname":"tasmota-DF322A-4650","IPAddress":"192.168.1.166","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C4:5B:BE:DF:32:2A","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:57:45.755 MQT: stat/tasmota_DF322A/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.234","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DF322A","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:57:45.765 MQT: stat/tasmota_DF322A/STATUS7 = {"StatusTIM":{"UTC":"2023-11-28T20:57:45","Local":"2023-11-28T21:57:45","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:18","Sunset":"16:57"}}
21:57:45.773 MQT: stat/tasmota_DF322A/STATUS10 = {"StatusSNS":{"Time":"2023-11-28T21:57:45"}}
21:57:45.784 MQT: stat/tasmota_DF322A/STATUS11 = {"StatusSTS":{"Time":"2023-11-28T21:57:45","Uptime":"0T00:34:38","UptimeSec":2078,"Heap":19,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":31,"Color":"4F4F4F0000","HSBColor":"262,0,31","White":0,"CT":153,"Channel":[31,31,31,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Sprenkle","BSSId":"F4:F5:E8:6A:44:8F","Channel":1,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

(This is post reboot dump. Once you hit the toggle button you cannot return to the console to obtain output)

22:14:36.980 MQT: stat/tasmota_DF322A/RESULT = {"GPIOs2":{"6656":"TM1637 DIO","6688":"DLP Tx","6720":"DLP Rx","6752":"SSD1351 DC","6784":"XPT2046 CS","6816":"CSE7761 Tx","6848":"CSE7761 Rx","6880":"VL53LXX XSHUT","6912":"MAX7219 CLK","6944":"MAX7219 DIN","6976":"MAX7219 CS","7008":"TFmini+ TX","7040":"TFmini+ RX","7072":"ZC Pulse","7104":"Input","7136":"I2S DOUT","7168":"I2S BCLK","7200":"I2S WS","7232":"I2S DIN","7264":"I2S WS IN","7296":"I2S BCLK IN","7328":"Interrupt","7360":"MCP2515 CS","7392":"HRG15 Tx","7424":"HRG15 Rx","7456":"VINDRIKTNING","7488":"BL0939 Rx","7520":"BL0942 Rx","7552":"HM330X SET","7584":"Heartbeat","7616":"Heartbeat_i","7648":"74x595 SRCLK","7680":"74x595 RCLK","7712":"74x595 OE","7744":"74x595 SER","7776":"SolaxX1 RTS","7808":"Option E","7840":"SDM230 Tx","7872":"SDM230 Rx","7904":"ADC MQ","7936":"CM110x TX","7968":"CM110x RX","8000":"BL6523 Tx","8032":"BL6523 Rx","8064":"ADE7880 IRQ","8096":"Reset","8128":"MS01","8160":"SDIO CMD","8192":"SDIO CLK","8224":"SDIO D0","8256":"SDIO D1","8288":"SDIO D2","8320":"SDIO D3","8352":"Flowrate","8384":"BP5758D Clk","8416":"BP5758D Dat","8448":"SM2335 Clk","8480":"SM2335 Dat","8512":"MP3 Busy","8544":"TM1621 CS","8576":"TM1621 WR","8608":"TM1621 RD","8640":"TM1621 DAT","8672":"Relay_b","8704":"Relay_bi","8736":"I2S MCLK","8768":"ModBr Tx","8800":"ModBr Rx","8832":"ADE7953 RST","8864":"NrgMbs Tx","8896":"NrgMbs Rx","8928":"ADE7953 CS","8960":"Dali RX","8992":"Dali TX","9024":"BP1658CJ Clk","9056":"BP1658CJ Dat","9088":"Dingtian CLK","9120":"Dingtian SDI","9152":"Dingtian Q7","9184":"Dingtian PL","9216":"Dingtian RCK","9248":"LD2410 Tx","9280":"LD2410 Rx","9312":"ModBr Tx Ena","9344":"NrgMbs Tx Ena","9376":"ME007 Tri","9408":"ME007 Rx","9440":"TuyaMCUBr Tx","9472":"TuyaMCUBr Rx","9504":"BioPDU PZEM0XX Tx","9536":"BioPDU PZEM016 Rx","9568":"BioPDU Bit","9600":"MCP23xxx Int","9632":"MCP23Sxx CS","9664":"PCF8574 Int","9696":"LoxO2 RX","9728":"GM861 Tx","9760":"GM861 Rx","9792":"Dingtian OE","9824":"HDMI CEC","9856":"HC8 Rx","9888":"I2S DAC"}}
22:14:37.013 SRC: MQTT
22:14:37.015 CMD: Grp 0, Cmd 'SWITCHDEBOUNCE', Idx 1, Len 0, Pld -99, Data ''
22:14:37.020 MQT: stat/tasmota_DF322A/RESULT = {"SwitchDebounce":50}
22:14:37.066 SRC: MQTT
[... cut ...]
**Console command**:      backlog Color1 #28114F0000 ; Dimmer 31 ; Power0 1
Press "Toggle" button on main web admin menu.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._ **Bulb to turn off**

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._  Information section from web admin page notes reboot reason as "Hardware Watchdog"

**(Please, remember to close the issue when the problem has been addressed)**
Sure.

s-hadinger, 2023-11-28:
I will try to reproduce, but it looks like a power issue on the device. Can you do `Status 12` right after the restart ?

jsprenkle, 2023-11-28:
> I will try to reproduce, but it looks like a power issue on the device. Can you do `Status 12` right after the restart ?

00:05:16.038 CMD: status12
00:05:16.046 MQT: stat/tasmota_DF322A/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota-DF322A","FriendlyName":["tasmota-DF322A"],"Topic":"tasmota_DF322A","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}

jsprenkle, 2023-11-28:
Is "LoadAvg 99" an issue with the watchdog timer?

sfromis, 2023-11-29:
The command you should try was `Status 12`, not `Status12` - spaces are very significant to computers :wink:
Seeing `LoadAvg 99` is no issue; but normal part of PWM operation.

Does it also happen at lower brightness?

s-hadinger, 2023-11-29:
> Is "LoadAvg 99" an issue with the watchdog timer?

No. LoadAvg 99 is normal when PMW is enabled

jsprenkle, 2023-11-29:
> The command you should try was `Status 12`, not `Status12` - spaces are very significant to computers 😉 Seeing `LoadAvg 99` is no issue; but normal part of PWM operation.
> 
> Does it also happen at lower brightness?

I used the status12 command as given the first time but got an unknown command response. I assume when I copied and pasted it the browser added some nice stuff like utf encoding. Anyway, here's the data you asked for:

```
15:23:32.187 CMD: Status12
15:23:32.195 MQT: stat/tasmota_DF322A/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota-DF322A","FriendlyName":["tasmota-DF322A"],"Topic":"tasmota_DF322A","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
```
Yes, it does happen at lower brightness. I used the 'dimmer' command. Here's the command that sets up the failure:
`backlog Color1 #28114F0000 ; Dimmer 31 ; Power0 1`

sfromis, 2023-11-29:
Not clear what you mean around `Status 12`. Just do **not** do it as `Status12`, but `Status 12` (**with** a space) as asked for. And it should be after a restart with the failure, without further restarts having been done.

jsprenkle, 2023-11-29:
Setup for failure:

```
17:17:54.512 CMD: backlog Color1 #28114F0000 ; Dimmer 31 ; Power0 1
17:17:54.533 MQT: stat/tasmota_DF322A/RESULT = {"POWER":"ON","Dimmer":31,"Color":"28114F0000","HSBColor":"262,78,31","White":0,"CT":153,"Channel":[15,6,31,0,0]}
17:17:54.745 MQT: stat/tasmota_DF322A/RESULT = {"POWER":"ON","Dimmer":31,"Color":"28114F0000","HSBColor":"262,78,31","White":0,"CT":153,"Channel":[15,6,31,0,0]}
17:17:54.947 MQT: stat/tasmota_DF322A/RESULT = {"POWER":"ON"}
17:17:54.951 MQT: stat/tasmota_DF322A/POWER = ON (retained)
```
Press toggle button on main menu
```

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F9, Count 107
00:00:00.057 QPC: Count 1
00:00:00.085 Project tasmota - tasmota-DF322A Version 13.2.0(tasmota)-2_7_4_9(2023-10-19T09:02:07)
00:00:00.525 WIF: Connecting to AP1 Sprenkle Channel 1 BSSId F4:F5:E8:6A:44:8F in mode 11n as tasmota-DF322A-4650...
00:00:01.762 WIF: Connected
00:00:02.012 HTP: Web server active on tasmota-DF322A-4650 with IP address 192.168.1.166
17:18:49.005 MQT: Attempting connection...
17:18:49.021 MQT: Connected
17:18:49.024 MQT: tele/tasmota_DF322A/LWT = Online (retained)
17:18:49.026 MQT: cmnd/tasmota_DF322A/POWER = 
17:18:49.031 MQT: tele/tasmota_DF322A/INFO1 = {"Info1":{"Module":"Kauf Bulb","Version":"13.2.0(tasmota)","FallbackTopic":"cmnd/DVES_DF322A_fb/","GroupTopic":"cmnd/tasmotas/"}}
17:18:49.034 MQT: tele/tasmota_DF322A/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-DF322A-4650","IPAddress":"192.168.1.166"}}
17:18:49.036 MQT: tele/tasmota_DF322A/INFO3 = {"Info3":{"RestartReason":"Hardware Watchdog","BootCount":38}}
17:18:49.038 MQT: stat/tasmota_DF322A/RESULT = {"POWER":"ON"}
17:18:49.040 MQT: stat/tasmota_DF322A/POWER = ON (retained)
17:18:52.418 QPC: Reset
17:18:53.407 MQT: tele/tasmota_DF322A/STATE = {"Time":"2023-11-29T17:18:53","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":45,"MqttCount":1,"POWER":"ON","Dimmer":31,"Color":"28114F0000","HSBColor":"262,78,31","White":0,"CT":153,"Channel":[15,6,31,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Sprenkle","BSSId":"F4:F5:E8:6A:44:8F","Channel":1,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:03"}}

Issued "status12"
```

17:19:34.458 CMD: status12
17:19:34.465 MQT: stat/tasmota_DF322A/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota-DF322A","FriendlyName":["tasmota-DF322A"],"Topic":"tasmota_DF322A","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
```

```

sfromis, 2023-11-29:
Do not waste time with `Status12`, it **has** to be `Status 12`, but it only provides details if used after a crash of Tasmota. After a restart which was not due to a crash, the error is the correct result.

jsprenkle, 2023-11-29:
> Do not waste time with `Status12`, it **has** to be `Status 12`, but it only provides details if used after a crash of Tasmota. After a restart which was not due to a crash, the error is the correct result. And "Hardware Watchdog" does not count as a "crash".
> 
> To illustrate, here's an example of using `Status 12` (with a space) after a crash: `18:33:47.167 MQT: xxxxxxxx/stat/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC", .....`

Good. Glad it provided some useful info, even if it just eliminating something from consideration.

s-hadinger, 2023-11-29:
I cannot reproduce. I did the same configuration on a Wemos D1 Mini, no crash. I really suspect a power issue on the device. It's not a software problem