arendst/Tasmota #20161: Sensor Temperature and Humidity AHT21

Haui74, 2023-12-04:
Good morning! 🙂
Has anyone got the AHT21 sensor integrated into Tasmota yet? Tasmota finds 0x38 during the I2C scan, but I don't see measurements (I2C driver 43 is also active).

barbudor, 2023-12-04:
Did you self compiled or using a rebuild binary?
It looks like the driver 43 can be compiled either for AHT1x or AHT2x but still report as driver 43.
Confusing coding decision

I can't tell for sure that compiling for AHT2x will work with AHT21 as only AHT20 is referenced in the doc and code but at least I wouldn't expect driver 43 to work if compiled as USE_AHT1x.

Requires
```
#ifdef USE_AHT1x
#undef USE_AHT1x
#endif
#ifndef USE_AHT2x
#define USE_AHT2x
#endif 
```

Haui74, 2023-12-04:
Its self compiled, i used "Sensor"+ use_AHT2x. 

> 0:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F7, Count 16
00:00:00.055 QPC: Count 1
00:00:00.077 Project tasmota - Tasmota Version 13.2.0.2(sensors)-2_7_4_9(2023-12-04T08:43:29)
00:00:00.256 I2C: EN16X found at 0x53
00:00:00.800 WIF: Connecting to AP1 Hauis Channel 3 BSSId EC:41:18:EB:E5:09 in mode 11n as tasmota-6D1253-4691...
00:00:03.777 WIF: Connected
00:00:04.030 HTP: Web server active on tasmota-6D1253-4691 with IP address 192.168.0.88
08:44:56.036 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"13.2.0.2(sensors)","FallbackTopic":"cmnd/DVES_6D1253_fb/","GroupTopic":"cmnd/tasmotas/"}}
08:44:56.041 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-6D1253-4691","IPAddress":"192.168.0.88"}}
08:44:56.051 RSL: INFO3 = {"Info3":{"RestartReason":"External System","BootCount":8}}
08:44:57.190 QPC: Reset
08:45:00.138 RSL: STATE = {"Time":"2023-12-04T08:45:00","Uptime":"0T00:00:11","UptimeSec":11,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Hauis","BSSId":"EC:41:18:EB:E5:09","Channel":3,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}
08:45:00.159 RSL: SENSOR = {"Time":"2023-12-04T08:45:00","EN16X":{"AQIS":2,"eCO2":512,"TVOC":79}}
08:45:22.130 CMD: I2Cscan
08:45:22.150 RSL: RESULT = {"I2CScan":"Device(s) found at 0x38 0x53"}
08:45:28.996 CMD: I2CDriver
08:45:28.003 RSL: RESULT = {"I2CDriver":"7,8,9,10,11,!12,13,14,15,17,18,20,24,29,31,36,41,42,43,44,46,48,69,76,84"}

sfromis, 2023-12-04:
You generally cannot add features to the sensors precomposed mix, but instead start from just "tasmota" instead of "tasmota-sensors", and add what you need.