arendst/Tasmota #20166: KNX sends wrong value even if hint in the text ist correct.

Elektron79, 2023-12-04:
### PROBLEM DESCRIPTION
KNX Button always sents Value 1, even if the text in logs toggels between "power on" and "power off".

19:47:46.168 KNX: Button 1 = 1 sent to 1/0/4
19:47:46.172 RSL: RESULT = {"POWER":"ON"}
19:47:46.174 RSL: POWER = ON
19:47:52.157 KNX: Button 1 = 1 sent to 1/0/4
19:47:52.161 RSL: RESULT = {"POWER":"OFF"}
19:47:52.163 RSL: POWER = OFF
19:47:59.538 KNX: Button 1 = 1 sent to 1/0/4
19:47:59.542 RSL: RESULT = {"POWER":"ON"}
19:47:59.544 RSL: POWER = ON
19:48:03.141 KNX: Button 1 = 1 sent to 1/0/4
19:48:03.146 RSL: RESULT = {"POWER":"OFF"}
19:48:03.148 RSL: POWER = OFF

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NodeMCU
- [ ] Tasmota binary firmware version number used: 13.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: NodeMCU_Pyflasher
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:47:46.168 KNX: Button 1 = 1 sent to 1/0/4
19:47:46.172 RSL: RESULT = {"POWER":"ON"}
19:47:46.174 RSL: POWER = ON
19:47:52.157 KNX: Button 1 = 1 sent to 1/0/4
19:47:52.161 RSL: RESULT = {"POWER":"OFF"}
19:47:52.163 RSL: POWER = OFF
19:47:59.538 KNX: Button 1 = 1 sent to 1/0/4
19:47:59.542 RSL: RESULT = {"POWER":"ON"}
19:47:59.544 RSL: POWER = ON
19:48:03.141 KNX: Button 1 = 1 sent to 1/0/4
19:48:03.146 RSL: RESULT = {"POWER":"OFF"}
19:48:03.148 RSL: POWER = OFF

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

19:47:46.168 KNX: Button 1 = 1 sent to 1/0/4
19:47:46.172 RSL: RESULT = {"POWER":"ON"}
19:47:46.174 RSL: POWER = ON
19:47:52.157 KNX: Button 1 = **0** sent to 1/0/4
19:47:52.161 RSL: RESULT = {"POWER":"OFF"}
19:47:52.163 RSL: POWER = OFF
19:47:59.538 KNX: Button 1 = 1 sent to 1/0/4
19:47:59.542 RSL: RESULT = {"POWER":"ON"}
19:47:59.544 RSL: POWER = ON
19:48:03.141 KNX: Button 1 = **0** sent to 1/0/4
19:48:03.146 RSL: RESULT = {"POWER":"OFF"}
19:48:03.148 RSL: POWER = OFF
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-12-04:
Missing your KNX configuration

barbudor, 2023-12-04:
Actually if you mapped the Button to be sent, you get what you asked: a message for each button action

If what you are interested is the relay state, then you should map the relay.

Elektron79, 2023-12-04:
Hi barbudor,

I Plan to use a simple Push Button to Switch the light on via an KNX Actor. Like i do it on the KNX connected Wall Switch. 

- Press the Button for the First Time -> light is switched on (this is working via Tasmota.
- Press the Button for the second Time -> light is switched Off (the Message on the Condoleezza Shows „Power Off“ But via KNX the „ON“-Message is sent again.

My expectation was, that the second Telegramm sende an off Message as mentioned in the Console log.

I can Share the KNX configuration but it‘s simple Button1 -> 1/0/4

If I can Support with more background just let me know…

Thank you for your help and wär Michael