arendst/Tasmota #20201: Shelly 1PM no longer starts on 230V with V13.2.0

Bodengriller, 2023-12-10:
I did an OTA upgrade from 9.1 to 13.2 on my [Shelly 1PM](https://tasmota.github.io/docs/devices/Shelly-1PM/) -> after that it won't start anymore.
There is a new WLAN "ESP-060496" - you can also connect to it, you also get the 192.168.4.2 assigned as IP, but you cannot open the 192.168.4.1.

I then flashed the Shelly via FTDI with V13.2 -> no problems. Restored config backup -> no problem -> 2x reboot -> no problem
FTDI removed and 230V connected -> Shelly does not come back.

Any idea what I can do or what the problem could be?

I have a second Shelly 1PM to test, which shows the same phenomena.

Very strange - no problems with V9.1 or 3,3V as Power Supply

[Waschmaschine-9_1_0_tasmota-2023-12-10_21_41_25.dmp](https://github.com/arendst/Tasmota/files/13629487/Waschmaschine-9_1_0_tasmota-2023-12-10_21_41_25.dmp)

barbudor, 2023-12-10:
Although the [Upgrade path](https://tasmota.github.io/docs/Upgrading/#upgrade-flow) say that you can go straight from 9.1 to latest version, I generally step by each major version.
That's may be superstition but 🤷‍♂️ 

Regarding serial flashing, FTDI adapters is known for providing poor quality power supply.
If you could redo the serial flashing with a stronger 3.3V, that should be better.
Recommended [Serial USB adapter](https://tasmota.github.io/docs/Getting-Started/#serial-programmer) should have a dedicated 3.3V regulator such as this one:
[image]
[image]

stefanbode, 2023-12-11:
I assume that you have problems with the increased power request coming with 13.0 can you check if the 12.0 version is working on the Shelly 1PM? There where some changes in TASMOTA with 13 that causes at least at the SONOFF devices problems with the 3.3V power supply. You error description matches to this one. A 7833 3.3V voltage regulator in parallel or a capacitor 470microF with some soldering knowledge can make it.

arendst, 2023-12-11:
I can only confirm a Shelly 1PM should still work with Tasmota v13.2:

```
00:00:00.001 HDW: ESP8266EX
00:00:00.046 CFG: Loaded from flash at FB, Count 31
00:00:00.106 CFG: Saved to flash at FB, Count 32, Bytes 4096
00:00:00.137 QPC: Reset
00:00:00.143 CFG: CR 474/699, Busy 0
00:00:00.155 ROT: Mode 1
00:00:00.156 NRG: Init driver 1
00:00:00.157 SRC: Restart
00:00:00.158 Project tasmota - Shelly 1PM Version 13.2.0(tasmota)-2_7_4_9(2023-10-19T09:02:07)
00:00:00.176 DSB: Sensors found 0
00:00:00.431 WIF: Attempting connection...
00:00:00.433 WIF: Network (re)scan started...
00:00:01.292 WIF: Attempting connection...
00:00:02.295 WIF: Attempting connection...
00:00:03.296 WIF: Attempting connection...
00:00:03.297 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -32, Encryption 1
00:00:03.635 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as shelly1pm...
00:00:04.757 WIF: Connected
00:00:05.010 HTP: Web server active on shelly1pm with IP address 192.168.2.144
00:00:05.558 RTC: UTC 2023-12-11T13:42:32, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
14:42:32.001 RTC: Synced by NTP
14:42:32.019 MQT: Attempting connection...
14:42:32.040 MQT: Connected
14:42:32.043 MQT: tele/shelly1pm/LWT = Online (retained)
14:42:32.045 MQT: cmnd/shelly1pm/POWER = 
14:42:32.047 MQT: Subscribe to cmnd/shelly1pm/#
14:42:32.048 MQT: Subscribe to cmnd/tasmotas/#
14:42:32.050 MQT: Subscribe to cmnd/ingebruik/#
14:42:32.051 MQT: Subscribe to cmnd/DVES_6097E9_fb/#
14:42:32.055 MQT: tele/shelly1pm/INFO1 = {"Info1":{"Module":"Shelly 1PM","Version":"13.2.0(tasmota)","FallbackTopic":"cmnd/DVES_6097E9_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:42:32.058 MQT: tele/shelly1pm/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"shelly1pm","IPAddress":"192.168.2.144"}}
14:42:32.060 MQT: tele/shelly1pm/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":7}}
14:42:32.063 MQT: stat/shelly1pm/RESULT = {"POWER":"OFF"}
14:42:32.064 MQT: stat/shelly1pm/POWER = OFF
14:42:33.476 NRG: Energy reset by invalid data
14:42:34.476 APP: Boot Count 7
14:42:34.579 CFG: Saved to flash at FA, Count 33, Bytes 4096
14:42:36.488 MQT: tele/shelly1pm/STATE = {"Time":"2023-12-11T14:42:36","Uptime":"0T00:00:11","UptimeSec":11,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:06"}}
14:42:36.498 MQT: tele/shelly1pm/SENSOR = {"Time":"2023-12-11T14:42:36","Switch1":"ON","ANALOG":{"Temperature":29.6},"ENERGY":{"TotalStartTime":"2019-05-18T13:49:50","Total":0.002,"Yesterday":0.002,"Today":0.000,"Period":0.00,"Power":0.00},"TempUnit":"C"}
14:42:50.037 HTP: Console
```

The above suggestion regarding internal power supply might well be the problem. I had the same issue on a Shelly 2.5 which now has an "external" capacitor (as I didn't have the needed small one) fixing the power issue.

Bodengriller, 2023-12-11:
> I assume that you have problems with the increased power request coming with 13.0 

>The above suggestion regarding internal power supply might well be the problem. 

You are probably right!
I have now tried a few things.
First I got my second Shelly 1PM to run with v13.2...something just went wrong during flashing - after a new flash it works perfectly with v13.2.

With the "old" colleague the problem remains -> with 3.3V it also works with v13.2.
At 230V it fails. -> therefore very obviously the internal power supply.
[image]

I then tested the following versions in descending order:
```
v12.5
v11.1
v10.0
```

With or up to v10.0, the old buddy works perfectly on 230V.

>A 7833 3.3V voltage regulator in parallel or a capacitor 470microF with some soldering knowledge can make it.
> I had the same issue on a Shelly 2.5 which now has an "external" capacitor (as I didn't have the needed small one) fixing the power issue.

Now I'm actually a bit afraid that one Shelly will die after the next.
I have a few of them in use.
On the other hand, I'm not sure why I want to use the latest Tasmota - it doesn't really make sense.
To be honest, I was still hoping inside that it was somehow due to my config

Do you have any instructions or pictures of where to solder the capacitor?
Apparently you should do this as a precaution

[image]
[image]