arendst/Tasmota #20206: Telegram bot support is not documented

bam80, 2023-12-11:
Telegram bot support was implemented but apparently not documented:
https://github.com/arendst/Tasmota/issues/8619

We need proper documentation rather than just source code comments:
https://github.com/arendst/Tasmota/issues/8619#issuecomment-1445345866

barbudor, 2023-12-11:
You are welcome to contribute to the documentation
Thanks for your help

bam80, 2023-12-11:
OK, if I give the feature a try I'll also help to document it maybe.

For now, I'm considering another option though:
_RadioShuttle MQTT push server_ 
https://www.radioshuttle.de/en/mqtt-en/push-notification/

Not sure which is a better solution, maybe both are valuable.

Still we need to document what we have, of course.