arendst/Tasmota #20217: SGP30

f112233, 2023-12-12:
<!- Спасибо за сообщение о проблеме для этого проекта. ПРОЧИТАЙТЕ ЭТО ПЕРВЫМ:

Этот шаблон выпуска предназначен для ОТЧЕТА ПРОГРАММНОГО ОБЕСПЕЧЕНИЯ Tasmota ПРОБЛЕМЫ ТОЛЬКО

ПОЖАЛУЙСТА, НЕ ОТКРЫВАЙТЕ ВОПРОС:
  - Если ваша версия Tasmota не является последней из ветки разработки, пожалуйста, обновите устройство перед отправкой вашей проблемы. Ваша проблема может быть уже решена. Последние предварительно скомпилированные двоичные файлы Tasmota можно скачать с http://ota.tasmota.com/tasmota/
  - Если у вас есть проблема, когда мигание было сделано через Tuya Convert, пожалуйста, обратитесь к Tuya Convert Team
  - Если ваша проблема является мигающей проблемой, пожалуйста, обратитесь к ней в чат поддержки [Tasmota](https://discord.gg/Ks2Kzd4)
  - Если ваша проблема связана с компиляцией, пожалуйста, обратитесь к чату поддержки [Tasmota](https://discord.gg/Ks2Kzd4)
  - Если ваша проблема была решена ранее (то есть, дублированная проблема), пожалуйста, спросите в оригинальной проблеме
  - Если ваша проблема связана с Wi-Fi или проблемой MQTT, пожалуйста, попробуйте шаги, предусмотренные в [FAQ](https://tasmota.github.io/docs/FAQ) и [Troubleshooting](https://tasmota.github.io/docs/Troubleshooting)

 Пожалуйста, потратьте несколько минут, чтобы заполнить запрошенную информацию ниже. Наша способность оказывать помощь сильно затруднена без нее. Запрашиваемые детали потенциально влияют на то, какие варианты следует использовать. Небольшое количество времени, которое вы тратите на заполнение шаблона, также поможет волонтерам, оказывающим вам помощь, сократить время, необходимое для оказания вам помощи.

НЕ УДАЛЯЙТЕ ТЕКСТ ИЗ этого шаблона! В противном случае вопрос будет автоматически закрыт.
->

### ОПИСАНИЕ ПРОБЛЕМЫ
_Ясное и краткое описание проблемы._

### ЗАПРОШЕННАЯ ИНФОРМАЦИЯ
_Убедитесь, что вы выполнили каждый шаг и проверили соответствующие поля перед отправкой вашей проблемы. Спасибо!_

- [ ] Прочитайте [Руководство и политику по внесению взносов](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) и [Кодекс поведения](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [] Поиск проблемы в [выпусках](https://github.com/arendst/Tasmota/issues)
- [ ] Поиск проблемы в [дискуссии](https://github.com/arendst/Tasmota/discussions)
- [ ] Искал проблему в [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Поиск проблемы в [chat](https://discord.gg/Ks2Kzd4)
- [] Используемое устройство (например, Sonoff Basic): _____
- [] Используемый номер версии бинарной прошивки Tasmota: _____
  - [] Предварительно скомпилированный
  - [] Самосоставленный
- [] Используемые инструменты для мигания: _____
- [ ] Обеспечить выход команды: `Backlog Template; Module; GPIO 255`:
```lua
  Вывод конфигурации здесь:

``
- [] Если использовать правила, обеспечьте выход этой команды: `Backlog Rule1; Rule2; Rule3`:
```lua
  Правила вывод здесь:

``
- [ ] Обеспечить выход этой команды: `Status 0`:
```lua
  СТАТУС 0 вывод здесь:

``
- [ ] Set `weblog` to 4 и затем, когда вы испытываете свою проблему, предоставить выход журнала Console:
```lua
  Выход консоли здесь:

``

### ВОСПРОИЗВОДИТЬ
_Шаги для воспроизведения поведения:_

### ОЖИДАЕМОЕ ПОВЕДЕНИЕ
_Ясное и краткое описание того, что вы ожидали._

### СКРИНШОТЫ
_Если применимо, добавьте скриншоты, чтобы помочь объяснить вашу проблему._

### ДОПОЛНИТЕЛЬНЫЙ КОНТЕКСТ
_Добавить любой другой контекст о проблеме здесь._

**(Пожалуйста, не забудьте закрыть проблему, когда проблема была решена)**

sfromis, 2023-12-12:
To raise issues, you need to use the English language, and provide the details.

Jason2866, 2023-12-13:
Closing since not provided the needed Infos and not in English

closed 2023-12-13 by Jason2866