arendst/Tasmota #20235: No ContentLength header set when using HTTP API via HTTP1.0 protocol

kryslt, 2023-12-14:
### PROBLEM DESCRIPTION
There should be Content-Length header set in the response headers when asking Tasmota HTTP server via HTTP1.0.
It is missing now.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Athom Plug V2
- [x] Tasmota binary firmware version number used: 13.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"Athom Plug V2","GPIO":[0,0,0,3104,0,32,0,0,224,576,0,0,0,0],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
{"Status":{"Module":0,"DeviceName":"Tasmota Test","FriendlyName":["Tasmota Test Station"],"Topic":"tasmota_A7EDD4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"2T14:33:23","StartupUTC":"2023-12-12T00:24:59","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2023-03-28T10:00:59","SaveCount":356,"SaveAddress":"F4000"},"StatusFWR":{"Version":"13.2.0(tasmota)","BuildDateTime":"2023-10-19T09:02:07","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285H16","CR":"353/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["tk_net",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","0405C80001000680003C5A0A002800000000","08000080","00806000","00004000","00000000"]},"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"},"StatusNET":{"Hostname":"tasmota-A7EDD4-3540","IPAddress":"192.168.2.41","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"217.77.165.81","DNSServer2":"217.77.165.211","Mac":"34:AB:95:A7:ED:D4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A7EDD4","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-12-14T14:58:22","Local":"2023-12-14T15:58:22","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:52"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2023-12-14T15:58:22","ENERGY":{"TotalStartTime":"2023-03-28T10:00:59","Total":2.072,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":233,"Current":0.000}},"StatusSTS":{"Time":"2023-12-14T15:58:22","Uptime":"2T14:33:23","UptimeSec":225203,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"tk_net","BSSId":"74:B5:7E:3E:EA:7C","Channel":6,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

1. Connect device to wifi at given ip.
2. Set browser HTTP version to HTTP1.0 (for Firefox type about:config and set network.http.version to 1.0)
3. Type http://{ip}/cm?cmnd=Status
4. Inspect response headers.

### EXPECTED BEHAVIOUR
Response headers are:

> <html><body>

Cache-Control | no-cache, no-store, must-revalidate
-- | --
Connection | close
Content-Type | application/json
Expires | -1
Pragma | no-cache
Server | Tasmota/13.2.0 (ESP8285H16)

</body>
</html>

They should be:

> <html><body>

Cache-Control | no-cache, no-store, must-revalidate
-- | --
Connection | close
Content-Type | application/json
Content-Length | xxx
Expires | -1
Pragma | no-cache
Server | Tasmota/13.2.0 (ESP8285H16)

</body>
</html>

### SCREENSHOTS
None.

### ADDITIONAL CONTEXT
I know Content-Length is not mandatory in HTTP1.0 response (https://stackoverflow.com/questions/15991173/is-the-content-length-header-required-for-a-http-1-0-response, https://stackoverflow.com/questions/27228343/what-are-the-consequences-of-not-including-a-content-length-header-in-a-server-r). But it SHOULD be used. Every standard server (nginx, apache) sends Content-Length headers in the response. When using low level communication with embedded system (eg. using UIP or LWIP stacks), you know from the headers what is the size of the entire data, and can allocate buffer with exact size upon parsing the headers.

When using HTTP1.1, Tasmota responds correctly with chunked encoding.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-12-14:
We can't. Sending the content-length implies that you know it in advance, hence buffer the response in memory. We can't because of the limited memory of ESP8266.

Instead we are using chunked encoding, meaning that chunks are sent whenever they are available. Chunked do contain per-chunk content-length.

barbudor, 2023-12-14:
> Every standard server (nginx, apache) sends Content-Length headers in the response. 

Yep. Try to run tasmota on the same hardware as those software and put that into a wall switch or a smart plug.

sfromis, 2023-12-14:
Is there any "harm" in not having the Content-Length header, outside of borderline cases not really pertaining to Tasmota? HTTP/1.1 was already the norm many years before Tasmota was created.

kryslt, 2023-12-14:
@s-hadinger:
I don't know the Tasmota code, but I bet it can be done. E.g. a double call mechanism can be used - measure (without any buffer) and fill (directly to DMA send buffer). Or a similar algorithm. We use even more constrained hardware (a LPCxx ARM Cortex M3 chip) and we generate complex XML/JSON files and always use Content-Length. We also work with Shelly products (smart plugs etc.) that use the same ESP hardware and they always have Content-Length in their HTTP1.0 responses. In HTTP1.0, Content-Length SHOULD be used. Anyway, although possible in HTTP1.1, the chunked encoding is rather unusual. I've not seen this anywhere else than in Tasmota and I came across many simple and constrained web server solutions. 

@sfromis:
I don't know if there is harm, I just report unusual thing. We still use the uIP-1.0 TCP-IP stack (because of resource constraints in the LPC mentioned above) and this stack only supports HTTP1.0. No support for chunked encoding in its webclient code.

sfromis, 2023-12-15:
As long as a big part of web sites are still using HTTP/1.1, it is not at all obsolete. Found a page dated yesterday claiming that HTTP/2 is only used by a bit more than 1/3 of sites (obviously such stats are never representative, and different stats can be found too). And while HTTP/2 indeed no longer uses "chunked", it still supports streaming without the length being known from the start. Using chunked is a "normal thing" when the length is unknown at the start of response generation, and makes more sense than contortions like double call to count bytes - and the 2nd run may not even give same number of bytes without locking all other activity, including the clock moving forward.

What **is** obsolete on the web is HTTP/1.0. But yes, you could say that the Tasmota+Arduino support for accepting a request with HTTP/1.0 is a bit "incomplete" when a client specifying the obsolete protocol will not get a header with `Transfer-Encoding: chunked` but still not get a length. Arguably it would be "more correct" to reject the request as unsupported, but not terribly practical.

s-hadinger, 2023-12-15:
Thanks for your comment but we will not change it in the short term. Doing this is a major refactoring. Generating twice the page can result in different sensor values. I see no benefit in doing such major refactoring 

Tasmota main purpose is mqtt

kryslt, 2023-12-15:
@s-hadinger 
I just reported unusual behavior. I know that between 2 runs the activity must be locked (or better, needs to be double buffered). This might be difficult to implement. The IP stack always needs to have the entire response in a buffer (for TCP retransmissions, I can see them happening in Wireshark for the big STATUS0 response), so maybe the Content-Length could be taken easily from that buffer. But I don't know if this is possible.
If you agree, I will close this issue as not easily fixable based on your responses.

@sfromis
The chunked transfer is pretty unusual in this scenario, never seen this anywhere else than in Tasmota. I can imagine using it for transmitting big data with previously unknown size, like music streaming. But not for transferring small files <1KB big. You are right, HTTP1.0 is obsolete. But not its basic principles, hence the human readable headers + content. IMO, for these simple data transfers, Content-Length approach should be used even in HTTP1.1 and onwards.

s-hadinger, 2023-12-15:
This wouldn't work. The memory of esp8266 is a strong constraint. Chunked encoding allows us to send many small chunks in many small TCP packets. You should see them in Wireshark. So there is now single TCP buffer with the entire response

s-hadinger, 2023-12-15:
Maybe I need to add more information. Tasmota used to have a big buffer for the response, but it generated many copies in memory to accommodate for unknown size and was a major reason for memory exhaustion on edp8266, especially with TLS enabled. 

We switched to chunk encoding as soon as it was supported by Arduino and it was a game changer for web support. Even if we wanted, there is unfortunately no way to turn back to what didn't work well

kryslt, 2023-12-15:
> This wouldn't work. The memory of esp8266 is a strong constraint. Chunked encoding allows us to send many small chunks in many small TCP packets. You should see them in Wireshark. So there is now single TCP buffer with the entire response

No, for the big STATUS0 response I can see 3 packets, one small with headers and 2 big with response. I assumed from that the entire response fitted into one buffer. Do you want to paste the entire Wireshark output here?

Jason2866, 2023-12-15:
Closing since answered, thx for sharing your ideas.

closed 2023-12-15 by Jason2866

kryslt, 2023-12-15:
@s-hadinger 
No, I'm talking about HTTP API all the time. 
Look at [PCAP.zip](https://github.com/arendst/Tasmota/files/13686354/PCAP.zip)
In the attached file there are the same responses in HTTP1.0 and HTTP1.1.
Both responses seem to use a large buffer. 
At least as large as one default packet / MTU size.
You can see in HTTP1.1 that there are more chunks sent in one packet.
Additionally, the individual chunks cross packet boundaries.
From this I assumed that a large buffer was used for the entire message, the size of which could easily be copied into the Content-Length field.

s-hadinger, 2023-12-15:
I'm out until Sunday. I will look at your pcap files then

s-hadinger, 2023-12-18:
There is indeed a buffer of around 1500 bytes before it's sent on the wire. It does not change the fact that the content length is not known in advance and is not stored in a single buffer at any moment.

kryslt, 2023-12-18:
Then it is probably only used for TCP retransmissions of the last packet.
And I understand that, depending on the present software structure, the Content-Length support might be very hard to implement.