arendst/Tasmota #20244: Trying  to use Tasmota for switch only monitoring no relays involved. Set up 2 switches on GPIO4 and GPIO5. Chose mode 15 for the switches so that they report to MQTT only and do not activate power. Doesn't work. All I get is POWER ON in the reports and only for the switch at GPIO 4. No reports for the switch at GPIO 5.

ivanvach, 2023-12-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ivanvach, 2023-12-15:
Trying to use Tasmota for switch only monitoring no relays involved. Set up 2 switches on GPIO4 and GPIO5. Chose mode 15 for the switches so that they report to MQTT only and do not activate power. Doesn't work. All I get is POWER ON in the reports and only for the switch at GPIO 4. No reports for the switch at GPIO 5.

sfromis, 2023-12-15:
You did not complete **any** of the requested details allowing us to check what happens, and maybe explain what you should've done.

ivanvach, 2023-12-15:
Sorry, fields above are closed for me to enter anything. Here is the response to Status 0

00:06:15.673 MQT: stat/tasmota_84BDA8/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_84BDA8","ButtonTopic":"test123","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:06:15.685 MQT: stat/tasmota_84BDA8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T23:02:07","StartupUTC":"2023-12-15T00:04:08","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2023-12-08T02:21:01","SaveCount":133}}
00:06:15.694 MQT: stat/tasmota_84BDA8/STATUS2 = {"StatusFWR":{"Version":"13.2.0(tasmota32)","BuildDateTime":"2023-10-19T09:05:13","Core":"2_0_14","SDK":"4.4.6.231011","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"425/699"}}
00:06:15.703 MQT: stat/tasmota_84BDA8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LITMATH",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000020"]}}
00:06:15.729 MQT: stat/tasmota_84BDA8/STATUS4 = {"StatusMEM":{"ProgramSize":1872,"Free":1007,"Heap":112,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","010A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
00:06:15.745 MQT: stat/tasmota_84BDA8/STATUS5 = {"StatusNET":{"Hostname":"tasmota-84BDA8-7592","IPAddress":"192.168.1.246","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"30:AE:A4:84:BD:A8","IP6Global":"fd0f:de1a:195e:203:32ae:a4ff:fe84:bda8","IP6Local":"fe80::32ae:a4ff:fe84:bda8","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
00:06:15.756 MQT: stat/tasmota_84BDA8/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_84BDA8","MqttUser":"ivan","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:06:15.765 MQT: stat/tasmota_84BDA8/STATUS7 = {"StatusTIM":{"UTC":"2023-12-15T23:06:15","Local":"2023-12-16T00:06:15","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:37","Sunset":"16:53"}}
00:06:15.778 MQT: stat/tasmota_84BDA8/STATUS10 = {"StatusSNS":{"Time":"2023-12-16T00:06:15","Switch15":"ON"}}
00:06:15.795 MQT: stat/tasmota_84BDA8/STATUS11 = {"StatusSTS":{"Time":"2023-12-16T00:06:15","Uptime":"0T23:02:07","UptimeSec":82927,"Heap":108,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":22,"Objects":261},"Wifi":{"AP":1,"SSId":"LITMATH","BSSId":"F4:E2:C6:24:9C:CE","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}}

sfromis, 2023-12-15:
`Status 0` output was not very interesting in this case, more so the output from `Backlog Template; Module; GPIO 255` and also the log (with `weblog 4` activate) when you actually experience the issue.

Readability of copy/pasted data is greatly enhanced by having triple backtick before/after the lines, to make it a block quote. And IIRC you could also edit the first post to add info.

ivanvach, 2023-12-15:
O.K. Here are 3 responses to  Backlog Template; Module; GPIO 255
First with both switches open
Second GPIO 4 to ground GPIO5 open
Third GPIO4 open GPIO5 to ground.

\\\ 00:22:01.011 CMD: Backlog Template; Module; GPIO 255
00:22:01.043 MQT: stat/tasmota_84BDA8/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:22:01.273 MQT: stat/tasmota_84BDA8/RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:22:01.482 MQT: stat/tasmota_84BDA8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"174":"Switch15"},"GPIO5":{"174":"Switch15"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
00:22:51.602 MQT: stat/tasmota_84BDA8/RESULT = {"POWER":"ON"}
00:22:51.604 MQT: stat/tasmota_84BDA8/POWER = ON
00:23:16.056 CMD: Backlog Template; Module; GPIO 255
00:23:16.095 MQT: stat/tasmota_84BDA8/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:23:16.323 MQT: stat/tasmota_84BDA8/RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:23:16.539 MQT: stat/tasmota_84BDA8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"174":"Switch15"},"GPIO5":{"174":"Switch15"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
00:23:40.609 MQT: stat/tasmota_84BDA8/RESULT = {"POWER":"OFF"}
00:23:40.611 MQT: stat/tasmota_84BDA8/POWER = OFF
00:23:58.489 CMD: Backlog Template; Module; GPIO 255
00:23:58.516 MQT: stat/tasmota_84BDA8/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:23:58.754 MQT: stat/tasmota_84BDA8/RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:23:58.961 MQT: stat/tasmota_84BDA8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"174":"Switch15"},"GPIO5":{"174":"Switch15"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}} \\\

ivanvach, 2023-12-15:
Here is what I'm getting when turning on the switch connected to GPIO4

00:35:38.877 MQT: stat/tasmota_84BDA8/RESULT = {"POWER":"ON"}
00:35:38.880 MQT: stat/tasmota_84BDA8/POWER = ON

I get NOTHING when turning the switch that is connected to GPIO5

sfromis, 2023-12-15:
You have a bad config. You are not using SwitchMode 15, but instead you have wrongly configured two Switch pins with index 15. That won't work. You need to use index 1 for the one you want to be the first, and then 2 for the other one. And then use the commands `SwitchMode1 15` and `SwitchMode2 15` to set the switchmode for both switches.

ivanvach, 2023-12-15:
@sfromis Thanks for the prompt attention to this matter. Got it. So the drop box next to the switch in the configuration is just to number them. Let me try and confirm that everything is O.K. If it is than I will close this issue.

ivanvach, 2023-12-15:
Works perfectly. Thanks again for the help. Closing this issue now.

closed 2023-12-15 by ivanvach