arendst/Tasmota #20251: Mem1=string ---> RESULT = {"Mem1":"0.000"}

afgcons, 2023-12-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

16:11:34.955 CMD: Mem1=usbnas.loc/cgi-bin/mocha.cgi
16:11:34.959 MQT: stat/zb1/RESULT = {**"Mem1":"0.000"**}
16:12:26.808 CMD: mem1=usbnas.loc
16:12:26.812 MQT: stat/zb1/RESULT = {**"Mem1":"0.000"**}
16:13:23.211 CMD: mem1=111111
16:13:23.216 MQT: stat/zb1/RESULT = {**"Mem1":"111111.000"**}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):   ZB-GW03-V1.2
- [x] Tasmota binary firmware version number used:  12.3.1(tasmota)
  - [x] Pre-compiled    2023-02-15T17:18:19
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:25:06.731 CMD: Status 0
17:25:06.736 MQT: stat/zb1/STATUS = {"Status":{"Module":0,"DeviceName":"ZB-GW03","FriendlyName":["Tasmota_ZB-GW03"],"Topic":"zb1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:25:06.772 MQT: stat/zb1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Software reset CPU","Uptime":"0T00:01:23","StartupUTC":"2023-12-16T22:23:43","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2023-05-08T05:58:52","SaveCount":51}}
17:25:06.799 MQT: stat/zb1/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2023-02-15T17:18:19","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":240,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"340/699"}}
17:25:06.815 MQT: stat/zb1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SL205G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000009","2805C80001000600003C5A00192800000000","000000A0","00004000","00004000","00000000"]}}
17:25:06.844 MQT: stat/zb1/STATUS4 = {"StatusMEM":{"ProgramSize":1302,"Free":1577,"Heap":156,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16400B","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8F800797","00500001","000000CF","010013C0","00000000","40024080","00200000","0400002C","00000080"],"Drivers":"1,2,3,4,8,9,10,12,20,21,23,38,41,50,52,62,82","Sensors":"1,2,3,5,6,127"}}
17:25:06.882 MQT: stat/zb1/STATUS5 = {"StatusNET":{"Hostname":"zb1-5428","IPAddress":"0.0.0.0","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"3C:E9:0E:89:35:34","Ethernet":{"Hostname":"zb1-5428-eth","IPAddress":"192.168.2.230","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"3C:E9:0E:89:35:37"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:25:06.929 MQT: stat/zb1/STATUS6 = {"StatusMQT":{"MqttHost":"vhost.loc","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_893534","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:25:06.945 MQT: stat/zb1/STATUS7 = {"StatusTIM":{"UTC":"2023-12-16T22:25:06","Local":"2023-12-16T17:25:06","StartDST":"2023-03-12T02:00:00","EndDST":"2023-11-05T02:00:00","Timezone":99,"Sunrise":"07:27","Sunset":"16:10"}}
17:25:06.975 MQT: stat/zb1/STATUS10 = {"StatusSNS":{"Time":"2023-12-16T17:25:06"}}
17:25:06.982 MQT: stat/zb1/STATUS11 = {"StatusSTS":{"Time":"2023-12-16T17:25:06","Uptime":"0T00:01:23","UptimeSec":83,"Heap":155,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

17:29:59.693 CMD: Mem1=string
17:29:59.698 MQT: stat/zb1/RESULT = {"Mem1":"0.000"}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Mem1=string
RESULT = {"Mem1":"string"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
[](https://tasmota.github.io/docs/Commands/#rules)
Mem<x> 	Manage up to 16 variables stored on flash (x = 1..16)
Mem returns all current values. Mem<x> returns the variable's current value.
<value> = store a **string value** in a variable
" = clear stored value in Mem<x>
**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-12-16:
Please check the syntax for command [`mem<x>`](https://tasmota.github.io/docs/Commands/#mem)

afgcons, 2023-12-16:
that was easy to fix ... thx!

closed 2023-12-16 by afgcons