arendst/Tasmota #20261: Switch not responding properly

andreasloe, 2023-12-18:
### PROBLEM DESCRIPTION
I am using a Wemos for a garage opener. Three "tools" are attached: 
- a reed sensor (detecting whether the garage is open, implemented as SWITCH), 
- a relay (simulating the push of a button at the opener, implemented as RELAY) and 
- a temperature sensor (implemented AM2301). 

The reed sensor does not display proper values (whether the garage is open or closed) in Tasmota.

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Lolin Wemos D1 
- [X] Tasmota binary firmware version number used: 13.3.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: https://tasmota.github.io/install/
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
stat/tasmota_8E6455/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"1216":"AM2301"},"GPIO5":{"161":"Switch2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
stat/tasmota_8E6455/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
stat/tasmota_8E6455/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
stat/tasmota_8E6455/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}```
- [X] Provide the output of this command: `Status 0`:
```lua
stat/tasmota_8E6455/STATUS = {"Status":{"Module":18,"DeviceName":"Garage","FriendlyName":["Garage"],"Topic":"tasmota_8E6455","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[13,15,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
stat/tasmota_8E6455/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"0T03:00:42","StartupUTC":"2023-12-18T08:35:57","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2023-12-17T20:48:50","SaveCount":55,"SaveAddress":"F5000"}}
stat/tasmota_8E6455/STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota)","BuildDateTime":"2023.12.12 14:33:12","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"373/699"}}
stat/tasmota_8E6455/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WLAN-120954",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0400800A","4606C80001000600003C5A0A192800000000","000000C0","00006000","00004000","00000000"]}}
stat/tasmota_8E6455/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["0407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
stat/tasmota_8E6455/STATUS5 = {"StatusNET":{"Hostname":"tasmota-8E6455-1109","IPAddress":"192.168.2.151","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:8E:64:55","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
stat/tasmota_8E6455/STATUS6 = {"StatusMQT":{"MqttHost":"raspfhem.fritz.box","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8E6455","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
stat/tasmota_8E6455/STATUS7 = {"StatusTIM":{"UTC":"2023-12-18T11:36:39","Local":"2023-12-18T12:36:39","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:53"}}
stat/tasmota_8E6455/STATUS10 = {"StatusSNS":{"Time":"2023-12-18T12:36:39","Tor":"ON","AM2301":{"Temperature":13.1,"Humidity":92.9,"DewPoint":12.0},"TempUnit":"C"}}
stat/tasmota_8E6455/STATUS11 = {"StatusSTS":{"Time":"2023-12-18T12:36:39","Uptime":"0T03:00:42","UptimeSec":10842,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"WLAN-120954","BSSId":"1E:E8:29:AA:44:A0","Channel":11,"Mode":"11n","RSSI":22,"Signal":-89,"LinkCount":1,"Downtime":"0T00:00:12"}}}```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
12:31:31.233 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:31","Tor":"OFF","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:31.282 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:31","Tor":"ON","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:32.513 DHT: Pin4 cycles (80/80)  74 29 74 36 74 35 74 35 74 35 75 35 74 102 74 100 93 102 74 36 ..
12:31:32.515 DHT: Pin4 read 03A2008328
12:31:33.715 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:33","Tor":"OFF","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:33.761 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:33","Tor":"ON","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:34.257 WIF: Prüfe Verbindung…
12:31:36.523 DHT: Pin4 cycles (80/80)  75 29 74 35 74 35 74 35 75 35 74 35 74 102 74 100 94 101 75 35 ..
12:31:36.525 DHT: Pin4 read 03A2008328
12:31:37.227 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:37","Tor":"OFF","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:37.324 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:37","Tor":"ON","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:37.575 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:37","Tor":"OFF","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:37.624 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:37","Tor":"ON","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:40.004 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:39","Tor":"OFF","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:40.050 MQT: tele/tasmota_8E6455/SENSOR = {"Time":"2023-12-18T12:31:40","Tor":"ON","AM2301":{"Temperature":13.1,"Humidity":93.0,"DewPoint":12.0},"TempUnit":"C"}
12:31:40.504 DHT: Pin4 cycles (80/80)  75 29 74 35 74 35 74 35 75 35 74 35 74 102 74 101 93 102 74 35 ..
12:31:40.507 DHT: Pin4 read 03A2008328```

### TO REPRODUCE
Enable Switch on D1 (as Switch2), attach reed contact and wait for door to close or open (a multimeter shows different resistance)

### EXPECTED BEHAVIOUR
If reed contact is closed the Switch should be ON resp. OFF when open.

Noschvie, 2023-12-18:
Why is your switch "Switch2" and not "Switch 1" ?

andreasloe, 2023-12-18:
Thanks for your help, Noschvie. 

There are several ways to open the garage (with 433MHz as well). Therefore, RELAY1 pushes the button (another way to open, different from 433MHz) and SWITCH2 shows whether the door is open or not. 

I need a TOGGLE for the relay and a reed contact for controlling the door.