arendst/Tasmota #20266: Alexa Fails to Detect Tasmota with Belkin WeMo Emulation; Hue Bridge Successful

RobertoD91, 2023-12-18:
### PROBLEM DESCRIPTION
When using 'Belkin WeMo' emulation, Alexa is unable to detect Tasmota. However, the 'Hue Bridge' emulation works fine.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff MINIR2
- [x] Tasmota binary firmware version number used: 13.3.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmota web installer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
17:34:47.674 MQT: stat/tasmota_01XXXX/RESULT = {"NAME":"Sonoff MINIR2","GPIO":[32,0,0,0,160,0,0,0,224,544,0,0,0,0],"FLAG":0,"BASE":1}
17:34:47.885 MQT: stat/tasmota_01XXXX/RESULT = {"Module":{"0":"Sonoff MINIR2"}}
17:34:48.091 MQT: stat/tasmota_01XXXX/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"544":"LedLink"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
17:35:22.226 MQT: stat/tasmota_01XXXX/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
17:35:22.437 MQT: stat/tasmota_01XXXX/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
17:35:22.638 MQT: stat/tasmota_01XXXX/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
17:36:23.712 MQT: stat/tasmota_01XXXX/STATUS = {"Status":{"Module":0,"DeviceName":"Luce DELETED_INFO","FriendlyName":["Luce DELETED_INFO"],"Topic":"tasmota_01XXXX","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:36:23.718 MQT: stat/tasmota_01XXXX/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-lite.bin.gz","RestartReason":"Software/System restart","Uptime":"0T16:45:04","StartupUTC":"2023-12-17T23:51:19","Sleep":50,"CfgHolder":4617,"BootCount":85,"BCResetTime":"2022-02-26T22:17:58","SaveCount":1618,"SaveAddress":"F5000"}}
17:36:23.722 MQT: stat/tasmota_01XXXX/STATUS2 = {"StatusFWR":{"Version":"13.3.0(lite)","BuildDateTime":"2023-12-12T14:28:57","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"396/699"}}
17:36:23.727 MQT: stat/tasmota_01XXXX/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DELETED_INFO",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
17:36:23.737 MQT: stat/tasmota_01XXXX/STATUS4 = {"StatusMEM":{"ProgramSize":495,"Free":508,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","0F000782","04108001","00000002","00000000","00000000","00000000","00000000","04000000","00000080","00000000"],"Drivers":"1,2,4,9,10,12,16,20,21,62","Sensors":""}}
17:36:23.745 MQT: stat/tasmota_01XXXX/STATUS5 = {"StatusNET":{"Hostname":"Luce-DELETED_INFO","IPAddress":"192.DELETED_INFO","Gateway":"192.DELETED_INFO","Subnetmask":"255.255.255.0","DNSServer1":"192.DELETED_INFO","DNSServer2":"0.0.0.0","Mac":"C4:DD:57:01:XX:XX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:36:23.750 MQT: stat/tasmota_01XXXX/STATUS6 = {"StatusMQT":{"MqttHost":"192.DELETED_INFO","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_01XXXX","MqttUser":"mqtt_admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:36:23.756 MQT: stat/tasmota_01XXXX/STATUS7 = {"StatusTIM":{"UTC":"2023-12-18T16:36:23","Local":"2023-12-18T17:36:23","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:53"}}
17:36:23.760 MQT: stat/tasmota_01XXXX/STATUS10 = {"StatusSNS":{"Time":"2023-12-18T17:36:23","Switch1":"ON"}}
17:36:23.767 MQT: stat/tasmota_01XXXX/STATUS11 = {"StatusSTS":{"Time":"2023-12-18T17:36:23","Uptime":"0T16:45:04","UptimeSec":60304,"Vcc":3.571,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HelloWorld2","BSSId":"38:10:D5:01:19:06","Channel":11,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- flash tasmota and erase memory
- setup wifi and model
- enable Belkin WeMo emulation
- search new device in alexa app -> don't work
- enable Hue Bridge emulation
- search new device in alexa app -> it work

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
- i have a physical alexa

**(Please, remember to close the issue when the problem has been addressed)**

RobertoD91, 2023-12-18:
Another Sonoff MINIR2 with Tasmota 12.2.0 work fine with Belkin WeMo emulation

s-hadinger, 2023-12-18:
I've never seen such problem. Please enable `weblog 4` and paste the logs when you do the scanning from Alexa

closed 2024-01-18 by github-actions[bot]