arendst/Tasmota #20285: EthDNSserver1 Command does not work as expected

UBWH, 2023-12-22:
### PROBLEM DESCRIPTION
Setting DNSServer1 on Ethernet Interface does not work

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [LilyGo ESP32 Ethernet Development Board (T-Internet-POE)](https://templates.blakadder.com/lilygo_T-Internet-POE.html) and [WT32-ETH01](https://templates.blakadder.com/wireless_tag_WT32-ETH01.html)
- [x] Tasmota binary firmware version number used: __13.3.0_
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ____Tasmota Web Installer _
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Backlog Template; Module; GPIO 255
07:25:23.364 RSL: RESULT = {"NAME":"LilyGO T-Internet-POE","GPIO":[0,1,672,1,1,0,0,0,1,6720,736,704,1,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
07:25:23.589 RSL: RESULT = {"Module":{"0":"LilyGO T-Internet-POE"}}
07:25:23.847 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
07:26:54.098 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"ON","Length":0,"Free":511,"Rules":""}}
07:26:54.317 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
07:26:54.567 RSL: RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:27:28.170 CMD: status 0
07:27:28.175 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_5BDBFC","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
07:27:28.210 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:03:45","StartupUTC":"2023-12-22T06:23:43","Sleep":50,"CfgHolder":4617,"BootCount":215,"BCResetTime":"2023-09-27T01:13:15","SaveCount":378}}
07:27:28.248 RSL: STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota32)","BuildDateTime":"2023-12-12T14:31:42","Core":"2_0_14","SDK":"4.4.6.231122","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"409/699"}}
07:27:28.265 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UniFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00400080","00006000","00004000","00000000"]}}
07:27:28.303 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1888,"Free":991,"Heap":137,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
07:27:28.369 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-5BDBFC-7164","IPAddress":"10.1.1.76","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.1.1.1","DNSServer2":"0.0.0.0","Mac":"44:17:93:5B:DB:FC","IP6Global":"fdfd:f5d8:e1fd::4617:93ff:fe5b:dbfc","IP6Local":"fe80::4617:93ff:fe5b:dbfc%st2","Ethernet":{"Hostname":"tasmota-5BDBFC-7164-eth","IPAddress":"10.1.1.79","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.1.1.1","DNSServer2":"0.0.0.0","Mac":"44:17:93:5B:DB:FF","IP6Global":"fdfd:f5d8:e1fd::4617:93ff:fe5b:dbff","IP6Local":"fe80::4617:93ff:fe5b:dbff%en1"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
07:27:28.428 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5BDBFC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:27:28.443 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-12-22T06:27:28","Local":"2023-12-22T07:27:28","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:55"}}
07:27:28.473 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-12-22T07:27:28"}}
07:27:28.480 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-12-22T07:27:28","Uptime":"0T00:03:45","UptimeSec":225,"Heap":136,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":47},"Wifi":{"AP":1,"SSId":"UniFi","BSSId":"DC:9F:DB:1C:E0:60","Channel":1,"Mode":"11n","RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
7:32:42.939 CMD: ethipaddress 192.168.22.100
07:32:42.942 SRC: WebConsole from 0.0.0.0
07:32:42.943 CMD: Grp 0, Cmd 'ETHIPADDRESS', Idx 1, Len 14, Pld 192, Data '192.168.22.100'
07:32:42.952 RSL: RESULT = {"EthIPAddress":"192.168.22.100"}
07:32:43.176 CFG: Saved, Count 380, Bytes 4096
07:32:44.156 ETH: IPv4 192.168.22.100, mask 255.255.255.0, gateway 10.1.1.1
07:33:26.056 CMD: EthSubnetmask 255.255.0.0
07:33:26.058 SRC: WebConsole from 0.0.0.0
07:33:26.060 CMD: Grp 0, Cmd 'ETHSUBNETMASK', Idx 1, Len 11, Pld 255, Data '255.255.0.0'
07:33:26.066 RSL: RESULT = {"EthSubnetmask":"255.255.0.0"}
07:33:27.084 CFG: Saved, Count 381, Bytes 4096
07:33:27.151 ETH: IPv4 192.168.22.100, mask 255.255.0.0, gateway 10.1.1.1
07:34:34.166 CMD: EthGateway 192.168.22.1
07:34:34.168 SRC: WebConsole from 0.0.0.0
07:34:34.170 CMD: Grp 0, Cmd 'ETHGATEWAY', Idx 1, Len 12, Pld 192, Data '192.168.22.1'
07:34:34.176 RSL: RESULT = {"EthGateway":"192.168.22.1"}
07:34:35.098 CFG: Saved, Count 382, Bytes 4096
07:34:35.161 ETH: IPv4 192.168.22.100, mask 255.255.0.0, gateway 192.168.22.1
[... 49 lines cut ...]
   "Subnetmask":"255.255.0.0",
   "DNSServer1":"10.1.1.1",                   <--------------  WRONG
   "DNSServer2":"8.8.9.9",
   "Mac":"44:17:93:5B:DB:FF",
    "IP6Global":"",
   "IP6Local":"fe80::4617:93ff:fe5b:dbff%en1"
  },
"Webserver":2,
"HTTP_API":1,
"WifiConfig":4,
"WifiPower":16.0
}}
```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
none

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
This seems related to this issue:
https://github.com/arendst/Tasmota/issues/14688
which is closed.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-12-22:
This is a limitation of the underlying `LwIP` library. You can have only 2 DNS servers that are shared among all interfaces, including IPv4/IPv6. Nothing we can do about it.

UBWH, 2023-12-22:
OK - that helps explain things.

Maybe that should be mentioned in the documentation.

Jason2866, 2023-12-22:
@UBWH The documentation is Open Source too. You can add ;-)

closed 2023-12-22 by Jason2866