arendst/Tasmota #20294: Consider moving this repo into the `tasmota` org

DeflateAwning, 2023-12-22:
It is really confusing having the docs repo at https://github.com/tasmota/docs (`tasmota/docs`), but the actual Tasmota repo not being in the `tasmota` org at all. 

Worse, the docs link to the https://github.com/arendst/tasmota repo in the top-right corner, but don't have a link to the docs repo.

Worse, some files (e.g., the `BUILDS.md`) file are duplicated in both repos.

sfromis, 2023-12-22:
Each page of `docs` does have a Github link to edit (or not) the page, the pencil top/right. Outside of editing, normal usage of docs does not include going to the Github repo.

DeflateAwning, 2023-12-22:
I see that now; I missed that edit button when I was getting started editing it. It's tough to see on a bad-contrast monitor/bad-contrast eyeball. My Ctrl+F for "edit" failed; it's very common for these sorts of docs to have an "Edit this page <pencil emoji>" at the bottom.

All that aside, the core point of this issue still stands: Consider moving this repo into the tasmota org. Tasmota is very popular, and it would be great if it was in the org created for its ecosystem.

Benefits:
1. Less confusion for (clueless) people like me
2. Prevents namespace security attacks, where someone registers a username similar to "arendst", like "arends" or "arandst", and publishes/shares a malicious version of Tasmota. 
3. Prevents confusion with users who are afraid that the "arendst/tasmota" version is a fake, when the "tasmota" org exists.
4. Maybe others.

barbudor, 2023-12-23:
Tasmota belongs to Theo Arendst so the code is in his account and he has no intention to "drop" it. This is common with most similar projects such as aircookie/wled, 1technophile/openmqttgateway

The Tasmota account has been created by other members of the core team to host various side projects around Tasmota such as the docs, the docker builder, decode-status tools etc...

The BUILDS.md and I2CDEVICES are part of arendst/Tasmota repo and are copied to the Doc repo to avoid redirecting people to the repo

DeflateAwning, 2023-12-23:
https://github.com/tasmota is not an account; it is a GitHub Organization (which appears to be owned by @jziolkowski). If @arendst was added as an owner to the Tasmota GitHub Organization (which I'm shocked he isn't already), I feel like that could work out well? 

> The BUILDS.md and I2CDEVICES are part of arendst/Tasmota repo and are copied to the Doc repo to avoid redirecting people to the repo

Is there a reason that we're trying to "avoid redirecting people to the [docs] repo"? I was _considering_ opening a PR to update the "BUILDS.md" file in `arendst/Tasmota` to just say "refer to the file in <link to equivalent file in `tasmota/docs` repo>", but obviously won't do that if that won't get merged.

jziolkowski, 2023-12-23:
Wait, I am?

Also, you should ask @arendst what he thinks of that. Tasmota GitHub organization is for that: all the meta and docs. The main project is owned solely by Theo under his account. Not my call.

barbudor, 2023-12-23:
> Is there a reason that we're trying to "avoid redirecting people to the [docs] repo"?

You got me wrong
I meant avoid a doc page to link to arendst/Tasmota/BUILDS.md
It's a better experience for the reader to keep browsing this information while in the docs.
Thus the copy

DeflateAwning, 2023-12-23:
I may be mistaken about the "tasmota" org ownership, but @jziolkowski is the only person who shows up on [the people page of the "tasmota" organization](https://github.com/orgs/tasmota/people).

I agree! I would love to see what @arendst thinks! I just wanted to provide a suggestion from the community about a way to improve.

> ... This is common with most similar projects such as aircookie/wled, 1technophile/openmqttgateway

Re: [aircookie/WLED](https://github.com/Aircoookie/WLED) and [1technophile/OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway), the [WLED](https://github.com/Aircoookie/WLED) and [WLED-Docs](https://github.com/Aircoookie/WLED-Docs) repos are both pinned under Aircookie's account, which makes it easier and clearer to find all these linked repos.  The OpenMQTTGateway repo tracks [the docs in the same repo as the source code for the project](https://github.com/1technophile/OpenMQTTGateway/tree/development/docs), which is a distinctly different design choice. Those are both examples of non-Organization-organized popular open source projects though, you're right.

To provide a few counter-examples (open source projects which keep everything in a GitHub organization), the [OpenSCAD project](https://github.com/openscad/openscad), made by Clifford Wolf and contributed to by hundreds of others, uses [a single unified GitHub Organization](https://github.com/openscad) to track their several repos (including main source, docker, documentation, etc.). A few other open source projects (with rights still owned by their creator, but which track their source in a GitHub org) incude:
* https://github.com/TigerVNC/tigervnc
* https://github.com/chirpstack/chirpstack
* https://github.com/syncthing/docs
* https://github.com/OSGeo/gdal
* https://github.com/thingsboard/thingsboard

In my opinion, Tasmota has reached the notoriety threshold and depth of content to have its own organization (e.g., several repos specifically for Tasmota, 20k+ stars on GitHub in its main repo); WLED (13k stars) and OpenMQTTGateway (3.3k stars) are also moving toward that point, but definitely aren't quite as wide-spread as Tasmota.

Jason2866, 2023-12-23:
As a maintainer of all Github Actions a clear answer: **NO** Since it is to much effort to change all. Don't have the time and motivation to do.

closed 2023-12-23 by Jason2866

DeflateAwning, 2023-12-23:
I'd be willing to help out with that transition! Would love to hear from @arendst before you just kill it the idea (and somewhat rudely at that).