arendst/Tasmota #20299: MQTT switch instructions on Domoticz/out subtopics are not recognized after 11.1.0

colbrac, 2023-12-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
MQTT switch instructions on Domoticz/out subtopics are not recognized after 11.1.0. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff S26
- [X] Tasmota binary firmware version number used: 9.2.0, 10.1.0, 11.1.0, 12.2.0, 12.5.0, 13.0.0, 13.1.0, 13.3.0.1 (dev of today)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:15:20.359 MQT: stat/tas_tv/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:15:20.568 MQT: stat/tas_tv/RESULT = {"Module":{"8":"Sonoff S2X"}}
12:15:20.774 MQT: stat/tas_tv/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:15:46.458 MQT: stat/tas_tv/STATUS = {"Status":{"Module":8,"DeviceName":"tas_tv","FriendlyName":["tas_tv"],"Topic":"tas_tv","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:15:46.494 MQT: stat/tas_tv/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:12","StartupUTC":"2023-12-23T11:12:34","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2023-08-12T22:27:28","SaveCount":59,"SaveAddress":"FA000"}}
12:15:46.525 MQT: stat/tas_tv/STATUS2 = {"StatusFWR":{"Version":"13.3.0.1(tasmota)","BuildDateTime":"2023-12-23T09:56:24","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"349/699"}}
12:15:46.546 MQT: stat/tas_tv/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["*growon*",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
12:15:46.579 MQT: stat/tas_tv/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
12:15:46.619 MQT: stat/tas_tv/STATUS5 = {"StatusNET":{"Hostname":"tas-tv-7037","IPAddress":"192.168.178.23","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.7","DNSServer2":"192.168.178.1","Mac":"48:55:19:7A:7B:7D","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:15:46.643 MQT: stat/tas_tv/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.7","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A7B7D","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:15:46.666 MQT: stat/tas_tv/STATUS7 = {"StatusTIM":{"UTC":"2023-12-23T11:15:46Z","Local":"2023-12-23T12:15:46","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:56"}}
12:15:46.685 MQT: stat/tas_tv/STATUS10 = {"StatusSNS":{"Time":"2023-12-23T12:15:46"}}
12:15:46.692 MQT: stat/tas_tv/STATUS11 = {"StatusSTS":{"Time":"2023-12-23T12:15:46","Uptime":"0T00:03:12","UptimeSec":192,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"*growon*","BSSId":"D8:0D:17:C9:67:4A","Channel":5,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use Domoticz and dummy switch, use MQTT with floor/room setting for domoticz/out subgroups  and Tasmota < 12.2.0 -> Sonoff switches succesfully.
Use same setup and only change Tasmota firmware to >11.1.0 -> Sonoff does not respond anymore.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota reacting to switching instructions in domoticz/out subgroups

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
As discussed here: https://github.com/arendst/Tasmota/discussions/19107, I note that significant changes have been made between 11.1.0 and 12.0.0 in [tasmota_xdrv_driver/xdrv_07_domoticz.ino](https://github.com/arendst/Tasmota/compare/d157b1c..7bc7be5?diff=split&w=#diff-3f7d8ab0ef2d415a7e37b27248546a94852f24143a14a265c79b29702b04f207). I assume a little error crept in.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-12-23:
Domoticz tends to flood MQTT with all kinds of messages. Most of them are irrelevant for a specific device.

In versions before v12.0.1 this lead to a lot of processing before finally determine that the message was not for this specific device.

After v11.1.0 an additional detection was introduced to allow early bail-out of irrelevant Domoticz calls. This detection only allows for a topic as defined in DOMOTICZ_OUT_TOPIC which is default ``domoticz/out``.

The additional code is here:
```
  int32_t relay_index = -1;

  // Quick check if this is mine using topic domoticz/out/{$idx}
  if (strlen(XdrvMailbox.topic) > strlen(DOMOTICZ_OUT_TOPIC)) {
    char* topic_index = &XdrvMailbox.topic[strlen(DOMOTICZ_OUT_TOPIC) +1];
    relay_index = DomoticzIdx2Relay(atoi(topic_index));
    if (relay_index < 0) {
      return true;  // Idx not mine
    }
  }
```

This code does not chk for sub-topics like floor and/or room.

As a workaround you could compile your own and change #define DOMOTICZ_OUT_TOPIC to any topic you like including your needed floor and room.

As a solution I'll see if I can change the current fixed location for relay index to finding the index at the end of the topic. Hold on......

arendst, 2023-12-23:
As an alternative you could try to configure Domoticz for sending with Index like I do:

[image]

Looking into it further I think you will need to reconfigure your Domoticz MQTT Publish Topic anyway in order for Tasmota to bail-out quickly using the test for {$idx} as I use.

arendst, 2023-12-23:
No fix planned. Reconfigure Domoticz for option ``Publish Topic: Index``

colbrac, 2023-12-23:
Thanks for the fast response and the suggestions. 

So I understand that: 
- Domoticz MQTT with option `Publish topic: Flat` will work but results in a lot more processing in all Tasmota devices. 
- Domoticz MQTT with option `Publish topic: Index` will work and more quickly stops processing of irrelevant messages in all Tasmota devices.

My reason for using the option `Publish topic: Floor/Room` is in fact to limit outgoing MQTT messages to only those devices that I have manually added to a floorplan. In this way, I minimize the amount of messages going through domoticz/out. 
 
I will stick to Tasmota version 11.1.0 as the easiest solution for me. 

NB: [the docs](https://tasmota.github.io/docs/Domoticz/) instruct to use option `Publish topic: Flat`. Perhaps good to update the docs to recommend option `Publish topic: Index` and note that option `Publish topic: Floor/Room` will not work.

arendst, 2023-12-23:
I'll do some further tests. I think there is a solution where if there is no idx in the topic I'll continue to look for the idx in  the message.

Hold on (once more)....

arendst, 2023-12-23:
Oh my. I never knew the floor/room option does lower the amount of MQTT messages so much:

```
15:28:56.551-025 DOM: Topic 'domoticz/out/512', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:28:57","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 512,"name" : "Actual Delivery (L1 + L2 + L3)","nvalue" : 0,"stype" : "Electric","svalue1" : "0.0","unit" : 8}'
15:28:56.603-025 DOM: Topic 'domoticz/out/514', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:28:57","RSSI" : 12,"description" : "","dtype" : "General","hwid" : "11","id" : "00000001","idx" : 514,"name" : "kWh Usage L1 (Calculated)","nvalue" : 0,"stype" : "kWh","svalue1" : "183.000","svalue2" : "394887.219","unit" : 1}'
15:28:56.652-026 DOM: Topic 'domoticz/out/515', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:28:57","RSSI" : 12,"description" : "","dtype" : "General","hwid" : "11","id" : "00000002","idx" : 515,"name" : "kWh Usage L2 (Calculated)","nvalue" : 0,"stype" : "kWh","svalue1" : "103.000","svalue2" : "280159.656","unit" : 1}'
15:28:56.701-027 DOM: Topic 'domoticz/out/516', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:28:57","RSSI" : 12,"description" : "","dtype" : "General","hwid" : "11","id" : "00000003","idx" : 516,"name" : "kWh Usage L3 (Calculated)","nvalue" : 0,"stype" : "kWh","svalue1" : "249.000","svalue2" : "497657.688","unit" : 1}'
15:29:00.198-027 DOM: Topic 'domoticz/out/526', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:01","RSSI" : 12,"description" : "return temperature","dtype" : "Temp","hwid" : "16","id" : "2710","idx" : 526,"name" : "EMS - CV retoursensor","nvalue" : 1,"stype" : "LaCrosse TX3","svalue1" : "34.1","unit" : 15}'
15:29:00.397-026 DOM: Topic 'domoticz/out/528', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:01","RSSI" : 12,"description" : "","dtype" : "General","hwid" : "16","id" : "0010001D","idx" : 528,"name" : "EMS bus gateway - Boiler flame current","nvalue" : 1,"stype" : "Current","svalue1" : "11.2","unit" : 29}'
15:29:00.449-027 DOM: Topic 'domoticz/out/551', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:01","LevelActions" : "||","LevelNames" : "Hot|Eco|Intelligent","LevelOffHidden" : "true","RSSI" : 12,"SelectorStyle" : "0","description" : "","dtype" : "Light/Switch","hwid" : "16","id" : "0010001E","idx" : 551,"name" : "EMS - Comfort mode","nvalue" : 1,"stype" : "Selector Switch","svalue1" : "0","switchType" : "Selector","unit" : 30}'
15:29:06.103-027 DOM: Topic 'domoticz/out/289', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "P1 Smart Meter","hwid" : "11","id" : "0001","idx" : 289,"name" : "sP1meter-Energy","nvalue" : 0,"stype" : "Energy","svalue1" : "12902808","svalue2" : "14575229","svalue3" : "0","svalue4" : "0","svalue5" : "539","svalue6" : "0","unit" : 1}'
15:29:06.202-025 DOM: Topic 'domoticz/out/290', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 290,"name" : "sP1verbruik-L1","nvalue" : 0,"stype" : "Electric","svalue1" : "184.0","unit" : 1}'
15:29:06.252-023 DOM: Topic 'domoticz/out/293', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 293,"name" : "sP1geleverd-L1","nvalue" : 0,"stype" : "Electric","svalue1" : "0.0","unit" : 4}'
15:29:06.302-023 DOM: Topic 'domoticz/out/291', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 291,"name" : "sP1verbruik-L2","nvalue" : 0,"stype" : "Electric","svalue1" : "103.0","unit" : 2}'
15:29:06.352-023 DOM: Topic 'domoticz/out/294', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 294,"name" : "sP1geleverd-L2","nvalue" : 0,"stype" : "Electric","svalue1" : "0.0","unit" : 5}'
15:29:06.402-023 DOM: Topic 'domoticz/out/292', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 292,"name" : "sP1verbruik-L3","nvalue" : 0,"stype" : "Electric","svalue1" : "250.0","unit" : 3}'
15:29:06.453-025 DOM: Topic 'domoticz/out/295', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 295,"name" : "sP1geleverd-L3","nvalue" : 0,"stype" : "Electric","svalue1" : "0.0","unit" : 6}'
15:29:06.502-025 DOM: Topic 'domoticz/out/511', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 511,"name" : "Actual Usage (L1 + L2 + L3)","nvalue" : 0,"stype" : "Electric","svalue1" : "537.0","unit" : 7}'
15:29:06.554-025 DOM: Topic 'domoticz/out/512', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "Usage","hwid" : "11","id" : "0000000","idx" : 512,"name" : "Actual Delivery (L1 + L2 + L3)","nvalue" : 0,"stype" : "Electric","svalue1" : "0.0","unit" : 8}'
15:29:06.603-025 DOM: Topic 'domoticz/out/514', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:07","RSSI" : 12,"description" : "","dtype" : "General","hwid" : "11","id" : "00000001","idx" : 514,"name" : "kWh Usage L1 (Calculated)","nvalue" : 0,"stype" : "kWh","svalue1 [... cut ...] /Off","unit" : 3}'
15:29:38.781-027 DOM: Topic 'domoticz/out/Floor1/Werkkamer', Data '{"Battery" : 255,"LastUpdate" : "2023-12-23 15:29:39","RSSI" : 12,"description" : "","dtype" : "Lighting 2","hwid" : "3","id" : "3030303","idx" : 403,"name" : "TheoTest","nvalue" : 1,"stype" : "AC","svalue1" : "15","switchType" : "On/Off","unit" : 3}'
15:29:44.372-027 WIF: Sending Gratuitous ARP
15:29:46.794-027 WIF: Checking connection...
```

I also see that this doesn't trigger my device ;-(
I'll fix it now.

colbrac, 2023-12-23:
Confirmed the fix works when I update the Sonoff S26 to the development release of 5pm.

Is an update of the docs included in the commit, recommending idx or room over flat for the Domoticz mqtt publish option?

closed 2023-12-23 by colbrac

arendst, 2023-12-23:
> Is an update of the docs included in the commit, recommending idx or room over flat for the Domoticz mqtt publish option?

Yep. Chk it out.