arendst/Tasmota #20301: Passwords with special characters are not set correctly via GUI

DerT94, 2023-12-23:
### PROBLEM DESCRIPTION
Passwords with special characters are not set correctly via GUI.
I tested it with a few and the error was reproducible.
Special characters with which the error occurred for me

- ;
- {
- ~

If I set my password via the console with the command `mqttpassword` it works.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Pro 3
- [ ] Tasmota binary firmware version number used: tasmota32.bin 13.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
19:25:40.739 MQT: stat/ShellyPro3_01/RESULT = {"NAME":"ShellyPro3","GPIO":[1,1,224,1,225,1,1,1,1,1,1,1,1,1,5600,1,0,1,1,5568,0,1,1,1,0,0,0,0,226,1,96,1,4736,97,162,1],"FLAG":0,"BASE":1}
19:25:40.966 MQT: stat/ShellyPro3_01/RESULT = {"Module":{"0":"ShellyPro3"}}
19:25:41.221 MQT: stat/ShellyPro3_01/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"224":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"225":"Relay2"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"226":"Relay3"},"GPIO33":{"0":"None"},"GPIO34":{"96":"Button_i1"},"GPIO35":{"0":"None"},"GPIO36":{"4736":"ADC Temp1"},"GPIO37":{"97":"Button_i2"},"GPIO38":{"162":"Switch3"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
19:26:31.767 MQT: stat/ShellyPro3_01/STATUS = {"Status":{"Module":0,"DeviceName":"ShellyPro3_01","FriendlyName":["ShellyPro3_01","",""],"Topic":"ShellyPro3_01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,1,6,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:26:31.826 MQT: stat/ShellyPro3_01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:06:55","StartupUTC":"2023-12-23T18:19:36","Sleep":50,"CfgHolder":4617,"BootCount":62,"BCResetTime":"2023-12-17T22:31:22","SaveCount":419}}
19:26:31.882 MQT: stat/ShellyPro3_01/STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota32)","BuildDateTime":"2023-12-12T14:31:42","Core":"2_0_14","SDK":"4.4.6.231122","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"551/699"}}
19:26:31.918 MQT: stat/ShellyPro3_01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI_TAJP_IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","0505C80001000600003C5A0A192800000000","00800080","00006000","00004000","00000000"]}}
19:26:31.987 MQT: stat/ShellyPro3_01/STATUS4 = {"StatusMEM":{"ProgramSize":1888,"Free":991,"Heap":110,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":8192,"FlashSize":8192,"FlashChipId":"17701C","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
19:26:32.083 MQT: stat/ShellyPro3_01/STATUS5 = {"StatusNET":{"Hostname":"ShellyPro3-01-5992","IPAddress":"X","Gateway":"X","Subnetmask":"X","DNSServer1":"X","DNSServer2":"0.0.0.0","Mac":"C8:F0:9E:88:57:68","IP6Global":"","IP6Local":"fe80::caf0:9eff:fe88:5768%st2","Ethernet":{"Hostname":"ShellyPro3-01-5992-eth","IPAddress":"0.0.0.0","Gateway":"192.168.170.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.0.3","DNSServer2":"0.0.0.0","Mac":"C8:F0:9E:88:57:6B","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
19:26:32.160 MQT: stat/ShellyPro3_01/STATUS6 = {"StatusMQT":{"MqttHost":"anonym","MqttPort":1883,"MqttClientMask":"ShellyPro3_01","MqttClient":"ShellyPro3_01","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:26:32.205 MQT: stat/ShellyPro3_01/STATUS7 = {"StatusTIM":{"UTC":"2023-12-23T18:26:32","Local":"2023-12-23T19:26:32","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:56"}}
19:26:32.244 MQT: stat/ShellyPro3_01/STATUS10 = {"StatusSNS":{"Time":"2023-12-23T19:26:32","Switch3":"ON","ANALOG":{"Temperature1":33.9},"TempUnit":"C"}}
19:26:32.283 MQT: stat/ShellyPro3_01/STATUS11 = {"StatusSTS":{"Time":"2023-12-23T19:26:32","Uptime":"0T00:06:56","UptimeSec":416,"Heap":104,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"Berry":{"HeapUsed":22,"Objects":300},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"WIFI_TAJP_IoT","BSSId":"62:22:32:13:C0:A1","Channel":1,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-12-23:
The docs says that you cannot use special characters in passwords. Thus no bug.

DerT94, 2023-12-24:
> The docs says that you cannot use special characters in passwords. Thus no bug.

Then I'll close the issue 
It is interesting that I never noticed this problem with older Tasmota versions. (<10)

closed 2023-12-24 by DerT94

s-hadinger, 2024-09-01:
If it fails with special characters, it's not on purpose. You are welcome to spot where the problem is and submit a Pull Request

sfromis, 2024-09-01:
The risk factor with special chars is not so much support or not, but when multiple tools are used with different ideas about special meaning of certain chars, or handling anything outside of 7-bit US-ASCII, small deviations in possibly multiple steps can easily lead to the result being broken, which is pretty bad for an "invisible" password. 

Anyway, it is not true that special chars are needed for password security.

s-hadinger, 2024-09-01:
Il afraid it's not about posting a feature request, but about submitting a code patch. And finding time to do so

s-hadinger, 2024-09-02:
If you come up with a small patch that does not increase the flash size and does not break anything, there’s a high chance it is merged. 

However we don’t even know if the problem is in Tasmota, or Arduino Core or esp-idf. 

It also depends on what type of special characters you mean. ASCII 7 bits should be acceptable. But if you aim at accepting Unicode, then you’re opening a huge can of worms.

barbudor, 2024-09-02:
There's no issues to set spécial characters from the console or mqtt
So I bet the issue is around url-encoding (or lack of) on the webserver