arendst/Tasmota #20303: Tasmota crashes on long devicename

thomas374b, 2023-12-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8285
- [ ] Tasmota binary firmware version number used: 13.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- in a spare window open a serial console to the device to observe
- navigate with your browser to the device webpage
- go to  Settings -> Misc Settings 
- give an reasonable long devicename and friendlyname to trigger the fault
- click on "save"

### ACTUAL BEHAVIOUR
The device reboots immediately. On serial console a crash dump is written. As long as the browser tries to reload the page, the device is stuck in a crash-reboot loop. This can be stopped only by redirecting the browser elsewhere or switching off the WLAN router.

### EXPECTED BEHAVIOUR
Tasmota should not crash at all.

### ADDITIONAL CONTEXT
The same behaviour can be triggered when giving the resulting Backlog command in the serial console. In my case the devicename and friendlyname had only 5 characters.  The crash was triggered when the payload data length for the backlog command was larger than 178 chars.  

The Info-Page showed that there were about 11kB free RAM space on my device.

I tried already to debug this issue but I was lost at function AddLogData() in file tasmota/tasmota_support/support.ino around line 2513

**(Please, remember to close the issue when the problem has been addressed)**

thomas374b, 2023-12-24:
Additional infos attached

arendst, 2023-12-24:
Let's start by you filling in the minimal requested information before cherrypicking the stacktrace.

sfromis, 2023-12-24:
> The Info-Page showed that there were about 11kB free RAM space on my device.

That is bad. You should not go that low.

barbudor, 2023-12-24:
Can't reproduce based on zero information provided
```
11:19:11.335 CMD: backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
11:19:11.410 MQT: stat/nodemcu/WEBPASSWORD = {"WebPassword":"****"}
11:19:11.616 RSL: SO = {"SetOption3":"OFF"}
11:19:11.866 RSL: SO = {"SetOption128":"ON"}
11:19:12.117 RSL: DEVICENAME = {"DeviceName":"CloudSens"}
11:19:12.367 RSL: FN = {"FriendlyName1":"CloudSens"}
11:19:12.617 RSL: FN = {"FriendlyName2":""}
11:19:12.868 RSL: FN = {"FriendlyName3":""}
11:19:13.118 RSL: FN = {"FriendlyName4":""}
11:19:13.367 RSL: FN = {"FriendlyName5":""}
11:19:13.617 RSL: FN = {"FriendlyName6":""}
11:19:13.929 RSL: FN = {"FriendlyName7":""}
11:19:14.138 RSL: FN = {"FriendlyName8":""}
11:19:14.341 RSL: TEMPLATE = {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
11:19:15.848 APP: Restarting
```

Even with much longer device name and friendly name
```
11:22:46.972 CMD: backlog WebPassword2;SO3 0;SO128 1;DeviceName Some very unreasonnable long device name without meaning;FN1 Another very long and meaning less friendly name;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
11:22:47.071 RSL: WEBPASSWORD = {"WebPassword":"****"}
11:22:47.277 RSL: SO = {"SetOption3":"OFF"}
11:22:47.527 RSL: SO = {"SetOption128":"ON"}
11:22:47.778 RSL: DEVICENAME = {"DeviceName":"Some very unreasonnable long device name without meaning"}
11:22:48.028 RSL: FN = {"FriendlyName1":"Another very long and meaning less friendly name"}
11:22:48.277 RSL: FN = {"FriendlyName2":""}
11:22:48.527 RSL: FN = {"FriendlyName3":""}
11:22:48.728 RSL: FN = {"FriendlyName4":""}
11:22:48.930 RSL: FN = {"FriendlyName5":""}
11:22:49.180 RSL: FN = {"FriendlyName6":""}
11:22:49.432 RSL: FN = {"FriendlyName7":""}
11:22:49.660 RSL: FN = {"FriendlyName8":""}
11:22:49.863 RSL: TEMPLATE = {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
11:22:51.271 APP: Restarting
```

Doing the same from the UI with even longer DN and FN, still no problems

arendst, 2023-12-24:
Enable ``SO130 1`` to see heap during backlog execution

thomas374b, 2023-12-25:
Setting the Option SO130 1 did not reveal more information. The crash appears right after logging the backlog command itself.
-------8<--------------------------------------
22:15:04.766 WIF: PrÃŒfe Verbindung...
backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
got command "backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
"
22:15:11.244 CMD: backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
22:15:11.254 SRC: Serial
22:15:11.257 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 187, Pld -99, Data 'WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}'

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x22455341 epc2=0x00000000 epc3=0x40105682 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb80 end: 3fffffc0 offset: 0190
3ffffd10:  4c46222c 3a224741 42222c30 22455341  
3ffffd20:  7d38313a 3fff0029 00000008 0000000b  
-------8<--------------------------------------

The problem occured on a firmware image where I included every functionality/drivers that I might have on all my (future) devices/projects. (I wanted to avoid to have too many special firmware images).

Now I removed all unnessesary stuff and re-compiled a version with much fewer functionality.  On that minimal firmware image (about 590kB, with 16.9kB free RAM) the error vanishes.

No I'm going to re-enable all that features step-by-step, re-compile and test in between.  I hope I can identify a USE_XXX compiler  directive that triggers the fault

thomas374b, 2023-12-25:
The error re-appears with definition
   #define	USE_SCRIPT_SUB_COMMAND
The firmare image is about 604kB large and there where 17.2k RAM free. I have no subcommand section defined in my script at all.
(and of course USE_SCRIPT was also defined, not using rules)

Then I tried re-enabling all my "goodies" except for USE_SCRIPT_SUB_COMMAND resulting in an 790k firmware image with 12.9k free RAM left. And the error stays away.

sfromis, 2023-12-25:
Looks like you did not have SetOption130 enabled, otherwise your console timestamps would be like `23:17:39.852-024` - with the extra number being free heap at that time. And when you post about numbers down to 11kB, you are too low.

Note that the numbers you post are most probably before extra RAM is being used by the things you activate, like in scripting. That's why it is still **very** interesting with `SetOption130` actually activated. And why 11KB in idle is very low.

thomas374b, 2023-12-25:
I retested in between with Script disabled  => the error vanishes

The error re-appears with option USE_SCRIPT_SUB_COMMAND compiled in _and_  script enabled.  Even with a dummy script stating only   >D

thomas374b, 2023-12-25:
--------------8<--------------------------------------------
23:32:12.019-013 CMD: backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
23:32:12.031-013 SRC: Serial
23:32:12.034-013 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 187, Pld -99, Data 'WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}'
23:32:12.056-013 >> BACKLOG, WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}, -99, 1 
23:32:12.076-013 >>-1

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x22455344 epc2=0x00000000 epc3=0x40101510 excvaddr=0x00000000 depc=0x00000000

--------------8<--------------------------------------------

I guess -013 means 13kB free RAM. on the Web-Info Page it shows 12.9kB free RAM

thomas374b, 2023-12-25:
For me the feature
USE_SCRIPT_SUB_COMMAND
is the evil. I can live with it leaving it undefined. I have here another trace (without sub_command feature) everything went OK with the heap going down to 8k and still everything is working. 

-------8<---------------------------
23:50:44.540-009 WEB: Chunk size 0
backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
got command "backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
"
23:50:46.529-012 CMD: backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
23:50:46.540-012 SRC: Serial
23:50:46.544-012 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 187, Pld -99, Data 'WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}'
23:50:46.576-012 SRC: Backlog
23:50:46.579-012 CMD: Grp 0, Cmd 'WEBPASSWORD', Idx 2, Len 0, Pld -99, Data ''
23:50:46.583-012 RSL: RESULT = {"WebPassword":"****"}
23:50:46.828-012 SRC: Backlog
23:50:46.831-012 CMD: Grp 0, Cmd 'SO', Idx 3, Len 1, Pld 0, Data '0'
23:50:46.835-012 RSL: RESULT = {"SetOption3":"OFF"}
23:50:46.947-008 WEB: Chunk size 158
23:50:46.949-008 WEB: Chunk size 0
23:50:47.077-012 SRC: Backlog
23:50:47.081-012 CMD: Grp 0, Cmd 'SO', Idx 128, Len 1, Pld 1, Data '1'
23:50:47.084-012 RSL: RESULT = {"SetOption128":"ON"}
23:50:47.327-012 SRC: Backlog
23:50:47.331-012 CMD: Grp 0, Cmd 'DEVICENAME', Idx 1, Len 9, Pld -99, Data 'CloudSens'
23:50:47.334-012 CFG: CR 431/699, Busy 0
23:50:47.337-012 RSL: RESULT = {"DeviceName":"CloudSens"}
23:50:47.684-012 CFG: in Flash gespeichert an 1FB, Anzahl 1657, Bytes 4096
23:50:47.696-012 SRC: Backlog
23:50:47.699-012 CMD: Grp 0, Cmd 'FN', Idx 1, Len 9, Pld -99, Data 'CloudSens'
23:50:47.702-012 CFG: CR 431/699, Busy 0
23:50:47.706-012 RSL: RESULT = {"FriendlyName1":"CloudSens"}
23:50:47.946-012 SRC: Backlog
23:50:47.949-012 CMD: Grp 0, Cmd 'FN', Idx 2, Len 1, Pld -99, Data '"'
23:50:47.952-012 CFG: CR 431/699, Busy 0
23:50:47.955-012 RSL: RESULT = {"FriendlyName2":""}
23:50:48.197-012 SRC: Backlog
23:50:48.200-012 CMD: Grp 0, Cmd 'FN', Idx 3, Len 1, Pld -99, Data '"'
23:50:48.203-012 CFG: CR 431/699, Busy 0
23:50:48.207-012 RSL: RESULT = {"FriendlyName3":""}
23:50:48.491-012 SRC: Backlog
23:50:48.494-012 CMD: Grp 0, Cmd 'FN', Idx 4, Len 1, Pld -99, Data '"'
23:50:48.498-012 CFG: CR 431/699, Busy 0
23:50:48.501-012 RSL: RESULT = {"FriendlyName4":""}
23:50:48.508-012 APP: starte neu
-------8<---------------------------

thomas374b, 2023-12-25:
here again a trace from a failing firmware image, the firmware is one of my "minimal" defines set. about 584kb large and 18.5kB free RAM, nearly everything excluded but included USE_SCRIPT and USE_SCRIPT_SUB_COMMAND, the script was enabled and empty (only >D)

It crashes on said backlog command or saving "other configurations" via webpage

------------------8<--------------------------------------------------
00:03:19.425-013 WEB: Chunk size 654
00:03:19.427-013 WEB: Chunk size 0
00:03:19.444-018 APP: Anzahl StartvorgÃ€nge 206
00:03:32.795-018 WIF: PrÃŒfe Verbindung...
backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
00:03:47.878-017 CMD: backlog WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}
00:03:47.890-018 SRC: Serial
00:03:47.893-017 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 187, Pld -99, Data 'WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}'
00:03:47.915-017 >> BACKLOG, WebPassword2;SO3 0;SO128 1;DeviceName CloudSens;FN1 CloudSens;FN2 ";FN3 ";FN4 ";FN5 ";FN6 ";FN7 ";FN8 ";Template {"NAME":"ESP8285","GPIO":[1,1,1,1,1,1,0,0,1,1,1,0,0,1],"FLAG":0,"BASE":18}, -99, 1 
00:03:47.935-018 >>-1

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x22455341 epc2=0x00000000 epc3=0x40101510 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb80 end: 3fffffc0 offset: 0190
3ffffd10:  4c46222c 3a224741 42222c30 22455341  
3ffffd20:  7d38313a 3fff0029 3ffffe17 40249f44  
3ffffd30:  646e6d63 4556442f 37395f53 30393435  
3ffffd40:  2f62665f 37343a00 3ffffd20 3ffefd04  
...
--------->8----------------------

after reboot
---------8<----------------------
00:00:03.075-019 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["22455341","00000000","40101510"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40249f44","4020aecc","40249ed0","40100cf7","40289479","4023b4bd","40249e00","4022c930","4020af00","4022c868","4023012c","40100672","402077a4","40230b44","40230b68","401002f8","4024014c","40101279"]},"BootCount":206}}
--------->8----------------------

thomas374b, 2023-12-26:
The error is caused probably in file
     tasmota/tasmota_xdrv_driver/xdrv_10_scripter.ino
in function
    bool Script_SubCmd(void)
starting around line 9142
because of 
     char cmdbuff[128];    // <= not enough space for holding the command string
and later
     strcpy(cp, command);     // because not using  strncpy(cp, command,127); instead
     
to replay the error compile a firmware with USE_SCRIPT and USE_SCRIPT_SUB_COMMAND and use the above mentioned
backlog command with long devicename. The backlog command payload lenght should exceed 178 bytes to trigger the fault

I replaced 128 with 256 in cmdbuff array, recompiled, tested and the error vanished

sfromis, 2023-12-26:
Not related to the crash, but your logs shows that there is another reason for your long backlog commands not working. The thing is that several of the settings you can change will cause a restart, including SetOption3. Hence, Tasmota is restarting before the Backlog sequence completes. One factor in this is a default Backlog inter-command delay of 200 ms; if you use `Backlog0` instead of `Backlog` this delay will not be taken.

arendst, 2023-12-26:
I'll do some tests and will fix this.

closed 2023-12-27 by arendst