arendst/Tasmota #20324: Problems using all 3 UART on ESP32 (tasmota drivers and berry)

FulvioSpelta, 2023-12-27:
### PROBLEM DESCRIPTION
On ESP32 using all 3 UART, 2 by tasmota native drivers (modbus and PMS5003) and the last by Berry the serial console (disabled to use uart0) restart. Other strange effects on different configuration. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic):  WT32-ETH01 V1.4
- [X] Tasmota binary firmware version number used: 13.3.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: tasmota online flasher (chrome browser)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:52:05.766 CMD: Backlog Template; Module; GPIO 255
21:52:05.769 SRC: WebConsole from 0.0.0.0
21:52:05.771 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
21:52:05.783 SRC: Backlog
21:52:05.785 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
21:52:05.793 MQT: adyda/smartbox-001/stat/RESULT = {"NAME":"WT32_ETH01","GPIO":[1,1,1,1,1,1,0,0,1,0,1,1,3840,576,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,0,1,1,0,0,1],"FLAG":0,"BASE":1}
21:52:06.013 SRC: Backlog
21:52:06.015 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
21:52:06.020 MQT: adyda/smartbox-001/stat/RESULT = {"Module":{"0":"WT32_ETH01"}}
21:52:06.263 SRC: Backlog
21:52:06.265 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
21:52:06.282 MQT: adyda/smartbox-001/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"7648":"Input1"},"GPIO2":{"0":"None"},"GPIO3":{"7649":"Input2"},"GPIO4":{"0":"None"},"GPIO5":{"1312":"DS18x201"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"258":"Relay_i3"},"GPIO13":{"0":"None"},"GPIO14":{"9440":"ModBr Rx"},"GPIO15":{"9408":"ModBr Tx"},"GPIO16":{"3840":"Output Hi"},"GPIO17":{"576":"LedLink_i"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"256":"Relay_i1"},"GPIO33":{"257":"Relay_i2"},"GPIO34":{"0":"None"},"GPIO35":{"160":"Switch1"},"GPIO36":{"161":"Switch2"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
21:53:06.681 CMD: Backlog Rule1; Rule2; Rule3
21:53:06.683 SRC: WebConsole from 0.0.0.0
21:53:06.685 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
21:53:06.709 SRC: Backlog
21:53:06.711 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
21:53:06.718 MQT: adyda/smartbox-001/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":339,"Free":172,"Rules":"ON Time#Minute DO Backlog ModBusSend {\"deviceaddress\": 1, \"functioncode\": 3, \"startaddress\": 40028, \"type\":\"uint16\", \"count\":5}; ModBusSend {\"deviceaddress\": 1, \"functioncode\": 3, \"startaddress\": 40057, \"type\":\"uint16\", \"count\":2}; ModBusSend {\"deviceaddress\": 1, \"functioncode\": 3, \"startaddress\": 40061, \"type\":\"uint16\", \"count\":4} ENDON"}}
21:53:06.939 SRC: Backlog
21:53:06.941 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
21:53:06.948 MQT: adyda/smartbox-001/stat/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":158,"Free":353,"Rules":"ON System#Boot DO Backlog EthIpAddress 192.168.10.60; EthSubnetmask 255.255.255.0; EthGateway 192.168.10.1; EthDNSServer1 8.8.8.8; EthDNSServer2 8.8.4.4 ENDON"}}
21:53:07.189 SRC: Backlog
21:53:07.191 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
21:53:07.198 MQT: adyda/smartbox-001/stat/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:53:38.704 CMD: Status 0
21:53:38.706 SRC: WebConsole from 0.0.0.0
21:53:38.708 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
21:53:38.716 MQT: adyda/smartbox-001/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","",""],"Topic":"smartbox-001","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[15,15,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:53:38.729 MQT: adyda/smartbox-001/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T10:41:24","StartupUTC":"2023-12-27T10:12:14","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2023-10-21T18:59:07","SaveCount":70}}
21:53:38.742 MQT: adyda/smartbox-001/stat/STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota32)","BuildDateTime":"2023-12-12T14:31:42","Core":"2_0_14","SDK":"4.4.6.231122","CpuFrequency":160,"Hardware":"ESP32-D0WD v1.1","CR":"453/699"}}
21:53:38.753 MQT: adyda/smartbox-001/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PariniNet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00008080","00204000","00004000","00000000"]}}
21:53:38.775 MQT: adyda/smartbox-001/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1888,"Free":991,"Heap":138,"StackLowMark":1,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
21:53:38.789 MQT: adyda/smartbox-001/stat/STATUS5 = {"StatusNET":{"Hostname":"smartbox-001-7692","IPAddress":"0.0.0.0","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"48:E7:29:AA:DE:0C","IP6Global":"","IP6Local":"","Ethernet":{"Hostname":"smartbox-001-7692-eth","IPAddress":"192.168.10.60","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"48:E7:29:AA:DE:0F","IP6Global":"","IP6Local":"fe80::4ae7:29ff:feaa:de0f%en1"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:53:38.801 MQT: adyda/smartbox-001/stat/STATUS6 = {"StatusMQT":{"MqttHost":"ec9e85dc56a0449bb685463ac0f1a1cb.s1.eu.hivemq.cloud","MqttPort":8883,"MqttClientMask":"smartbox_%06X","MqttClient":"smartbox_AADE0C","MqttUser":"smartboxes","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:53:38.812 MQT: adyda/smartbox-001/stat/STATUS7 = {"StatusTIM":{"UTC":"2023-12-27T20:53:38","Local":"2023-12-27T21:53:38","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:58"}}
21:53:38.827 MQT: adyda/smartbox-001/stat/STATUS10 = {"StatusSNS":{"Time":"2023-12-27T21:53:38","Switch1":"OFF","Switch2":"OFF"}}
21:53:38.840 MQT: adyda/smartbox-001/stat/STATUS11 = {"StatusSTS":{"Time":"2023-12-27T21:53:38","Uptime":"0T10:41:24","UptimeSec":38484,"Heap":136,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":7,"Objects":106},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF"}}

```
[... cut ...]
_Add any other context about the problem here._

I'm able and available to compile tasmota and perform tests.
I'm not a professional developer so i need hints to troubleshoot the problem, anyway i can help testing.

**(Please, remember to close the issue when the problem has been addressed)**
[231223_SerialConsoleTests_1.pdf](https://github.com/arendst/Tasmota/files/13781546/231223_SerialConsoleTests_1.pdf)

[231223_SerialConsoleTests_2.pdf](https://github.com/arendst/Tasmota/files/13781548/231223_SerialConsoleTests_2.pdf)

arendst, 2023-12-28:
There is currently no user support for all available UART's. One is always used for serial console.

I'll have a look if it's possible to release the serial console if a users claims it's UART for other purposes.

Note that Tasmota on ESP32 does not support software serial.

arendst, 2023-12-28:
Done some tests and I see that some drivers work perfectly well when they need to take-over the serial console (like TasmotaClient and SerialBridge).

I'll have a look at berry if it claims the serial interface if it needs uart0. Hold on....

FulvioSpelta, 2023-12-28:
Just to give you more info. Currently i'm using this configuration:
[image]
where simply i have removed the PMS5003 driver but i've disabled serialconsole and i'm using berry.

I'm reading ok register from modbus (modbussend command) and a simple "modem at command driver" that sends a "payload" using the lora modem (serial.write command).

I'm available to support. Thanks

arendst, 2023-12-28:
Try latest dev branch and report back

sfromis, 2023-12-28:
Doing a test before these changes, I had 3 serial ports working with Berry, picking up logs from other Tasmota devices. Only issue was loss of data when bursts overran the buffers, which I saw no option for extending.

FulvioSpelta, 2023-12-28:
did you test using the same driver that i've used (modbus, PMS5003) ?

FulvioSpelta, 2023-12-29:
Hi, at first thanks for your fedback.

I've downloaded the repository and compiled the tasmota32 environment from development branch without problems.
Erased and flashed the WT32-ETH01 device - verified version: Tasmota 13.3.0.1

Note
Starting having a terminal on serialconsole.

A the very first start the log on the serialconsole are ok
After defined the wifi to connect only this log are written (the configuration excluding wifi are at default):
[image]

I've struggled a bit to configure the module to use eth (having both wifi and eth seem to heavily slow down the tasmota but this is OT now).

I've defined the same configuration of the test posted:
[image]

Using this version i've successfully used both modbus and berry.
I've done just some tests, I'll use more deeply in the next days.

Details:
Start log (power on)
```lua
00:00:00.002-214/50 HDW: ESP32-D0WD v1.1 
00:00:00.014-213/50 UFS: FlashFS mounted with 304 kB free
00:00:00.025 CFG: Loaded from File, Count 33
00:00:00.036 QPC: Count 1
00:00:00.037 CFG: CR 403/699, Busy 0
00:00:00.047 TYA: Active=0
00:00:00.055 ROT: Mode 1
00:00:00.066 BRY: GC from 3692 to 3492 bytes, objects freed 4/20 (in 1 ms) - slots from 36/61 to 28/61
00:00:00.098 CFG: No '*.autoconf' file found
00:00:00.104 BRY: GC from 5852 to 4446 bytes, objects freed 6/40 (in 1 ms) - slots from 64/91 to 41/91
00:00:00.105 BRY: Berry initialized, RAM used 4446 bytes
00:00:00.113 BRY: No 'preinit.be'
00:00:00.117 MBS: Serial UART2
00:00:00.118 MBS: MBR SW ser init at 9600 baud
00:00:00.119 SRC: Restart
00:00:00.131 Project tasmota - Tasmota Version 13.3.0.1(tasmota32)-2_0_14(2023-12-29T00:45:45)
00:00:01.621 ETH: Attempting connection...
00:00:01.623 ETH: Connected at 100Mbps Full Duplex, Mac 48:E7:29:AA:DE:0F, Hostname tasmota-AADE0C-7692-eth
00:00:01.676 ETH: IPv4 192.168.10.61, mask 255.255.255.0, gateway 192.168.10.1
00:00:01.693 DSB: Sensors found 0
00:00:01.695 PMS: Serial UART1
00:00:01.718 BRY: No 'autoexec.be'
00:00:01.736 NTP: Sync time...
00:00:02.740 WIF: DNS failed for 2.pool.ntp.org after 1002 ms
00:00:02.756 NTP: Sync time...
00:00:03.098 ETH: IPv6 Local fe80::4ae7:29ff:feaa:de0f%en1
00:00:03.758 WIF: DNS failed for 2.europe.pool.ntp.org after 1001 ms
00:00:03.777 HTP: Web server active on tasmota-AADE0C-7692-eth with IP address 192.168.10.61
00:00:03.782 NTP: Sync time...
00:00:04.784 WIF: DNS failed for 2.nl.pool.ntp.org after 1002 ms
00:00:04.801 NTP: Sync time...
00:00:04.803 WIF: DNS resolved '192.168.10.1' (192.168.10.1) in 1 ms
00:00:05.806 NTP: No reply from 192.168.10.1
00:00:05.823 NTP: Sync time...
00:00:06.759 WIF: DNS resolved '0.pool.ntp.org' (162.159.200.1) in 934 ms
00:00:06.783 RTC: UTC 2023-12-29T01:03:14Z, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
02:03:14.000 RTC: Synced by NTP
02:03:14.045 HTP: Console
02:03:14.093 RSL: INFO1 = {"Info1":{"Module":"WT32_ETH01","Version":"13.3.0.1(tasmota32)","FallbackTopic":"cmnd/DVES_AADE0C_fb/","GroupTopic":"cmnd/tasmotas/"}}
02:03:14.098 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Ethernet":{"Hostname":"tasmota-AADE0C-7692-eth","IPAddress":"192.168.10.61"}}}
02:03:14.101 RSL: INFO3 = {"Info3":{"RestartReason":"RTC Watch dog reset digital core and rtc module","BootCount":12}}
02:03:14.105 RSL: RESULT = {"POWER1":"OFF"}
02:03:14.106 RSL: POWER1 = OFF
02:03:14.107 RSL: RESULT = {"POWER2":"OFF"}
02:03:14.108 RSL: POWER2 = OFF
02:03:14.110 RSL: RESULT = {"POWER3":"OFF"}
02:03:14.111 RSL: POWER3 = OFF
[... 47 lines cut ...]
I've sent data from berry to terminal, received data in berry (see in the log the LOR: lines), read registers in modbus
```lua
02:08:55.456 BRY: Serial UART0
02:08:55.458 LOR: RakModem hard reset
02:08:55.460 SRC: Berry
02:08:55.467 RSL: RESULT = {"POWER3":"ON"}
02:08:55.469 RSL: POWER3 = ON
02:08:55.521 SRC: Berry
02:08:55.527 RSL: RESULT = {"POWER3":"OFF"}
02:08:55.529 RSL: POWER3 = OFF
02:10:00.066 LOR: HB sent: at+send=1:00
 counter:1
02:10:10.357 LOR: Received Raw = p
02:10:10.657 LOR: Received Raw = i
02:10:10.859 LOR: Received Raw = p
02:10:10.967 LOR: Received Raw = p
02:10:11.257 LOR: Received Raw = o
02:10:11.556 LOR: Received Raw = 

02:10:30.612 CMD: ModBusSend {"deviceaddress": 1, "functioncode": 3, "startaddress": 40028, "type":"uint16", "count":5}
02:10:30.613 SRC: WebConsole from 0.0.0.0
02:10:30.616 CMD: Grp 0, Cmd 'MODBUSSEND', Idx 1, Len 90, Pld -99, Data '{"deviceaddress": 1, "functioncode": 3, "startaddress": 40028, "type":"uint16", "count":5}'
02:10:30.624 RSL: RESULT = {"ModbusSend":"Done"}
02:10:30.829 RSL: RESULT = {"ModbusReceived":{"DeviceAddress":1,"FunctionCode":3,"StartAddress":40028,"Length":15,"Count":5,"Values":[0,0,0,0,0]}}
```

FulvioSpelta, 2023-12-29:
Update: up & running for 5h with berry and modbus used.

I'll add in hw (not only in configuration) the pms5003 sensor and report.

FulvioSpelta, 2023-12-29:
Non directly related to uart problems, i kindly ask for an hint then i'll delete from here.

After the first configuration (ethernet and wifi off) i'm not able to save the conf anymore.
Example: rule1 (modbus readings, mqtt configuration) are lost when the tasmota restart.
More: if i upload an autoexec.be file using file manager it tells "upload" but (after upload finished) no file is listed even checking "show hidden".

Attached the current conf (no mqtt due, requiring restart, is deleted just after "save").
Do you see something that can generate this strange issue ?
Thanks

File json generated from .dmp downloaded from module using decode_config last version.
[Config_tasmota_AADE0C_7692_13.3.0.1_uart.txt](https://github.com/arendst/Tasmota/files/13792130/Config_tasmota_AADE0C_7692_13.3.0.1_uart.txt)

FulvioSpelta, 2023-12-30:
Hi,
i've modified the tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_serial.ino as you has done in the commit in the master (13_3_0) version and built a specific test system based on esp32-wroom32 in order to avoid potential "side effect" from the ethernet interface of the WT32-ETH01 module. Here the results_

Environment ESP32-Wroom32

Test 1 - 13.3.0
Defined just modbus and berry -> test ok
Added pms5003 -> test ko (same as first post)

Test 2 - 13.3.0 + changes in xdrv_52_3_berry_serial.ino from commit [d5a4f84](https://github.com/arendst/Tasmota/commit/d5a4f8441bfc0b98d6b482b684cc542d5ed4969b)
Defined just modbus and berry -> test ok
Added pms5003 -> test ok

I think that the bus is fixed. I'll use this "patched" version till next tasmota official release.
Thanks a lot

arendst, 2023-12-30:
Happy new year too.

PR #21496 (2024-06-22): FIX: Allow use of UART0 with enabled USB_CDC_CONSOLE
## Description:
When tasmota is compiled with `USE_USB_CDC_CONSOLE`, the ´ClaimSerial` method will return immediately and will not release UART0

This makes sense, as long as USB is used for console, but if we fall back to serial, because no USB is connected, we must release the console, so that all 3 UARTs on the ESP32 can be used.
(Possible broken since https://github.com/rPraml/Tasmota/commit/84ced0fcab6f3af26602cd8fc23df5ba7bcc8c1f)

In my case I tried to connect 3 SML meters to a ESP32 and I always had some garbage in the web console, w [... cut]
closed 2023-12-30 by arendst