arendst/Tasmota #20341: Assistance needed in adding Sunton ESP32-8048S070 ESP32-S3 7inch Display support

markusdd, 2023-12-29:
Hi. After consulting the LVGL, Berry and Display Docs it seems that we need a device template and universal display driver config for this board to work with Tasmota.
These seem to be pretty nice, but also very new devices. 

The links to it is here:
https://de.aliexpress.com/item/1005004952726089.html?pdp_npi=4%40dis%21EUR%21%E2%82%AC%2025%2C14%21%E2%82%AC%2025%2C14%21%21%21192.91%21192.91%21%402101c2d217038451419932055d6c3e%2112000031135431788%21sh01%21DE%21172906472%21&spm=a2g0o.store_pc_home.productList_2007684931608.pic_3&gatewayAdapt=glo2deu

There is a also (luckily) a relatively good docs package available here:
http://pan.jczn1688.com/directlink/1/ESP32%20module/7.0inch_ESP32-8048S070.zip?spm=a2g0o.detail.1000023.1.735afVh9fVh9NX&file=7.0inch_ESP32-8048S070.zip

A TLDR from the docs that I could gather is as follows:
- 800x480 TFT screen, 7 inch
- base MCU is a ESP32-S3-WROOM-1 Module with 16MB Flash and 8MB PSRAM
- it uses 80MHz QIO mode
- it uses OPI PSRAM
- touch driver is a GT911
- TFT driver is an EK9716, which seems to be connected using the ESP32-S3 "exclusive" RGB mode

So what is needed is essentially a Tasmota32-S3-QIO-OPI-LVGL image as a base. Is that available as a prebuild somewhere?
Web Installer has this, but it seems this does not have LVGL? 
[image]

As for the display and touch I have skimmed through their docs and demos. They use and adapted lv_conf.h (but seems relatively straightforward). Also, they have the pin config for the GT911 and the RGB mode in arduino GFX driver in the code. I ahve attached the relevant 3 files.
[touch_display_lvconf.zip](https://github.com/arendst/Tasmota/files/13794921/touch_display_lvconf.zip)

relevant sections seem to be:

lv-conf.h:
```c
/*====================
   COLOR SETTINGS
 *====================*/

/*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)*/
#define LV_COLOR_DEPTH 16

/*Swap the 2 bytes of RGB565 color. Useful if the display has an 8-bit interface (e.g. SPI)*/
#define LV_COLOR_16_SWAP 0

/*Enable more complex drawing routines to manage screens transparency.
 *Can be used if the UI is above another layer, e.g. an OSD menu or video player.
 *Requires `LV_COLOR_DEPTH = 32` colors and the screen's `bg_opa` should be set to non LV_OPA_COVER value*/
#define LV_COLOR_SCREEN_TRANSP 0

/* Adjust color mix functions rounding. GPUs might calculate color mix (blending) differently.
 * 0: round down, 64: round up from x.75, 128: round up from half, 192: round up from x.25, 254: round up */
#define LV_COLOR_MIX_ROUND_OFS (LV_COLOR_DEPTH == 32 ? 0: 128)

/*Images pixels with this color will not be drawn if they are chroma keyed)*/
#define LV_COLOR_CHROMA_KEY lv_color_hex(0x00ff00)         /*pure green*/
```

LvglWidgets.ino:
```c
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
);
// option 1:
// 7寸 50PIN 800*480
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
  bus,
//  800 /* width */, 0 /* hsync_polarity */, 8/* hsync_front_porch */, 2 /* hsync_pulse_width */, 43/* hsync_back_porch */,
//  480 /* height */, 0 /* vsync_polarity */, 8 /* vsync_front_porch */, 2/* vsync_pulse_width */, 12 /* vsync_back_porch */,
//  1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);

    800 /* width */, 0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    480 /* height */, 0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);
```

touch.h:
```c
/* uncomment for GT911 */
 #define TOUCH_GT911
 #define TOUCH_GT911_SCL 20
 #define TOUCH_GT911_SDA 19
 #define TOUCH_GT911_INT -1
 #define TOUCH_GT911_RST 38
 #define TOUCH_GT911_ROTATION ROTATION_NORMAL
 #define TOUCH_MAP_X1 800
 #define TOUCH_MAP_X2 0
 #define TOUCH_MAP_Y1 480
 #define TOUCH_MAP_Y2 0

//...

int touch_last_x = 0, touch_last_y = 0;

#if defined(TOUCH_FT6X36)
#include <Wire.h>
#include <FT6X36.h>
FT6X36 ts(&Wire, TOUCH_FT6X36_INT);
bool touch_touched_flag = true, touch_released_flag = true;

#elif defined(TOUCH_GT911)
#include <Wire.h>
#include <TAMC_GT911.h>
TAMC_GT911 ts = TAMC_GT911(TOUCH_GT911_SDA, TOUCH_GT911_SCL, TOUCH_GT911_INT, TOUCH_GT911_RST, max(TOUCH_MAP_X1, TOUCH_MAP_X2), max(TOUCH_MAP_Y1, TOUCH_MAP_Y2));
```

Now my goal is: I want to get this going to build a berry lvgl GUI with Tasmota on that thing, but despite all the docs I am a bit lost now as to what to prepare when and where.

Based on the info I dug up, can someone assist? I am also happy to then feed back my findings and commit a device template once I have it working.

sfromis, 2023-12-29:
I'm pretty sure that `tasmota32s3-qio_opi.bin` includes LVGL, as this feature is included in the CDC variant of the same build.

markusdd, 2023-12-29:
> I'm pretty sure that `tasmota32s3-qio_opi.bin` includes LVGL, as this feature is included in the CDC variant of the same build.
> 
> I'd try with the "Sunton-S3-Display" auto-configuration. While this was likely created for the 5" variant, they seem to be pretty similar, same resolution.

helpful info, thx. They are indeed similar regading the MCU but the touch driver is a different one.

I have now gone through their demo files and came up wiht the following ini file:

```ini
:H,EK9716,800,480,16,RGB,41,40,39,42,15,7,6,5,4,9,46,3,8,16,1,14,21,47,48,45,16
:S,2,1,1,0,40,20
:V,0,210,30,16,0,22,13,10,1
:0,00
:1,01
:2,02
:3,03
:TI1,5d,20,19,-1,38
#
```

Throughts?

sfromis, 2023-12-29:
I'd first get the display to work without touch.

FWIF (IDK how much), this table suggests touch GT911 for both 5" and 7"
When I wrote "pretty similar" that was not at all about the MCU, but the rest.

markusdd, 2023-12-29:
I think touch is non-critical. As you say, this seems very similar to the sunton 5" and also other displays which use this controller, so I feel confident about that, it is essentially the I2C address and the lines.

I am currently working through the schematics from the docs package to compare the pins but if they really do match all that is needed should be the display config from above.

sfromis, 2023-12-29:
The `display.ini` file in the 5" autoconf is:
```
:H,ST7262,800,480,16,RGB,40,41,39,42,2,15,16,4,45,48,47,21,14,8,3,46,9,1,5,6,7,14
:S,2,1,1,0,40,20
:V,0,8,4,8,0,8,4,8,1
:0,00
:1,01
:2,02
:3,03
:TI1,5d,*,*,-1,38
#
```

markusdd, 2023-12-29:
yes, I have checked the schematics now.
The wiring for touch, SD card, UART etc is identical, but the panels have wildly different parameters and pin assignments.
See the comparison here:
[image]

So in my book that means we can reuse the template but need a new display.ini.

Can you point me to where the autoconf data is located in the repo?

Also: do you agree with my observation?

sfromis, 2023-12-29:
https://github.com/tasmota/autoconf/tree/main/raw/esp32s3/SunTon-S3-Display

Yeah, what you post does look like gpio assignments needing to be remapped. What's important are what specific display controller chips the panels have, and setup of those.

markusdd, 2023-12-29:
yes.

the 5" uses a ST7262 as driver and the 7" uses a EK9716.

So essentially I need to make a PR to the autoconf repo if my display.ini works.

it's a bit of a bummer that the current module is just called SunTon-S3-Display as this obviously now just applies to the 5 inch.

I will propose to rename it to SunTon-S3-Display-5inch and add a SunTon-S3-Display-7inch next to it.

closed 2023-12-30 by s-hadinger

sfromis, 2023-12-30:
Did you also get touch to work?

markusdd, 2023-12-30:
> Did you also get touch to work?

Yes. As mentioned somewhat buried above the capacitive version (there is no resistive one for 7") is identical to the 5" capacitive one, so the :TI1 line is the same