arendst/Tasmota #20344: Error when running ShutterSetClose during shutters calibration

rlegrand, 2023-12-29:
### PROBLEM DESCRIPTION
Tasmota returns ERROR when running the ShutterSetClose command.
My device is [this one](https://templates.blakadder.com/etersky_WF-CS01.html).
For the conf I did several tests but the one [provided here](https://templates.blakadder.com/WF-CS02_EU) is the last I tried after a reset of the device.
Thanks for any help ! 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): etersky_WF-CS01
- [x] Tasmota binary firmware version number used: 13.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Through wifi
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
19:21:41.285 CMD: Backlog Template; Module; GPIO 255
19:21:41.311 RSL: RESULT = {"NAME":"WF-CS02 Tuya","GPIO":[544,0,289,162,226,33,0,0,32,224,290,225,288,0],"FLAG":0,"BASE":18}
19:21:41.545 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
19:21:41.751 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
19:23:47.031 CMD: Backlog Rule1; Rule2;
19:23:47.060 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":39,"Free":472,"Rules":"on Power3#State=1 do ShutterStop1 endon"}}
19:23:47.290 RSL: RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":233,"Free":278,"Rules":"on Power1#State=0 do LEDPower1 0 endon on Power1#State=1 do LEDPower1 1 endon on Power2#State=0 do LEDPower2 0 endon on Power2#State=1 do LEDPower2 1 endon on Power3#State=0 do LEDPower3 0 endon on Power3#State=1 do LEDPower3 1 endon"}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
19:24:38.881 CMD: Status 0
19:24:38.888 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_F751BA","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,4,4,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:24:38.919 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:31:38","StartupUTC":"2023-12-29T17:53:00","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2023-12-29T18:51:17","SaveCount":19,"SaveAddress":"F9000"}}
19:24:38.949 RSL: STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota)","BuildDateTime":"2023-12-12T14:29:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"387/699"}}
19:24:38.977 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RLIBOX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A029","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
19:24:38.009 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":637,"Free":364,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
19:24:39.026 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-F751BA-4538","IPAddress":"192.168.86.30","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.86.1","DNSServer2":"0.0.0.0","Mac":"98:F4:AB:F7:51:BA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:24:39.056 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F751BA","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:24:39.083 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-12-29T18:24:39","Local":"2023-12-29T19:24:39","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
19:24:39.109 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-12-29T19:24:39"}}
19:24:39.137 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-12-29T19:24:39","Uptime":"0T00:31:39","UptimeSec":1899,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"RLIBOX","BSSId":"E4:5E:1B:8E:DE:5A","Channel":6,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:06"}}}
19:24:39.163 RSL: STATUS13 = {"StatusSHT":{}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
19:25:43.388 CMD: ShutterSetClose
19:25:43.390 SRC: WebConsole from 192.168.86.159
19:25:43.392 CMD: Grp 0, Cmd 'SHUTTERSETCLOSE', Idx 1, Len 0, Pld -99, Data ''
19:25:43.397 RSL: RESULT = {"Command":"Error"}
```

### TO REPRODUCE
Use the same device and run:
```lua
Reset 1
Template {"NAME":"WF-CS02 Tuya","GPIO":[157,0,53,11,23,18,0,0,17,21,54,22,52],"FLAG":0,"BASE":18}
SetOption80 1
SetOption13 1
Backlog PulseTime1 0; PulseTime2 0
Pulsetime3 1
Backlog Interlock 1,2,3; Interlock ON
PowerRetain 1
Rule1 on Power3#State=1 do ShutterStop1 endon
Rule1 1
Rule2 on Power1#State=0 do LEDPower1 0 endon on Power1#State=1 do LEDPower1 1 endon on Power2#State=0 do LEDPower2 0 endon on Power2#State=1 do LEDPower2 1 endon on Power3#State=0 do LEDPower3 0 endon on Power3#State=1 do LEDPower3 1 endon
Rule2 1
SwitchMode2 4
SwitchMode3 4
ShutterSetClose
```

### EXPECTED BEHAVIOUR
No error.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
The only thing I can think about is it looks like there is something related to tuya in this device ( as the template name says [here](https://templates.blakadder.com/WF-CS02_EU) ) and the calibration doc tells tuya is not supported for shutters calibration.
But I don't know if it's really the issue while the conf proposed for the device is a preamble to the shutter calibration so I guess it should work.

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-12-30:
This works to spec. After starting calibration no other commands are supported. Pls let me know if you need further support. The calibration is still beta state and only work if you strictly follow the path.

stefanbode, 2023-12-30:
Calibration is supported for all devices that reports the power usage on the shutter during operation. Anyhow still struggle with reboots on some of the hardware doing the measurement.