arendst/Tasmota #20345: Tasmota KNX: No Status Feedback, maybe IP_TTL not correct

hyperjojo, 2023-12-29:
### PROBLEM DESCRIPTION
I flashed several different Shellys with Tasmota KNX and also successfully integrated some into my KNX.

However, the Shelly RGBW2 have a problem with the status feedback. Configuration is the same as the functioning devices (both on the KNX side and in the Tasmota). All devices logged into the same access point, no special settings.

Noticeable: When I look at the difference in feedback in Wireshark, I notice that the RGBW2 indicate a time to live of 1 in the IPv4 header. Since the packets are routed to a different VLAN for me (multicast router), the TTL has already expired on the first hop and is no longer routed (this is also what my router's traces show me: Discarded, time exceeded, hop limit exceeded in transit for KNX).

Does anyone have any idea why the Tasmota on the RGBW2 (with ESP8266EX) behaves differently in the IP stack than, for example, a Plus1Mini (with ESP32-C3 v0.4)? 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly RGBW2
- [x] Tasmota binary firmware version number used: 13.3.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:30:07.797 RSL: RESULT = {"NAME":"Shelly RGBW2","GPIO":[0,0,288,0,419,1,0,0,416,32,418,417,0,0],"FLAG":0,"BASE":18}
21:30:07.000 RSL: RESULT = {"Module":{"0":"Shelly RGBW2"}}
21:30:08.205 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"288":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"419":"PWM4"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"32":"Button1"},"GPIO14":{"418":"PWM3"},"GPIO15":{"417":"PWM2"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:31:10.780 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-LEDStrip-Fluroben","FriendlyName":["Tasmota-LEDStrip-Fluroben"],"Topic":"tasmota_28587B","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:31:10.815 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:03:18","StartupUTC":"2023-12-29T20:27:52","Sleep":50,"CfgHolder":4617,"BootCount":157,"BCResetTime":"2023-12-21T23:51:19","SaveCount":736,"SaveAddress":"FB000"}}
21:31:10.847 RSL: STATUS2 = {"StatusFWR":{"Version":"13.3.0(knx)","BuildDateTime":"2023-12-12T14:28:56","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"430/699"}}
21:31:10.866 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT@Liwa4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0A008001","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
21:31:10.899 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"156085","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","9F9AC587","04268011","000000CF","010013C0","8000F181","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,27,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
21:31:10.932 RSL: STATUS5 = {"StatusNET":{"Hostname":"Tasmota-LEDStrip-Fluroben","IPAddress":"192.168.108.173","Gateway":"192.168.108.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.108.1","DNSServer2":"0.0.0.0","Mac":"48:3F:DA:28:58:7B","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:31:10.959 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-12-29T20:31:10","Local":"2023-12-29T21:31:10","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
21:31:10.976 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-12-29T21:31:10"}}
21:31:10.984 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-12-29T21:31:10","Uptime":"0T00:03:18","UptimeSec":198,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":36,"Color":"4F005C00","HSBColor":"291,100,36","White":0,"Channel":[31,0,36,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IoT@Liwa4","BSSId":"0A:A0:57:9C:4F:65","Channel":6,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:01:35"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
[image]

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-12-30:
Ah, I didn't realize KNX uses IP Multicast. I'm not aware of what is the correct value for TTL in multicast, considering that most use-cases will limit to the VLAN itself. I will dig into the library if I find something.

s-hadinger, 2023-12-30:
I must be missing something, I tried to configure a fake KNX device but I'm seeing any multicast traffic from the Tasmota device. I tried `knxtx_val1 1` or even mapping to a Button.

I do see the following in the logs: `15:52:56.576 KNX: KNX TX 1 = 1.00 sent to 2/2/1`

hyperjojo, 2023-12-31:
Will it help if I provide additional screenshots or data? If yes, what data do you need?

s-hadinger, 2024-01-03:
I would just like to have a minimal setup that can send multicast UDP packets

hyperjojo, 2024-01-03:
I have several Shelly RGBW2 in use. How can I help you now? Should I run any tests and record using Wireshark? Give me a short instruction on what to do.

s-hadinger, 2024-01-04:
I did a test with ESP32 over ethernet and I confirm the TTL is 255. I don't know why but I can't capture the multicast packet over wifi on my computer; which makes testing with ESP8266 challenging. I will keep trying.

hyperjojo, 2024-01-04:
Maybe your Wifi Access Point can capture the traffic on the Wifi Interface? This is how I captured the multicast traffic.

barbudor, 2024-01-04:
@s-hadinger 
> I did a test with ESP32 over ethernet and I confirm the TTL is 255. I don't know why but I can't capture the multicast packet over wifi on my computer; which makes testing with ESP8266 challenging. I will keep trying.

Over WLAN you probably have to subscribe to the MCast for the AP to send you the packets
On WLAN, basic switches (not managed) may forward the MCast to every ports

barbudor, 2024-01-04:
found that Python gist which could be convenient:
https://gist.github.com/friek/6c10535b2e43f0e0867ce21b0679a337

hyperjojo, 2024-01-29:
any idea?

barbudor, 2024-01-29:
@hyperjojo
Do you confirm that the only device that has a TTL of 1 is also the only one based on ESP8266 ?
All other working devices are ESP32 based ?

barbudor, 2024-01-29:
I confirm that in the ESP8266 WifiUDP module, the default TTL for Multicast is 1:
```
  virtual int beginPacketMulticast(IPAddress multicastAddress,
                                   uint16_t port, 
                                   IPAddress interfaceAddress,
                                   int ttl = 1);
```

The equivalent on ESP32 doesn't offer the possibility to choose the TTL and seems to use 255 as standard

I suggest that we enforce 255 in the ESP8266 version of the KNX library

I will provide you with a test binary

barbudor, 2024-01-29:
Here is a test binary (both bin and bin.gz)
Please report your test results 

[test tasmota-knx.zip](https://github.com/arendst/Tasmota/files/14090002/test.tasmota-knx.zip)

hyperjojo, 2024-01-30:
> @hyperjojo Do you confirm that the only device that has a TTL of 1 is also the only one based on ESP8266 ? All other working devices are ESP32 based ?

yes, i will test the binary the next days...

thanks!
hyperjojo

Jason2866, 2024-01-30:
A correct working/configured multicast router shouldn't drop multicast packets with multicast TTL=1

hyperjojo, 2024-01-30:
> Here is a test binary (both bin and bin.gz) Please report your test results

this works fine!
Thank you for your help!

PR #20616 (2024-01-30): Enforce KNX TTL=255 on ESP8266
## Description:

**Related issue (if applicable):** fixes #20345

On ESP32, Multicast UDP TTL is by default set to 255 but only 1 on ESP8266
This align the TTL to 255 when used from KNX library

Submitted to test confirmation by @hyperjojo

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.6
  - [ ] The code change is teste [... cut]