arendst/Tasmota #20350: Berry LVGL Image button set_src fails if IMGBTN_STATE_RELEASED not used

ukoda, 2023-12-29:
### PROBLEM DESCRIPTION
The image button set_src function only works if the first argument is lv.IMGBTN_STATE_RELEASED.  If lv.IMGBTN_STATE_PRESSED, lv.IMGBTN_STATE_CHECKED_PRESSED or lv.IMGBTN_STATE_CHECKED_RELEASED is used it will fail with incomplete screen render.  After a while the Berry console becomes available and shows the error message "BRY: failed to run compiled code 'type_error' - Unexpected argument type 'i', expected 'lv.lv_imgbtn_state'".

Code snippet, full code at end of this post:
```berry
ui_Main_ImgButton = lv.imgbtn(ui_StatePanel)
ui_Main_ImgButton.set_src(lv.IMGBTN_STATE_RELEASED, nil, "A:/sleeping113x80.png", nil)          # Renders without error
# These three all fail with error:
# BRY: failed to run compiled code 'type_error' - Unexpected argument type 'i', expected 'lv.lv_imgbtn_state'
ui_Main_ImgButton.set_src(lv.IMGBTN_STATE_PRESSED, nil, "A:/campfire113x80.png", nil)           # Fails
ui_Main_ImgButton.set_src(lv.IMGBTN_STATE_CHECKED_PRESSED, nil, "A:/sleeping113x80.png", nil)   # Fails
ui_Main_ImgButton.set_src(lv.IMGBTN_STATE_CHECKED_RELEASED, nil, "A:/campfire113x80.png", nil)  # Fails
ui_Main_ImgButton.set_width(113)
ui_Main_ImgButton.set_height(80)
ui_Main_ImgButton.set_align(lv.ALIGN_CENTER)
```

My code is an attempted port of an existing native C app I created and currently in use. My objective is to retain the existing functionality while bringing in the advantages Tasmota offers.

NB: The existing C code is just using the image button as a status image that is toggled using the checkbox state.  It was the easy way to to do things with the prior environment.  I realise there is probably other, better, ways to do this but figure this issue will hit someone else as the Berry LVGL support sees wider use and therefore worth reporting rather than simply trying to working around.

I am aware of issue [11784](https://github.com/arendst/Tasmota/discussions/11784).  I have searched the Tasmota source code and it would appear IMGBTN_STATE_PRESSED etc are present and do not appear to be treated differently from IMGBTN_STATE_RELEASED.  In parallel with waiting for feedback on this issue I will dig deeper to try and understand how Berry works in this case.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [Y] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [Y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [Y] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [Y] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [N] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [Y] Device used (e.g., Sonoff Basic): [ESP32-2432S028R](https://templates.blakadder.com/sunton_ESP32-2432S028.html)
- [Y] Tasmota binary firmware version number used: v13.3.0 development 13a9c3b5b
  - [N] Pre-compiled
  - [Y] Self-compiled
- [Y] Flashing tools used: esptool.py
- [Y] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:36:51.592 CMD: Backlog Template; Module; GPIO 255
10:36:51.631 MQT: ema/tasmota_469064/stat/RESULT = {"NAME":"ESP32-2432S028","GPIO":[6210,1,800,0,448,0,1,1,672,704,736,768,449,450,1,1,0,992,1,1,0,737,480,1,0,0,0,0,705,7264,4768,1,0,0,0,673],"FLAG":0,"BASE":1}
10:36:51.836 MQT: ema/tasmota_469064/stat/RESULT = {"Module":{"0":"ESP32-2432S028"}}
10:36:52.052 MQT: ema/tasmota_469064/stat/RESULT = {"GPIO0":{"6210":"Option A3"},"GPIO1":{"0":"None"},"GPIO2":{"800":"SPI DC1"},"GPIO3":{"0":"None"},"GPIO4":{"448":"PWM_i1"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"672":"SPI MISO1"},"GPIO13":{"704":"SPI MOSI1"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"768":"SPI CS1"},"GPIO16":{"449":"PWM_i2"},"GPIO17":{"450":"PWM_i3"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"992":"Backlight"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"737":"SPI CLK2"},"GPIO26":{"480":"Buzzer"},"GPIO27":{"0":"None"},"GPIO32":{"705":"SPI MOSI2"},"GPIO33":{"7264":"XPT2046 CS"},"GPIO34":{"4768":"ADC Light1"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"673":"SPI MISO2"}}

```
- [N] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [Y] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:38:02.406 CMD: Status 0
10:38:02.413 MQT: ema/tasmota_469064/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_469064","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:38:02.419 MQT: ema/tasmota_469064/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://REDACTED/tasmota32-lvgl.bin","RestartReason":"Software reset CPU","Uptime":"0T01:00:54","StartupUTC":"2023-12-29T20:37:08","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2023-12-27T18:03:04","SaveCount":64}}
10:38:02.423 MQT: ema/tasmota_469064/stat/STATUS2 = {"StatusFWR":{"Version":"13.3.0.1(lvgl-haspmota)","BuildDateTime":"2023-12-28T10:12:59","Core":"2_0_14","SDK":"4.4.6.231122","CpuFrequency":240,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"419/699"}}
10:38:02.427 MQT: ema/tasmota_469064/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ACCL2GHz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
10:38:02.437 MQT: ema/tasmota_469064/stat/STATUS4 = {"StatusMEM":{"ProgramSize":2318,"Free":561,"Heap":106,"StackLowMark":19,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","8FAAC7DF","00048021","000006CF","010013C0","40000981","40004080","00200000","5400003D","800200A0","00000000"],"Drivers":"1,2,3,4,5,8,9,10,12,13,16,20,24,26,27,29,38,50,52,54,55,56,62,67,68,82,88","Sensors":"1,2,3,5,6,8,9,85,127","I2CDriver":"7,9,10,58,59,77"}}
10:38:02.446 MQT: ema/tasmota_469064/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota-469064-4196","IPAddress":"10.20.1.217","Gateway":"10.20.1.1","Subnetmask":"255.255.0.0","DNSServer1":"10.20.1.1","DNSServer2":"0.0.0.0","Mac":"54:43:B2:46:90:64","IP6Global":"fd8c:4962:c2d6::5643:b2ff:fe46:9064","IP6Local":"fe80::5643:b2ff:fe46:9064%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.20.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
10:38:02.451 MQT: ema/tasmota_469064/stat/STATUS6 = {"StatusMQT":{"MqttHost":"REDACTED","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_469064","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:38:02.456 MQT: ema/tasmota_469064/stat/STATUS7 = {"StatusTIM":{"UTC":"2023-12-29T21:38:02Z","Local":"2023-12-30T10:38:02","StartDST":"2023-09-24T02:00:00","EndDST":"2023-04-02T02:00:00","Timezone":99,"Sunrise":"05:49","Sunset":"20:51"}}
10:38:02.469 MQT: ema/tasmota_469064/stat/STATUS10 = {"StatusSNS":{"Time":"2023-12-30T10:38:02","ANALOG":{"Illuminance1":1}}}
10:38:02.480 MQT: ema/tasmota_469064/stat/STATUS11 = {"StatusSTS":{"Time":"2023-12-30T10:38:02","Uptime":"0T01:00:54","UptimeSec":3654,"Heap":105,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":11,"Objects":143},"POWER1":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"C2:56:27:BB:F8:7C","Channel":1,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
[... cut ...]
ui_RoomValueLabel = lv.label(ui_TemperaturePanel)
ui_RoomValueLabel.set_width(50)
ui_RoomValueLabel.set_height(40)
ui_RoomValueLabel.set_x(0)
ui_RoomValueLabel.set_y(5)
ui_RoomValueLabel.set_align(lv.ALIGN_TOP_RIGHT)
ui_RoomValueLabel.set_text("20")
ui_RoomValueLabel.set_style_text_align(lv.TEXT_ALIGN_CENTER, lv.PART_MAIN | lv.STATE_DEFAULT)
ui_RoomValueLabel.set_style_text_font(lg_font, lv.PART_MAIN | lv.STATE_DEFAULT)
```

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-12-30:
Oops, you are right. The type mapping from `lv_imgbtn_state_t` to `int` was missed. It is now fixed in the latest development release #20354.

Thanks for reporting.

And nice display !!!

closed 2023-12-30 by s-hadinger

ukoda, 2023-12-30:
I can confirm that has resolved my issue.  EMA and I thank you for the prompt fix :-)

ukoda, 2023-12-30:
@s-hadinger a similar bug may exist with the clear_flag function which works with lv.OBJ_FLAG_HIDDEN but fails with lv.OBJ_FLAG_SCROLLABLE.  It gives the error "BRY: failed to run compiled code 'type_error' - Unexpected argument type 'b', expected 'i'".

It is time for bed now so I will do more digging tomorrow and look to create a new issue if needed.

s-hadinger, 2023-12-30:
This is because you put `lv.OBJ_FLAG_SCROLLABLE = false` at the beginning of the script. You just masked the actual value of `lv.OBJ_FLAG_SCROLLABLE` being `16`

PR #20354 (2023-12-30): LVGL fix type for lv_imgbtn
## Description:

**Related issue (if applicable):** fixes #20350

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.14
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
 [... cut]