arendst/Tasmota #20364: Tasmota does not support absolute URLs http packets using the API

arlyon, 2023-12-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I am using tasmota 13.3.0. When attempting to issue http requests to the API, it appears the server does not support the _absolute form_ of the request target, returning a 404.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): LocalBytes PM
- [x] Tasmota binary firmware version number used: 13.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: built-in ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:46:39.954 RSL: RESULT = {"NAME":"LocalBytes PM","GPIO":[0,0,0,32,2720,2656,0,0,2624,320,224,0,0,0],"FLAG":0,"BASE":18}
19:46:40.174 RSL: RESULT = {"Module":{"0":"LocalBytes PM"}}
19:46:40.428 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:47:02.822 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9E1A04","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:47:02.854 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:37","StartupUTC":"2023-12-31T18:39:25","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2023-07-06T14:19:25","SaveCount":221,"SaveAddress":"F9000"}}
19:47:02.887 RSL: STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota)","BuildDateTime":"2023-12-12T14:29:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285H16","CR":"367/699"}}
19:47:02.906 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Cat",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
19:47:02.938 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":637,"Free":364,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
19:47:02.970 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-9E1A04-6660","IPAddress":"192.168.1.71","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.254","DNSServer2":"0.0.0.0","Mac":"80:64:6F:9E:1A:04","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:47:02.994 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.209","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9E1A04","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:47:02.017 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-12-31T18:47:02","Local":"2023-12-31T19:47:02","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:02"}}
19:47:02.034 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:47:02.053 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-12-31T19:47:02","ENERGY":{"TotalStartTime":"2023-07-06T14:19:25","Total":28.170,"Yesterday":0.005,"Today":0.004,"Power":0,"ApparentPower":7,"ReactivePower":7,"Factor":0.01,"Voltage":250,"Current":0.027}}}
19:47:02.072 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-12-31T19:47:02","Uptime":"0T00:07:37","UptimeSec":457,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Cat","BSSId":"04:70:56:95:73:BA","Channel":6,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
N/A
```

### TO REPRODUCE

```
GET /cm?cmnd=Power%20ON HTTP/1.1
Accept: */*
Accept-Encoding: gzip, deflate, br
Connection: keep-alive
Host: 192.168.1.71
User-Agent: HTTPie/3.2.2

HTTP/1.1 200 OK
Accept-Ranges: none
Cache-Control: no-cache, no-store, must-revalidate
Connection: close
Content-Type: application/json
Expires: -1
Pragma: no-cache
Server: Tasmota/13.3.0 (ESP8285H16)
Transfer-Encoding: chunked

{
    "POWER": "ON"
}
```

```
GET http://192.168.1.71/cm?cmnd=Power%20ON HTTP/1.1
host: 192.168.1.71

File Not Found

URI: http://192.168.1.71/cm
Method: GET
Arguments: 1
 cmnd: Power ON
```

### EXPECTED BEHAVIOUR

I'd expect it to handle those properly

https://developer.mozilla.org/en-US/docs/Web/HTTP/Messages

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-01-01:
I will have a look. But keep in mind that we are using Arduino Core for the web server part, so we rely on it for these features. 

I'm not sure the absolute form is required if you already provide the Host header

barbudor, 2024-01-01:
@arlyon what is your use case?
With which browser and in which circumstances do you get this and need this to be supported? 

As per the standard this is actually mandatory for proxies, and although the standard says that servers are required to support absolute URI, I can't seem to find a case where this would be really needed from a device.
Are you accessing your Tasmotas through a proxy?

Supporting absolute URI is a security risk if the URI points outside of the device itself. Which means that additional checks should be performed to be sure that the URI points to the device itself.
IMHO, 404 seems a not too bad response so far

sfromis, 2024-01-01:
The RFC gives this explanation:
> To allow for transition to the absolute-form for all requests in some future version of HTTP, a server MUST accept the absolute-form in requests, even though HTTP/1.1 clients will only send them in requests to proxies.

Thus, the value of this detail is "pretty limited".