arendst/Tasmota #20371: INA3221 results not shown when no connection to configured MQTT.

toszczyp, 2024-01-02:
Self compiled Tasmota version 13.1.0.2 by Arends (because INA3221) is not in default configuration.
When no connection to MQTT configured in Configuration/Configure MQTT then only   this line is shown on main scrreen "INA3221   Voltage Current Power" - this is header only - no any additional lines with results. 
When connection returns all results are displaied as expected. To reproduce it is enougth to replace MQTT IP address in configuration to any IP with no MQTT in configuration.
Expected behaviour: display INA3221 results if npo connection to MQTT too.

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-01-02:
I confirm that there is no dependencies to the MQTT connection in the code
Docs have been updated regarding command [`Sensor100`](https://tasmota.github.io/docs/Commands/#sensor100)
Please check that you are using the command correctly
More details in the PR: https://github.com/arendst/Tasmota/pull/16786

toszczyp, 2024-01-02:
I found that issue is not connected to INA3221 but to not executed rule.
So now i updated title - Rule not  executed when no connection to configured MQTT.
See in logs attached - here you see that in tasmota_ok.txt  command Sensor100 is executed so all works but when no connection as in tasmota_error.txt  then rule is not executed - so expected behavior is to execute rules  if no connection to mqtt too.

[tasmota_error.txt](https://github.com/arendst/Tasmota/files/13810858/tasmota_error.txt)
[tasmota_ok.txt](https://github.com/arendst/Tasmota/files/13810860/tasmota_ok.txt)

toszczyp, 2024-01-02:
I found that in case https://github.com/arendst/Tasmota/issues/7552
 _[ascillato](https://github.com/ascillato) commented [on Jan 24, 2020](https://github.com/arendst/Tasmota/issues/7552#issuecomment-578304573)
As designed. SYSTEM#BOOT only triggers if mqtt is on. If you want a trigger for an earlier event prior all the initialization, you can use ON POWER1#BOOT DO ...._

so it looks that I have the same case  
_RUL: SYSTEM#BOOT performs "backlog; Sensor100 1, 0.1, 0.1, 0.1; VoltRes 3;WattRes 2"_
finally:  works as designed :-(