arendst/Tasmota #20384: tasmota32-ZbBridge .. ZbInfo only displays a partial list of Zigbee devices

tmoore22, 2024-01-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
ZbInfo only displays a partial list of Zigbee devices (15/25) missing first 10x devices that were registered..

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [y] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [y] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [y] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [y] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [y] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge Pro
- [y] Tasmota binary firmware version number used: 13.3.0
  - [y] Pre-compiled
  - [ ] Self-compiled
- [y] Flashing tools used: espflasher
- [y] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:38:11.030 MQT: stat/tasmota_zbbridge/RESULT = {"NAME":"Sonoff Zigbee Pro","GPIO":[0,0,576,0,480,0,0,0,0,1,1,5792,0,0,0,3552,0,320,5793,3584,0,640,608,32,0,0,0,0,0,1,0,0,0,0,0,0],"FLAG":0,"BASE":1}
14:38:11.229 MQT: stat/tasmota_zbbridge/RESULT = {"Module":{"0":"Sonoff Zigbee Pro"}}
14:38:11.437 MQT: stat/tasmota_zbbridge/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"5792":"Zigbee Rst1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"3552":"Zigbee Tx"},"GPIO20":{"0":"None"},"GPIO21":{"320":"Led_i1"},"GPIO22":{"5793":"Zigbee Rst2"},"GPIO23":{"3584":"Zigbee Rx"},"GPIO24":{"0":"None"},"GPIO25":{"640":"I2C SDA1"},"GPIO26":{"608":"I2C SCL1"},"GPIO27":{"32":"Button1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:33:41.935 CMD: Status 0
14:33:41.942 MQT: stat/tasmota_zbbridge/STATUS = {"Status":{"Module":0,"DeviceName":"ZbBridge Hub","FriendlyName":["ZbBridge Hub"],"Topic":"tasmota_zbbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:33:42.008 MQT: stat/tasmota_zbbridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-zbbrdgpro.bin","RestartReason":"Software reset CPU","Uptime":"3T02:30:04","StartupUTC":"2023-12-31T12:03:38","Sleep":0,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2023-12-24T23:28:11","SaveCount":135}}
14:33:42.071 MQT: stat/tasmota_zbbridge/STATUS2 = {"StatusFWR":{"Version":"13.3.0(zbbrdgpro)","BuildDateTime":"2023-12-12T14:31:25","Core":"2_0_14","SDK":"4.4.6.231122","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"421/699"}}
14:33:42.125 MQT: stat/tasmota_zbbridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HeavyCloud",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000081","40006082","00004000","00000000"]}}
14:33:42.191 MQT: stat/tasmota_zbbridge/STATUS4 = {"StatusMEM":{"ProgramSize":1465,"Free":582,"Heap":139,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","8F8A87CF","00548001","000000CF","010013C0","C0000981","40024080","00200000","5400082C","00000080","00000000"],"Drivers":"1,2,3,4,5,8,9,10,12,16,20,21,23,24,27,29,35,38,41,50,52,59,62,68,82","Sensors":"1,2,3,5,6,127","I2CDriver":"7"}}
14:33:42.256 MQT: stat/tasmota_zbbridge/STATUS5 = {"StatusNET":{"Hostname":"tasmota-zbbridge-7180","IPAddress":"192.168.0.222","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.254","DNSServer2":"0.0.0.0","Mac":"E4:65:B8:0F:7C:0C","IP6Global":"","IP6Local":"fe80::e665:b8ff:fe0f:7c0c%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.0.254","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
14:33:42.324 MQT: stat/tasmota_zbbridge/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.0.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F7C0C","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:33:42.380 MQT: stat/tasmota_zbbridge/STATUS7 = {"StatusTIM":{"UTC":"2024-01-03T14:33:42","Local":"2024-01-03T14:33:42","StartDST":"2024-03-31T01:00:00","EndDST":"2024-10-27T01:00:00","Timezone":99,"Sunrise":"08:24","Sunset":"16:06"}}
14:33:42.438 MQT: stat/tasmota_zbbridge/STATUS10 = {"StatusSNS":{"Time":"2024-01-03T14:33:42"}}
14:33:42.489 MQT: stat/tasmota_zbbridge/STATUS11 = {"StatusSTS":{"Time":"2024-01-03T14:33:42","Uptime":"3T02:30:04","UptimeSec":268204,"Heap":138,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":34},"Wifi":{"AP":1,"SSId":"HeavyCloud","BSSId":"B0:BE:76:59:51:7C","Channel":1,"Mode":"11n","RSSI":100,"Signal":-32,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
14:29:43.139 MQT: tele/tasmota_zbbridge/front_door_sensor/SENSOR = {"ZbInfo":{"front_door_sensor":{"Device":"0x8163","Name":"front_door_sensor","IEEEAddr":"0x00124B000F02A699","ModelId":"DWS003","Manufacturer":"HiveHome.com","Endpoints":[6],"Config":["A06.2","T06"],"ZoneType":21,"ZoneStatus":12800,"Temperature":13.5,"Reachable":true,"BatteryPercentage":25,"BatteryLastSeenEpoch":1704206345,"LastSeen":332,"LastSeenEpoch":1704291851,"LinkQuality":109}}}
14:29:43.205 MQT: tele/tasmota_zbbridge/portche_light/SENSOR = {"ZbInfo":{"portche_light":{"Device":"0xDD0C","Name":"portche_light","IEEEAddr":"0xA4C138BB39AC2AEC","ModelId":"TS0011","Manufacturer":"_TZ3000_hhiodade","Endpoints":[1],"Config":["O01","L01.1"],"Power":0,"Reachable":true,"LastSeen":113,"LastSeenEpoch":1704292070,"LinkQuality":109}}}
14:29:43.267 MQT: tele/tasmota_zbbridge/bathroom_sensor/SENSOR = {"ZbInfo":{"bathroom_sensor":{"Device":"0x4038","Name":"bathroom_sensor","IEEEAddr":"0x00124B000E36AACD","ModelId":"MOT003","Manufacturer":"HiveHome.com","Endpoints":[6],"Config":["A06.1","T06","I06"],"ZoneType":13,"ZoneStatus":12592,"Temperature":24.5,"Occupancy":0,"Reachable":true,"BatteryPercentage":90,"BatteryLastSeenEpoch":1704220255,"LastSeen":577,"LastSeenEpoch":1704291606,"LinkQuality":83}}}
14:29:43.335 MQT: tele/tasmota_zbbridge/bathroom_door_sensor/SENSOR = {"ZbInfo":{"bathroom_door_sensor":{"Device":"0x4E0F","Name":"bathroom_door_sensor","IEEEAddr":"0x00124B000F02A116","ModelId":"DWS003","Manufacturer":"HiveHome.com","Endpoints":[6],"Config":["A06.2","T06"],"ZoneType":21,"ZoneStatus":12801,"Temperature":24.75,"Reachable":true,"BatteryPercentage":66,"BatteryLastSeenEpoch":1704221427,"LastSeen":684,"LastSeenEpoch":1704291499,"LinkQuality":138}}}
14:29:43.399 MQT: tele/tasmota_zbbridge/bathroom_light/SENSOR = {"ZbInfo":{"bathroom_light":{"Device":"0x3957","Name":"bathroom_light","IEEEAddr":"0x00158D000172232F","ModelId":"FWBulb01","Manufacturer":"Aurora","Endpoints":[1],"Config":["O01","L01.1"],"Power":1,"Dimmer":254,"Reachable":true,"LastSeen":576,"LastSeenEpoch":1704291607,"LinkQuality":83}}}
[... cut ...]
_Steps to reproduce the behavior:_
ZbInfo ....

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Listing ALL Zigbee devices, not just a partial list (as described in tasmota docs)
https://tasmota.github.io/docs/Zigbee/#advanced-topic-device-information

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-01-03:
It looks like you exhausted the size of the console output. ZbInfo does show all devices, but the console buffer is only 6KB on ESP32, which is exactly the size of the first block.

tmoore22, 2024-01-03:
> If you check what you get on a MQTT subscription, you may find all the devices there. Or maybe capture logging via syslog.

its not a problem, they are all listed on the tasmota homescreen ... although I wanted to raise the issue because in the tasmota documentation it says "ALL" devices. Maybe its worth just mentioning that it applied only up to the console limit.
[image]

s-hadinger, 2024-01-03:
Not exactly. `ZbInfo` does show all the devices as per documentation. It's just that the console buffer is not large enough to hold them all. But if you look at serial, mqtt or syslog, they are all there.

That's precisely why I implemented `ZbInfo <index>` so you can iterate manually if the number gets too large.

sfromis, 2024-01-03:
You could also set up some Berry code to capture the log output, and do what you like with it. Just don't print it while capturing, as this will cause a loop.

kelsotowers, 2024-01-09:
Hope you don’t mind me asking a newbie question of people with knowledge of Tamara on a Sonoff zigbee bridge pro? I’m struggling to flash mine - can’t even read the flash ID so I’m not getting it into flash mode. I’m powering it with 3.3V via the header and holding down gpio 0 when I power it on. To prove my setup, a sonoff basic goes into flash mode straight away. Can anyone offer any advice please?

sfromis, 2024-01-09:
Typical issue is that you are using a weak USB/serial adapter not able to power the ESP32.
https://tasmota.github.io/docs/Getting-Started/#serial-programmer

kelsotowers, 2024-01-09:
Thank you for your reply. My setup uses an external 3.3V supply and I set it current limited at 0.6A - and the max current I see drawn at any time is 120mA. Also it won’t even read the flash_id so I don’t think it’s that? Unless 0.6A isn’t enough, easy to increase. I will read around the subject - to be honest I only just realised it was an ESP32, first one I’ve had. Most of my devices are Sonoff S20 or Shelly1

tmoore22, 2024-01-09:
@kelsotowers the only way I could get mine to flash was by using espflasher, other flash tools would not work for me.

Jason2866, 2024-01-09:
Try a different USB cable for your USB serial adapter if you don't have it directly connected to the PC. It needs a good cable here. Most of the time thicker ones are better.
A good setup can flash every esp32x variant with a speed of `1500000`

arendst, 2024-01-10:
Pls do not hijack this now closed issue.

Original answer is exhausted console buffer.

closed 2024-01-10 by arendst