arendst/Tasmota #20387: Wifi password with special characters fails to connect

cmonty14, 2024-01-03:
### PROBLEM DESCRIPTION
I have received a new router and started migration all wifi clients to this new router.
With my old router I used this password:
`XahiRari4[Vadd`
I can share this password here because the wifi is disabled.
As you can see this password includes 1 special character.
There were no issues with any Tasmota device connected to this wifi.

However there are major issues with new router when using specials characters in password phrase.

I'm [aware](https://tasmota.github.io/docs/FAQ/#cannot-connect-to-wi-fi) that wifi password with special characters won't work with Tasmota. However it worked with old router, and I was expecting that is works with new router, too.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund EP2
- [x] Tasmota binary firmware version number used: 13.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
  tasmota/Gosund-EP2-03/RESULT = {"NAME":"Gosund EP2","GPIO":[320,1,576,1,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
  tasmota/Gosund-EP2-03/RESULT = {"Module":{"0":"Gosund EP2"}}
  tasmota/Gosund-EP2-03/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:57:04.737 MQT: tasmota/Gosund-EP2-03/STATUS = {"Status":{"Module":0,"DeviceName":"Gosund-EP2-03","FriendlyName":["Gosund EP2 Gerät 03"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:57:04.744 MQT: tasmota/Gosund-EP2-03/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"External System","Uptime":"0T01:43:21","StartupUTC":"2024-01-03T16:13:43","Sleep":50,"CfgHolder":4617,"BootCount":283,"BCResetTime":"2022-12-22T09:11:59","SaveCount":3470,"SaveAddress":"F4000"}}
18:57:04.748 MQT: tasmota/Gosund-EP2-03/STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota)","BuildDateTime":"2023.12.12 14:33:12","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"420/699"}}
18:57:04.753 MQT: tasmota/Gosund-EP2-03/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["babbelnet_nomap","wlan68753SHN_nomap"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
18:57:04.766 MQT: tasmota/Gosund-EP2-03/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["0407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
18:57:04.772 MQT: tasmota/Gosund-EP2-03/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3489","IPAddress":"192.168.1.33","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"24:62:AB:34:CD:A1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:57:04.778 MQT: tasmota/Gosund-EP2-03/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.20.5","MqttPort":1883,"MqttClientMask":"tasmota-3489","MqttClient":"tasmota-3489","MqttUser":"loxberry","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:57:04.785 MQT: tasmota/Gosund-EP2-03/STATUS7 = {"StatusTIM":{"UTC":"2024-01-03T17:57:04","Local":"2024-01-03T18:57:04","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:05"}}
18:57:04.789 MQT: tasmota/Gosund-EP2-03/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:57:04.798 MQT: tasmota/Gosund-EP2-03/STATUS10 = {"StatusSNS":{"Time":"2024-01-03T18:57:04","ENERGY":{"TotalStartTime":"2022-12-22T09:11:59","Total":0.330,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
18:57:04.804 MQT: tasmota/Gosund-EP2-03/STATUS11 = {"StatusSTS":{"Time":"2024-01-03T18:57:04","Uptime":"0T01:43:21","UptimeSec":6201,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"babbelnet_nomap","BSSId":"04:B4:FE:2C:95:78","Channel":1,"Mode":"11n","RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:12"}}}
18:57:12.435 MQT: tasmota/Gosund-EP2-03/STATE = {"Time":"2024-01-03T18:57:12","Uptime":"0T01:43:29","UptimeSec":6209,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"babbelnet_nomap","BSSId":"04:B4:FE:2C:95:78","Channel":1,"Mode":"11n","RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:12"}}
18:57:12.448 MQT: tasmota/Gosund-EP2-03/SENSOR = {"Time":"2024-01-03T18:57:12","ENERGY":{"TotalStartTime":"2022-12-22T09:11:59","Total":0.330,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Configuration
2. Configure WiFi
3. WiFi parameters
4. WiFi Password

### EXPECTED BEHAVIOUR
If wifi connections works with old router using special characters it should work with new rooter, too.

sfromis, 2024-01-03:
Yes, docs indeed stipulate no special chars in passwords, meaning no "safely" predictable outcome when stepping outside. Different tools and encodings means that the outcome depends on details, including how you input the passwords into Tasmota.

As long as you are veering outside of the docs, what you see is by definition no "bug", and while Tasmota does not validate password for adherence to the stipulation, you should expect risks of trouble when using other passwords. And the trouble zones both depends on what special chars are used, and on versions of Tasmota and the Arduino Core.

cmonty14, 2024-01-03:
I'm sorry but I don't fully understand your reponse.

Is usage of special characters in wifi passwords possible or not?
If it's possible, must I set the password using console?

sfromis, 2024-01-03:
Docs says that "Do not use special characters or white spaces in the password". It works if following that.

If you do not want to follow that, you're basically on your own experimenting with different methods, and see what works at the moment. You cannot be sure that what you may have found out will also work next year.

But, as I wrote, no validation happens, so you can "often" get something to work, also without you having any guidelines about how to not follow the docs. Using commands or build time defaults does have less zones of risk for how characters are handled.

cmonty14, 2024-01-04:
Based on my tests the number of characters is relevant, too.
Means +14 characters is not working.

sfromis, 2024-01-04:
I had no trouble connecting to a test AP with the 26-character password `abcdefghijklmnopqrstuvwxyz`

closed 2024-01-04 by Jason2866