arendst/Tasmota #20435: Tasmota ui toggle button does not send "cmnd" command to console

hotmtk, 2024-01-08:
### PROBLEM DESCRIPTION
Tasmota UI Toggle button does not send "cmnd" topic to console. It sends only "stat" topic

My network consists of Sonff Mini R2 with Tasmota firmware, Mosquitto Mqtt Broker, and one Iot MQTT Panel - as Android client
Everything is set up and works near perfect. When I Publish from Broker and from Android client, the Relay switches On an Off.
But when I Toggle the button from UI of Tasmota Sonoff Switch, it doesn't send to the Broker "cmnd/tasmota_9E59DD/POWER" topic, it sends only "stat/tasmota_9E59DD/POWER".
Therefore neither the Broker is receiving the cmnd topic to forward it to Android client, and to modify the correct state of On/OFF virtual button inside it.  

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): SonOff Mini R2
- [x ] Tasmota binary firmware version number used: Sonoff MINIR2 Tasmota Program Version	10.0.0(lite) Build Date & Time	2021-10-9T07:55:32
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: [https://github.com/njh/sonoff-ota-flash-cli]
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
- 22:19:59.539 CMD: Backlog Template; Module; GPIO 255
22:19:59.561 MQT: stat/tasmota_9E59DD/RESULT = {"NAME":"Sonoff MINIR2","GPIO":[32,0,0,0,160,0,0,0,224,544,0,0,0,0],"FLAG":0,"BASE":1}
22:19:59.787 MQT: stat/tasmota_9E59DD/RESULT = {"Module":{"0":"Sonoff MINIR2"}}
22:20:00.040 MQT: stat/tasmota_9E59DD/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"544":"LedLink"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:33:26.525 CMD: cmnd/tasmota_9E59DD/Status 0
22:33:26.532 MQT: stat/tasmota_9E59DD/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9E59DD","ButtonTopic":"tasmota_9E59DD","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":1}}
22:33:26.539 MQT: stat/tasmota_9E59DD/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:54:15","StartupUTC":"2024-01-08T20:39:11","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2024-01-04T15:58:02","SaveCount":483,"SaveAddress":"F4000"}}
22:33:26.545 MQT: stat/tasmota_9E59DD/STATUS2 = {"StatusFWR":{"Version":"10.0.0(lite)","BuildDateTime":"2021-10-19T07:55:32","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"382/699"}}
22:33:26.549 MQT: stat/tasmota_9E59DD/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SamyGO","NPP"],"TelePeriod":600,"Resolution":"558180C0","SetOption":["000080E9","2805C8000100060000005A00000000000000","00000080","00006000","00004080"]}}
22:33:26.559 MQT: stat/tasmota_9E59DD/STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F000783","04108001","00000006","010001C0","00000000","00004004","00000000","00000020"],"Drivers":"1,2,3,4,9,10,12,16,20,21,37","Sensors":"3"}}
22:33:26.567 MQT: stat/tasmota_9E59DD/STATUS5 = {"StatusNET":{"Hostname":"tasmota-9E59DD-6621","IPAddress":"192.168.5.153","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.5.5","DNSServer2":"192.168.5.3","Mac":"9C:9C:1F:9E:59:DD","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:33:26.572 MQT: stat/tasmota_9E59DD/STATUS6 = {"StatusMQT":{"MqttHost":"hotmtk.go.ro","MqttPort":11833,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9E59DD","MqttUser":"MQTT_DEV_9E59DD","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:33:26.578 MQT: stat/tasmota_9E59DD/STATUS7 = {"StatusTIM":{"UTC":"2024-01-08T21:33:26","Local":"2024-01-08T22:33:26","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:42","Sunset":"17:10"}}
22:33:26.583 MQT: stat/tasmota_9E59DD/STATUS10 = {"StatusSNS":{"Time":"2024-01-08T22:33:26","Switch1":"ON"}}
22:33:26.589 MQT: stat/tasmota_9E59DD/STATUS11 = {"StatusSTS":{"Time":"2024-01-08T22:33:26","Uptime":"0T00:54:15","UptimeSec":3255,"Vcc":3.495,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":2,"SSId":"NPP","BSSId":"04:18:D6:B4:E8:40","Channel":11,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Use a Tasmota switch with Tasmota firmware on it, Mosquitto MQTT Broker, and Iot MQTT Panel - as Android client. Set up all to work together.
Using web ui toggle button, does not change the state of Android virtual button, as it not receiving cmnd topic, just only stat.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Here is Mosquitto log:

Here I push virtual button from IoT Android Mqtt application:
1704720823: Received PUBLISH from IotMQTTPanel_mobil (d0, q0, r1, m0, 'cmnd/tasmota_9E59DD/POWER', ... (3 bytes))
1704720823: Sending PUBLISH to IotMQTTPanel_mobil (d0, q0, r0, m0, 'cmnd/tasmota_9E59DD/POWER', ... (3 bytes))
1704720823: Sending PUBLISH to DVES_9E59DD (d0, q0, r0, m0, 'cmnd/tasmota_9E59DD/POWER', ... (3 bytes))
1704720823: Received PUBLISH from DVES_9E59DD (d0, q0, r0, m0, 'stat/tasmota_9E59DD/RESULT', ... (15 bytes))
1704720823: Received PUBLISH from DVES_9E59DD (d0, q0, r1, m0, 'stat/tasmota_9E59DD/POWER', ... (3 bytes))

Here I push toggle button from tasmota web ui:
1704720844: Received PUBLISH from DVES_9E59DD (d0, q0, r0, m0, 'stat/tasmota_9E59DD/RESULT', ... (15 bytes))
1704720844: Received PUBLISH from DVES_9E59DD (d0, q0, r1, m0, 'stat/tasmota_9E59DD/POWER', ... (3 bytes))

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-01-08:
Then you have not understood how Tasmota works

Working as expected

a `cmmd/` topic is an INPUT to tasmota so the external world can send a COMMAND to Tasmota such as `Power 1`
a 'stat/`topic is an OUTPUT from Tasmota so the extenal workls is made aware of a change in Tasmotat state or a response to a command

When you press the web UI button it  is normal that the message send is a `stat/` and not a `cmnd/`

The Android client should update it's state ONLY on the reception os hte `stat/` message (even if it issued the `cmnd/) itself

Working as designed

closed 2024-01-09 by Jason2866

barbudor, 2024-01-09:
And by the way, I have used IotMQTTPanel for more than a couple of years using the above "rule"
Working like a charm

I just uninstalled it after a recent update asking me to agree on much too many cookies with no option to reject.
I just can accept such behavior from an app