arendst/Tasmota #20450: Tasmota cannot connect to wireless network named "0"

kcsoft, 2024-01-09:
### PROBLEM DESCRIPTION
Tasmota cannot connect to wifi SSID named "0". (character zero)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: 13.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Try to connect to a wireless network named "0" (character zero, no quotes).

### EXPECTED BEHAVIOUR
It should connect regardless of the SSID name.

### SCREENSHOTS
[image]

When running `ssid1 0` in console:
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-01-09:
Is it really worth spending troubleshooting time in this case? Moreover, you didn't provide logs as requested.

sfromis, 2024-01-09:
Docs updated to mention that the SSID command will clear the value if the input was "0" (along with "1" being reset to default from build).

Jason2866, 2024-01-09:
Thx for pointing at. Documentation has been updated.

closed 2024-01-09 by Jason2866