arendst/Tasmota #20461: Sonoff ZBPro with ESP32 random crash

tasict, 2024-01-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sonoff ZBPro with ESP32 random crash when enter to main menu page.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

11:30:48.654 MQT: tele/ZBPro/INFO1 = {"Info1":{"Module":"Sonoff Zigbee Pro","Version":"13.3.0.2(zbbrdgpro)","FallbackTopic":"cmnd/DVES_607D04_fb/","GroupTopic":"cmnd/tasmotas/"}}
11:30:48.681 MQT: tele/ZBPro/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"ZBPro","IPAddress":"10.1.8.25","IP6Global":"","IP6Local":"-"}}
11:30:48.707 MQT: tele/ZBPro/INFO3 = {"Info3":{"RestartReason":{"Exception":6,"Reason":"IntegerDivideByZero","EPC":"40102f6f","EXCVADDR":"00000000","CallChain":["40102f6c","40122668","401085bd","4014649c","401407db","40103451","400fe31a","400fffa5","40101fcd","4013c729","401085bd","401465ba","40152069","400e4e2b"]},"BootCount":261}}

- [O ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [O] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [O ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [O ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [O ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [O ] Device used (e.g., Sonoff Basic): _Sonoff ZB Pro____
- [O ] Tasmota binary firmware version number used: _13.3.0.2____
  - [ O] Pre-compiled
  - [ ] Self-compiled
- [O] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:41:21.283 MQT: stat/ZBPro/RESULT = {"NAME":"Sonoff Zigbee Pro","GPIO":[0,0,576,0,480,0,0,0,0,1,1,5792,0,0,0,3552,0,320,5793,3584,0,640,608,32,0,0,0,0,0,1,0,0,0,0,0,0],"FLAG":0,"BASE":1}
11:41:21.482 MQT: stat/ZBPro/RESULT = {"Module":{"0":"Sonoff Zigbee Pro"}}
11:41:21.690 MQT: stat/ZBPro/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"5792":"Zigbee Rst1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"3552":"Zigbee Tx"},"GPIO20":{"0":"None"},"GPIO21":{"320":"Led_i1"},"GPIO22":{"5793":"Zigbee Rst2"},"GPIO23":{"3584":"Zigbee Rx"},"GPIO24":{"0":"None"},"GPIO25":{"640":"I2C SDA1"},"GPIO26":{"608":"I2C SCL1"},"GPIO27":{"32":"Button1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
11:41

```
- [O ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:42:07.476 MQT: stat/ZBPro/STATUS = {"Status":{"Module":0,"DeviceName":"ZBPro","FriendlyName":["ZBPro"],"Topic":"ZBPro","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:42:07.499 MQT: stat/ZBPro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/tasmota32-zbbrdgpro.bin","RestartReason":"Software reset CPU","Uptime":"0T00:11:24","StartupUTC":"2024-01-11T03:30:43","Sleep":0,"CfgHolder":4617,"BootCount":261,"BCResetTime":"2023-11-11T09:50:51","SaveCount":493}}
11:42:07.521 MQT: stat/ZBPro/STATUS2 = {"StatusFWR":{"Version":"13.3.0.2(zbbrdgpro)","BuildDateTime":"2024-01-03T15:16:41","Core":"2_0_14","SDK":"4.4.6.231122","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"406/699"}}
11:42:07.540 MQT: stat/ZBPro/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NO46-2.4G","NO46"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000695A0A192800000000","00000080","40006082","00004000","00000000"]}}
11:42:07.567 MQT: stat/ZBPro/STATUS4 = {"StatusMEM":{"ProgramSize":1471,"Free":384,"Heap":129,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","8F8A87CF","00548001","000000CF","010013C0","C0000981","40024080","00200000","5400082C","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!8,9,10,12,!16,!20,!21,23,24,!27,29,!35,38,!41,50,52,!59,62,!68,82","Sensors":"1,2,3,5,6,127","I2CDriver":"7"}}
11:42:07.591 MQT: stat/ZBPro/STATUS5 = {"StatusNET":{"Hostname":"ZBPro","IPAddress":"10.1.8.25","Gateway":"10.1.8.1","Subnetmask":"255.255.252.0","DNSServer1":"10.1.8.1","DNSServer2":"8.8.8.8","Mac":"40:22:D8:60:7D:04","IP6Global":"fd00:88::4222:d8ff:fe60:7d04","IP6Local":"fe80::4222:d8ff:fe60:7d04%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.1.8.1","DNSServer2":"8.8.8.8","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
11:42:07.609 MQT: stat/ZBPro/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.ctsite.tk","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_607D04","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":15}}
11:42:07.629 MQT: stat/ZBPro/STATUS7 = {"StatusTIM":{"UTC":"2024-01-11T03:42:07Z","Local":"2024-01-11T11:42:07","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+08:00","Sunrise":"06:40","Sunset":"17:21"}}
11:42:07.647 MQT: stat/ZBPro/STATUS10 = {"StatusSNS":{"Time":"2024-01-11T11:42:07"}}
11:42:07.667 MQT: stat/ZBPro/STATUS11 = {"StatusSTS":{"Time":"2024-01-11T11:42:07","Uptime":"0T00:11:24","UptimeSec":684,"Heap":125,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Berry":{"HeapUsed":16,"Objects":241},"Wifi":{"AP":2,"SSId":"NO46","BSSId":"E4:38:83:8C:F8:AE","Channel":11,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:42:07.688 MQT: stat/ZBPro/STATUS12 = {"StatusSTK":{"Exception":6,"Reason":"IntegerDivideByZero","EPC":"40102f6f","EXCVADDR":"00000000","CallChain":["40102f6c","40122668","401085bd","4014649c","401407db","40103451","400fe31a","400fffa5","40101fcd","4013c729","401085bd","401465ba","40152069","400e4e2b"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

tasict, 2024-01-11:
In fact, I once tried to resolve this issue by removing all paired devices, but it is still possible to encounter this message. Therefore, I don't believe it is a problem specific to individual devices.

s-hadinger, 2024-01-11:
Unless you have some special codes or modules running, I don't see what could cause a "division by zero" exception when displaying the main page. Did you compile yourself? If so do you have the map file?

If not, we can try to flash it with a new version and use the map file to find the exact location of the crash

tasict, 2024-01-11:
Yes, that's exactly what I need. I'm using the official OTA server with daily build of the DEV branch. How can I find the map file?

s-hadinger, 2024-01-11:
They are here https://github.com/tasmota/install/tree/firmware/firmware/map

Jason2866, 2024-01-11:
Random exceptions are typical symptoms for hardware issue (most of the time power supply). If the same firmware is running without an issue on an identic second device it is most probably a hardware defect.

sfromis, 2024-01-11:
I'm occasionally seeing that exception on another device, which is basically sitting around doing nothing. I've not really tried to look into what happens, but my base assumption is that hardware, potentially power, plays a role.

s-hadinger, 2024-01-11:
That's at least 3 indépendant reports of a similar crash for division by zero. I would really be curious to check the map file to understand where it happens

Jason2866, 2024-01-11:
@tasict Can you try this build?
[tasmota32-zbbrdgpro-ard3.bin.zip](https://github.com/arendst/Tasmota/files/13907492/tasmota32-zbbrdgpro-ard3.bin.zip)
 Compiled from actual development branch with Tasmota core 3.0.0.
Extract the zip before upload the file via file upgrade!
Since the error is very seldom and strange it could be a compiler toolchain bug. -> Core 3.0.0 uses a newer toolchain.

tasict, 2024-01-12:
> @tasict Can you try this build? [tasmota32-zbbrdgpro-ard3.bin.zip](https://github.com/arendst/Tasmota/files/13907492/tasmota32-zbbrdgpro-ard3.bin.zip) Compiled from actual development branch with [...]

Of course, thank you very much for your assistance. I have updated the firmware you provided, and initially, it seems that the crash issue has not occurred. However, since this problem occurs randomly, I will continue to observe for a while and report the results back. Thanks again for your help.

tasict, 2024-01-12:
@Jason2866 Unfortunately, after five hours, it happened again. The log after the reboot is as follows:
-
00:00:00.002 HDW: ESP32-D0WD-V3 v3.0 
00:00:00.303 UFS: FlashFS mounted with 1240 kB free
00:00:00.330 CFG: Loaded from File, Count 611
00:00:00.338 QPC: Count 1
00:00:00.341 I2C: Bus1 using GPIO26(SCL) and GPIO25(SDA)
00:00:00.580 BRY: Berry initialized, RAM used 3964 bytes
00:00:00.652 Project tasmota - ZBPro Version 13.3.0.2(zbbrdgpro)-3_0_0(2024-01-11T19:15:50)
00:00:02.683 BRY: Successfully loaded 'autoexec.be'
00:00:04.489 WIF: Connecting to AP2 NO46 Channel 11 BSSId E4:38:83:8C:F8:AE in mode 11n as ZBPro...
00:00:07.049 WIF: Connected
00:00:07.305 HTP: Web server active on ZBPro with IP address 10.1.8.25
12:28:03.298 MQT: Attempting connection...
12:28:03.348 MQT: Connected
12:28:03.351 MQT: tele/ZBPro/LWT = Online (retained)
12:28:03.353 MQT: cmnd/ZBPro/POWER = 
12:28:03.360 MQT: tele/ZBPro/INFO1 = {"Info1":{"Module":"Sonoff Zigbee Pro","Version":"13.3.0.2(zbbrdgpro)","FallbackTopic":"cmnd/DVES_607D04_fb/","GroupTopic":"cmnd/tasmotas/"}}
12:28:03.383 MQT: tele/ZBPro/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"ZBPro","IPAddress":"10.1.8.25","IP6Global":"","IP6Local":"fe80::4222:d8ff:fe60:7d04%st1"}}
12:28:03.407 MQT: tele/ZBPro/INFO3 = {"Info3":{"RestartReason":{"Exception":6,"Reason":"IntegerDivideByZero","EPC":"40128639","EXCVADDR":"00000000","CallChain":["40128636","4010b5a9","40125481","400de9f1","400e71ba","40122ead","4012be72","4012df4d","4012eb3f","400e8c22","40125481","400de9d1","400d93c1","4014ce27"]},"BootCount":376}}
12:28:05.701 QPC: Reset
12:28:07.701 MQT: tele/ZBPro/STATE = {"Time":"2024-01-12T12:28:07","Uptime":"0T00:00:10","UptimeSec":10,"Heap":135,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":430,"MqttCount":1,"Berry":{"HeapUsed":15,"Objects":214},"Wifi":{"AP":2,"SSId":"NO46","BSSId":"E4:38:83:8C:F8:AE","Channel":11,"Mode":"11n","RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:00:04"}}
12:28:10.815 ZIG: rebooting ZNP device
12:28:13.054 MQT: tele/ZBPro/RESULT = {"ZbState":{"Status":1,"Message":"CCxxxx ZNP booted","RestartReason":"Power-up","MajorRel":2,"MinorRel":7}}
12:28:13.181 MQT: tele/ZBPro/RESULT = {"ZbState":{"Status":50,"MajorRel":2,"MinorRel":7,"MaintRel":1,"Revision":20230507}}
12:28:13.256 MQT: tele/ZBPro/RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
12:28:15.870 MQT: tele/ZBPro/RESULT = {"ZbState":{"Status":40,"NewState":9,"Message":"Started as coordinator"}}
12:28:15.895 MQT: tele/ZBPro/RESULT = {"ZbState":{"Status":51,"IEEEAddr":"0x00124B00257B1318","ShortAddr":"0x0000","DeviceType":7,"DeviceState":9,"NumAssocDevices":4,"AssocDevicesList":["0x1D72","0xCAD1","0xF819","0x778E"]}}
12:28:15.967 MQT: tele/ZBPro/RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
12:28:15.978 ZIG: Zigbee started

s-hadinger, 2024-01-12:
@Jason2866 can you share the corresponding map file?

Jason2866, 2024-01-12:
@s-hadinger 
[tasmota32-zbbrdgpro-ard3.map.gz](https://github.com/arendst/Tasmota/files/13917335/tasmota32-zbbrdgpro-ard3.map.gz)

s-hadinger, 2024-01-12:
Ha ha. The crash is in `void ZigbeeShow(bool json)` and the only division that can go wrong is:

`uint32_t ct_k = (((1000000 / light.getCT()) + 25) / 50) * 50;`

CT should not be zero, it's always between 153 and 500, but there may be some corner cases.

It will fix this.

PR #20481 (2024-01-12): Fix random crash in Zigbee main page
## Description:

**Related issue (if applicable):** fixes #20461

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.5
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.14
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement [... cut]
closed 2024-01-12 by s-hadinger