arendst/Tasmota #20498: When define USE_ADC_VCC compilation failed

kofec, 2024-01-15:
### PROBLEM DESCRIPTION
Today [15-01-2024] I was trying to compile development branch on gitpod with option ennabled "#define USE_ADC_VCC " and it failed:
`/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x20): undefined reference to `AdcPin(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x24): undefined reference to `AdcRead1(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x28): undefined reference to `AdcRange()'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x83): undefined reference to `AdcPin(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x91): undefined reference to `AdcRead1(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x99): undefined reference to `AdcRange()'` 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
`/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x20): undefined reference to `AdcPin(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x24): undefined reference to `AdcRead1(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x28): undefined reference to `AdcRange()'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x83): undefined reference to `AdcPin(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x91): undefined reference to `AdcRead1(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x99): undefined reference to `AdcRange()'` 

```

### TO REPRODUCE
Start compile on gitpod with enable "#define USE_ADC_VCC ": 

### EXPECTED BEHAVIOUR
Compile will be successful as it is when disable/remove this option:
Environment            Status    Duration
---------------------  --------  ------------
tasmota-rangeextender  SUCCESS   00:00:28.912

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-01-15:
Have you undefined USE_ADC?
Both are exclusive

kofec, 2024-01-15:
I just tired:
```
#ifdef USE_ADC
  #undef USE_ADC
#endif
#define USE_ADC_VCC                              // Display Vcc in Power status. Disable for use as Analog input on selected devices
```
And still the same:
```
Linking .pio/build/tasmota-rangeextender/firmware.elf
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x20): undefined reference to `AdcPin(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x24): undefined reference to `AdcRead1(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x28): undefined reference to `AdcRange()'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x83): undefined reference to `AdcPin(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x91): undefined reference to `AdcRead1(unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota-rangeextender/src/tasmota.ino.cpp.o:(.text._Z13GVMonitorTaskv+0x99): undefined reference to `AdcRange()'
collect2: error: ld returned 1 exit status
*** [.pio/build/tasmota-rangeextender/firmware.elf] Error 1
```

arendst, 2024-01-15:
I think the problem is latest addition of gpioviewer. I will fix this asap.

arendst, 2024-01-15:
Fixed in latest dev branch

kofec, 2024-01-15:
I confirmed:
```
Environment            Status    Duration
---------------------  --------  ------------
tasmota-rangeextender  SUCCESS   00:00:29.059
```