arendst/Tasmota #20499: Tasmota confuses comma and period.

HofeBY, 2024-01-15:
### PROBLEM DESCRIPTION
In my understanding of mathematics it is like this,
that numbers are before and fractions are after the decimal point.
For us, the point is only good for identifying the 1000 stamps.
At least that's how I see it and so does my calculator.

I have included an example of the errors in the picture and the corresponding configuration files.
The error point instead of comma when transmitting the sensor values is also visible in the log of the mqtty transmission.
Image TM13.3.0.2.main-http.jpg

In my case, the mqtty values are incorrectly transmitted to IOBroker, but then corrected there and with a comma
displayed. There is also a picture of this in-IOBroker.jpg.

As soon as you read sensor values into a variable, the point is also used instead of the comma.
This can also be seen with appropriate output on the http interface.
Now I write the values from variables into a local CSV file on my SML meter reader every day at the beginning of the day,
which would be best structured so that columns are written with semicolons and numerical values with commas.
Then you can easily read such a CSV file into Excel/OpenOffice Calc etc.
But here too, the wrong period must first be replaced with a comma.

I use a self-compiled developer version of "arendst" with "Script",
so that it also runs on new and older ESP8285/8266.
It would then be logical in a corrected version that you would have to adapt the variable definition in the script.

It wouldn't be bad for simple data processing if this could be corrected.

Thank you for your attention and perhaps for correcting this error.
Otherwise, great project and thanks to the developers

HofeBY

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): AZ-Delivery ES32 Dev KitC V4
- [ ] Tasmota binary firmware version number used: 13.3.0.2 and also 13.3.0.3
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [x ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
No rule support included in binary

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
LOGfile:
00:00:00.003 HDW: ESP32-D0WDQ6 v1.0 
00:00:00.125 UFS: FlashFS mounted with 288 kB free
00:00:00.162 CFG: Loaded from File, Anzahl 1091
00:00:00.172 QPC: Count 1
00:00:00.178 I2C: Bus1 using GPIO22(SCL) and GPIO21(SDA)
00:00:00.179 SCR: ufilesystem found
00:00:00.266 SCR: nv=34, tv=4, vns=152, vmem=710, smem=8192
00:00:00.268 I2C: DS3231 found at 0x68
00:00:00.273 RTC: UTC 2024-01-14T23:35:31Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
[... 237 lines cut ...]
;Diese Zeilen werden auf http Seite dargestellt.
http here is shown the missuse of Point instead of Comma 
AMT681-Tagesbezug %0day%.%0month%.%0year%:  {m} %4AtD% kWh
AMT681-Tagesbezug gestern:  {m} %4AtG% kWh
AMT681-Tageslieferung %0day%.%0month%.%0year%:  {m} %4AlD% kWh
AMT681-Tageslieferung gestern:  {m} %4AlG% kWh

>M 1
+1,2,s,1,9600,AMT681,0,10

1,=h -AMT681 SML Meter case1&2
; look Points in Text 0.0.9, 1.8.0, 2..8.0, 16.7.0
; below is translate to Comma on http !
; Point instead of Comma on http
; value on http with Comma OK, but in Log of mqtty
1%Ar%00009ff@#,0.0.9 Meter ID,,Meter_id,0
1%Ar%10800%Au%1.8.0 HT&NT Bezug,kWh,A1_8_0,4
1%Ar%20800%Au%2.8.0 HT&NT Lieferung,kWh,A2_8_0,4
1%Aq%@1,16.7.0 akt.Leistung,W,A16_7_0,1
1,=h<hr>
#

**(Please, remember to close the issue when the problem has been addressed)**

[TasmotaPointComma.zip](https://github.com/arendst/Tasmota/files/13937291/TasmotaPointComma.zip)

Jason2866, 2024-01-15:
Closing, answered.

closed 2024-01-15 by Jason2866

SteWers, 2024-01-23:
The PR #20573 resolves "case1".
"case2" is caused by the faulty script. The [documentation](https://tasmota.github.io/docs/Scripting-Language/#commands) tells how to solve it. Try something like `dp1,4` in the `>B`-section.

HofeBY, 2024-01-24:
Hallo SteWers
"case2" work´s direkt with your tip great !  Thank´s , I didn´t found it in documetation.

case1:
I compiled it with current TM13.3.0.4 and install it on a Testsystem.
partitial good.
Output Lines are doubled.
one Line with the wrong version.
one Line with the wrong version.
second Line with correct "." instead of "."

[image]

SteWers, 2024-01-24:
> @HofeBY
Output Lines are doubled.
one Line with the wrong version.

I think, you haven't removed the old line 2673. Please wait until the PR is merged.

HofeBY, 2024-01-24:
Hmm,
hope to do it right, substitute redline by green ones.
 [look](https://github.com/arendst/Tasmota/pull/20573)

SteWers, 2024-01-24:
No, you did not remove the old line:
[image]
"My" tasmota version is ahead with the development version.
Wait until the PR is merged. Than sync your branch and compile it again.
Please don't discuss that in the PR.

SteWers, 2024-01-24:
>@HofeBY
mqtty stream has now "," instead allowed "." stopping recording in IOBroker
"script section >J", here it is needed that "." is used, it seem that ";" is not allowd in mqtty. sorry

You can try something like this:

```
>D
A180=0
A280=0
A1670=0
A180Str=""
A280Str=""
A1670Str=""

>T
A180=AMT681#A1_8_0
A280=AMT681#A2_8_0
A1670=AMT681#A16_7_0
A180Str=A180
A280Str=A280
A1670Str=A1670
```

Then use the numerical variables in `>W` and the string variables in `>J`.