arendst/Tasmota #20521: Exception 6 IntegerDivideByZero When set ShutterMode 4

rt400, 2024-01-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When set ShutterMode 4 i got Exception 6 IntegerDivideByZero

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32 Lolin
- [X] Tasmota binary firmware version number used: 13.3.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Esptool_____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
{"Module":{"1":"ESP32-DevKit"}}
{"GPIO0":{"32":"Button1"},"GPIO1":{"33":"Button2"},"GPIO2":{"0":"None"},"GPIO3":{"34":"Button3"},"GPIO4":{"354":"Counter3"},"GPIO5":{"576":"LedLink_i"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"225":"Relay2"},"GPIO14":{"256":"Relay_i1"},"GPIO15":{"230":"Relay7"},"GPIO16":{"352":"Counter1"},"GPIO17":{"353":"Counter2"},"GPIO18":{"260":"Relay_i5"},"GPIO19":{"418":"PWM3"},"GPIO20":{"0":"None"},"GPIO21":{"229":"Relay6"},"GPIO22":{"231":"Relay8"},"GPIO23":{"232":"Relay9"},"GPIO24":{"0":"None"},"GPIO25":{"258":"Relay_i3"},"GPIO26":{"417":"PWM2"},"GPIO27":{"227":"Relay4"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
18:42:10.789 MQT: stat/Shading_System/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":122,"Free":389,"Rules":"ON Power1#state DO  power7 %value% endon ON Power3#state DO  power8 %value% endon ON Power5#state DO  power9 %value% endon"}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
{"Status":{"Module":1,"DeviceName":"Shading System","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","","","",""],"Topic":"Shading_System","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:41:25.144 MQT: stat/Shading_System/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"MejahezHome","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T20:58:20","StartupUTC":"2024-01-15T19:43:05","Sleep":0,"CfgHolder":4617,"BootCount":46,"BCResetTime":"2024-01-10T16:19:27","SaveCount":308}}
18:41:25.153 MQT: stat/Shading_System/STATUS2 = {"StatusFWR":{"Version":"12.5.0(tasmota32)","BuildDateTime":"2023-04-17T08:07:16","Core":"2_0_7","SDK":"4.4.4.20230310","CpuFrequency":160,"Hardware":"ESP32-D0WD rev.1","CR":"445/699"}}
18:41:25.161 MQT: stat/Shading_System/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mejahez-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
18:41:25.183 MQT: stat/Shading_System/STATUS4 = {"StatusMEM":{"ProgramSize":1525,"Free":1984,"Heap":147,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D4BC482D","000A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77"}}
18:41:25.203 MQT: stat/Shading_System/STATUS5 = {"StatusNET":{"Hostname":"ShadingSystem","IPAddress":"192.168.0.53","Gateway":"192.168.0.1","Subnetmask":"255.255.254.0","DNSServer1":"192.168.1.254","DNSServer2":"192.168.1.249","Mac":"94:3C:C6:32:EB:3C","IP6Global":"","IP6Local":"fe80::963c:c6ff:fe32:eb3c","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.254","DNSServer2":"192.168.1.249","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:41:25.218 MQT: stat/Shading_System/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.254","MqttPort":1883,"MqttClientMask":"Shading_System","MqttClient":"Shading_System","MqttUser":"homeassistant","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:41:25.232 MQT: stat/Shading_System/STATUS7 = {"StatusTIM":{"UTC":"2024-01-16T16:41:25","Local":"2024-01-16T18:41:25","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+02:00","Sunrise":"09:38","Sunset":"18:21"}}
18:41:25.251 MQT: stat/Shading_System/STATUS10 = {"StatusSNS":{"Time":"2024-01-16T18:41:25","COUNTER":{"C1":25090,"C2":4990,"C3":10517},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter2":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter3":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}}
18:41:25.269 MQT: stat/Shading_System/STATUS11 = {"StatusSTS":{"Time":"2024-01-16T18:41:25","Uptime":"0T20:58:20","UptimeSec":75500,"Heap":151,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":44},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","PWM":{"PWM1":0,"PWM2":0,"PWM3":0},"Wifi":{"AP":1,"SSId":"Mejahez-IoT","BSSId":"76:83:C2:26:58:3E","Channel":1,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:02"}}}
18:41:25.284 MQT: stat/Shading_System/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":250,"Close":250,"50perc":50,"Delay":0,"Opt":"1001","Calib":[300,500,700,900,1000],"Mode":"4"},"SHT1":{"Relay1":3,"Relay2":4,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"4"},"SHT2":{"Relay1":5,"Relay2":6,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"4"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I flash the last dev 13.3.0.3 and set SO80 1, and when i set in console ShutterMode 4 the consoe stuck and i see that he make reboot with this Exception:
```
00:00:03.882 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":6,"Reason":"IntegerDivideByZero","EPC":"40139b8c","EXCVADDR":"00000000","CallChain":["40139b89","40139cb9","40162ace","40133e3b","401108ca","401653a1","401654c5","40150f6f","40103f41","400f17ba","400f2e35","400f3828","40152489","401108ca","4015cd16","401654f1","400e695f"]},"BootCount":28}}
```
If i downgrade to 12.5.0 and the problem gone.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I already ask in Discord and @s-hadinger found the problem: 
```
The only 2 divisions that can go wrong are:
Shutter[i].close_velocity =  Shutter[i].open_max / Shutter[i].close_time / 2 ;
and
Shutter[i].close_velocity_max     = ShutterGlobal.open_velocity_max*Shutter[i].open_time / Shutter[i].close_time; 
So defeinitely Shutter[i].close_time is zero
```
And he already mention @stefanbode to take a look on it...

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2024-01-17:
Hi thanks for bringing this up. By default open and closetime is 10 on init. Therefore quite strange. Be at the mobile now and cannot really look through the log files. 

Keep you posted.

stefanbode, 2024-01-17:
Can you change the `shuttermotordelay 10`?

Steppermotors will never work without a delay, but the current code allows this to be 0. I assume that not in all cases this is catched. Thx

stefanbode, 2024-01-17:
I submitted a fix that in my eyes could be responsible. It is not what @s-hadinger and you think. The close time is correct in any case. I assume that the "mandatory" motordelay actuall still 0 is the root cause.

rt400, 2024-01-17:
> I submitted a fix that in my eyes could be responsible. It is not what @s-hadinger and you think. The close time is correct in any case. I assume that the "mandatory" motordelay actuall still 0 is the root cause.

Thanks. Now its working with the last DEV.. 
when i set ShutterMode 4 (without your idea about the delay) its work without Exception...
Can i close this issue ?

stefanbode, 2024-01-17:
Yes please