arendst/Tasmota #20589: Problem getting measured values in script from two BME280 sensors

JR777-1, 2024-01-25:
### PROBLEM DESCRIPTION
I'm using two BME280 devices with different addresss 0x76 and 0x77.
The data from the sensors are displayed correctly on the webui.
BME280-76 Temperature 22.2°C
BME280-76 Humidity 39.8%
...
BME280-77 Temperature 22.3°C
BME280-77 Humidity 40.1%
...

This is working as expected.

So i want to use the measured values in my script.
With two connected sensors the names shown when i read "Status 10" in console are "BME280-76" and "BME280-77".
But i cannot get the values from the sensors with the code "temp=BME280-76#Temperature" or "temp=BME280-77#Temperature".
I tried using the other index separator with SetOption64 1 without success.
With only one connected sensor i get the values correctly using "temp=BME280#Temperature". 
Is there a bug getting the sensor values within a script?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [ ] Tasmota binary firmware version number used: 13.2.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

JR777-1, 2024-01-25:
[image]

JR777-1, 2024-01-25:
[image]

sfromis, 2024-01-25:
As the Scripter docs says:
> Remark: JSON variable names (like all others) may not contain math operators like - , you should set [SetOption64 1](https://tasmota.github.io/docs/Commands/#setoption64) to replace `-` (dash) with `_` (underscore).

JR777-1, 2024-01-25:
> As the Scripter docs says:
> 
> > Remark: JSON variable names (like all others) may not contain math operators like - , you should set [SetOption64 1](https://tasmota.github.io/docs/Commands/#setoption64) to replace `-` (dash) with `_` (underscore).

So now i'm a little confused. 
I've already done this without getting the values (see first posting). 
Now set Option to 1 and it works. #confused 
But hey, thank you very much for your help
Maybe this information will be helpful in the documentation, see https://tasmota.github.io/docs/BME280/

closed 2024-01-25 by JR777-1