arendst/Tasmota #20608: TuyaMCU actions no commands when MQTT Broker is connected

JustH4ppy, 2024-01-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have a Sichler dehumidifier, which is currently not a supported device. It has a TYWE1S chip, which I flashed successfully. I was able to successfully assign TuyaRx and TuyaTx and was able to assign the fnids to dpids. I was able to toggle the relays I need successfully via the webinterface. As soon as I configure MQTT the device does not turn on/off anymore when toggeling the relays. No matter if actioned via Webinterface or MQTT.
When I remove the Host from the MQTT configuration it starts working again. When MQTT is enabled but the MQTT Broker is stopped, it is working too.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sichler NX-3291-675
- [X] Tasmota binary firmware version number used: 13.3.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotize
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:20:21.089 CMD: Backlog Template; Module; GPIO 255
13:20:21.092 SRC: WebConsole from 192.168.178.148
13:20:21.095 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
13:20:21.160 SRC: Backlog
13:20:21.163 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
13:20:21.170 MQT: stat/Luftentfeuchter/RESULT = {"NAME":"Luftentfeuchter","GPIO":[0,0,0,0,0,0,0,0,0,2304,0,2272,0,0],"FLAG":0,"BASE":54}
13:20:21.375 SRC: Backlog
13:20:21.377 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
13:20:21.383 MQT: stat/Luftentfeuchter/RESULT = {"Module":{"54":"Tuya MCU"}}
13:20:21.627 SRC: Backlog
13:20:21.630 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
13:20:21.640 MQT: stat/Luftentfeuchter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"2304":"Tuya Rx"},"GPIO14":{"0":"None"},"GPIO15":{"2272":"Tuya Tx"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
13:20:23.447 TYA: Send "55aa001c00080118011c0d14170798"
13:20:25.413 WIF: Sending Gratuitous ARP
13:20:26.539 WIF: Checking connection...
13:20:31.448 TYA: Send "55aa00000000ff"
13:20:31.457 DMP: 55 AA 00 00 00 01 01 01
13:20:31.459 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
13:20:31.462 TYA: Heartbeat

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:21:16.550 CMD: Status 0
13:21:16.552 SRC: WebConsole from 192.168.178.148
13:21:16.555 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
13:21:16.562 MQT: stat/Luftentfeuchter/STATUS = {"Status":{"Module":54,"DeviceName":"Luftentfeuchter","FriendlyName":["Luftentfeuchter",""],"Topic":"Luftentfeuchter","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
13:21:16.571 MQT: stat/Luftentfeuchter/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:05:52","StartupUTC":"2024-01-28T12:15:24","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2024-01-28T11:42:05","SaveCount":130,"SaveAddress":"FA000"}}
13:21:16.577 MQT: stat/Luftentfeuchter/STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota)","BuildDateTime":"2023-12-12T14:29:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"401/699"}}
13:21:16.583 MQT: stat/Luftentfeuchter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Verbinden...",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
13:21:16.596 MQT: stat/Luftentfeuchter/STATUS4 = {"StatusMEM":{"ProgramSize":637,"Free":364,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
13:21:16.606 MQT: stat/Luftentfeuchter/STATUS5 = {"StatusNET":{"Hostname":"Luftentfeuchter-3417","IPAddress":"192.168.178.44","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.180","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:3F:ED:59","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:21:16.611 MQT: stat/Luftentfeuchter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.180","MqttPort":1883,"MqttClientMask":"Luftentfeuchter","MqttClient":"Luftentfeuchter","MqttUser":"mqtt","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:21:16.619 MQT: stat/Luftentfeuchter/STATUS7 = {"StatusTIM":{"UTC":"2024-01-28T12:21:16","Local":"2024-01-28T13:21:16","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:26","Sunset":"17:39"}}
13:21:16.626 MQT: stat/Luftentfeuchter/STATUS10 = {"StatusSNS":{"Time":"2024-01-28T13:21:16"}}
13:21:16.632 MQT: stat/Luftentfeuchter/STATUS11 = {"StatusSTS":{"Time":"2024-01-28T13:21:16","Uptime":"0T00:05:52","UptimeSec":352,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Verbinden...","BSSId":"44:4E:6D:EE:B3:D6","Channel":6,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
With MQTT configured -> not working
13:22:54.458 APP: Boot Count 23
[... 97 lines cut ...]
13:24:56.103 {"TuyaReceived":{"Data":"55AA0002000001","Cmnd":2}}
13:24:56.105 TYA: RX MCU configuration Mode=0
13:24:56.106 TYA: Read MCU state
13:24:56.115 TYA: Send "55aa0008000007"
13:24:56.128 DMP: 55 AA 00 07 00 05 02 04 00 01 01 13
13:24:56.130 {"TuyaReceived":{"Data":"55AA00070005020400010113","Cmnd":7,"CmndData":"0204000101","DpType4Id2":1,"2":{"DpId":2,"DpIdType":4,"DpIdData":"01"}}}
13:24:56.131 TYA: fnId=0 is set for dpId=2
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Add/Remove MQTT Host IP. Toggle Power1 via Webinterface

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device turning On/Off when toggling DpID 101 which is working when MQTT is not configured

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
MQTT Broker used: eclipse-mosquiotto and emqx. With both the same problem

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-01-28:
I'm not sure to follow

In the 1st log block where you say "MQTT connect => Not working, I can clearly see:
```
13:23:12.122 MQT: stat/Luftentfeuchter/RESULT = {"POWER1":"OFF"}
13:23:12.126 MQT: stat/Luftentfeuchter/POWER1 = OFF
13:23:12.202 DMP: 55 AA 00 07 00 05 65 01 00 01 00 72
13:23:12.205 {"TuyaReceived":{"Data":"55AA00070005650100010072","Cmnd":7,"CmndData":"6501000100","DpType1Id101":0,"101":{"DpId":101,"DpIdType":1,"DpIdData":"00"}}}
13:23:12.208 TYA: fnId=11 is set for dpId=101
13:23:12.210 TYA: RX Relay-1 --> MCU State: Off Current State:Off
```
That I can translate by
- Web action click on Web button to set Power OFF
- Tasmota send a command (opcode 6) to dpId 101 with value 0
- MCU replies with a status (opcode 7) where dpId 101 has value 0

And right after the same dialog with value 0 for OFF

So it looks like it is working as expected

The 2nd block of code with MQTT not connected doesn't show a valid exchange on dpId 101

JustH4ppy, 2024-01-28:
Hey,

sorry, maybe the second logs with MQTT not configured were not complete. Here again with MQTT not configured and device turning On/Off correctly:
```
17:35:32.940 SRC: WebGui from 192.168.178.148
17:35:32.955 TYA: Send "55aa00060005650100010172"
17:35:32.979 RSL: RESULT = {"POWER1":"ON"}
17:35:32.981 RSL: POWER1 = ON
17:35:33.049 DMP: 55 AA 00 07 00 05 65 01 00 01 01 73
17:35:33.051 {"TuyaReceived":{"Data":"55AA00070005650100010173","Cmnd":7,"CmndData":"6501000101","DpType1Id101":1,"101":{"DpId":101,"DpIdType":1,"DpIdData":"01"}}}
17:35:33.054 TYA: fnId=11 is set for dpId=101
17:35:33.056 TYA: RX Relay-1 --> MCU State: On Current State:On
17:35:33.059 DMP: 55 AA 00 07 00 05 02 04 00 01 01 13
17:35:33.061 {"TuyaReceived":{"Data":"55AA00070005020400010113","Cmnd":7,"CmndData":"0204000101","DpType4Id2":1,"2":{"DpId":2,"DpIdType":4,"DpIdData":"01"}}}
17:35:33.064 TYA: fnId=0 is set for dpId=2
17:35:33.066 DMP: 55 AA 00 07 00 05 0C 05 00 01 00 1D
17:35:33.068 {"TuyaReceived":{"Data":"55AA000700050C050001001D","Cmnd":7,"CmndData":"0C05000100","DpType5Id12":"0x00","12":{"DpId":12,"DpIdType":5,"DpIdData":"00"}}}
17:35:33.070 TYA: fnId=0 is set for dpId=12
17:35:33.072 DMP: 55 AA 00 07 00 05 65 01 00 01 01 73
17:35:33.074 {"TuyaReceived":{"Data":"55AA00070005650100010173","Cmnd":7,"CmndData":"6501000101","DpType1Id101":1,"101":{"DpId":101,"DpIdType":1,"DpIdData":"01"}}}
17:35:33.077 TYA: fnId=11 is set for dpId=101
17:35:33.078 TYA: RX Relay-1 --> MCU State: On Current State:On
17:35:33.080 DMP: 55 AA 00 07 00 05 66 01 00 01 00 73
17:35:33.083 {"TuyaReceived":{"Data":"55AA00070005660100010073","Cmnd":7,"CmndData":"6601000100","DpType1Id102":0,"102":{"DpId":102,"DpIdType":1,"DpIdData":"00"}}}
17:35:33.086 TYA: fnId=12 is set for dpId=102
17:35:33.088 TYA: RX Relay-2 --> MCU State: Off Current State:Off
17:35:33.092 DMP: 55 AA 00 07 00 05 67 01 00 01 01 75
17:35:33.094 {"TuyaReceived":{"Data":"55AA00070005670100010175","Cmnd":7,"CmndData":"6701000101","DpType1Id103":1,"103":{"DpId":103,"DpIdType":1,"DpIdData":"01"}}}
17:35:33.097 TYA: fnId=0 is set for dpId=103
17:35:33.315 CFG: Saved to flash at F6, Count 166, Bytes 4096
17:35:36.252 HTP: Console
17:35:40.453 TYA: Send "55aa00000000ff"
17:35:40.462 DMP: 55 AA 00 00 00 01 01 01
17:35:40.464 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:35:40.466 TYA: Heartbeat
17:35:51.456 TYA: Send "55aa00000000ff"
17:35:51.466 DMP: 55 AA 00 00 00 01 01 01
17:35:51.468 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:35:51.471 TYA: Heartbeat
17:35:52.518 WIF: Checking connection...
17:36:02.447 TYA: Send "55aa00000000ff"
17:36:02.456 DMP: 55 AA 00 00 00 01 01 01
17:36:02.459 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:36:02.461 TYA: Heartbeat
```

> That I can translate by
> 
> MQTT command to set Power OFF
> Tasmota send a command (opcode 6) to dpId 101 with value 0
> MCU replies with a status (opcode 7) where dpId 101 has value 0
> And right after the same dialog with value 0 for OFF
> 
> So it looks like it is working as expected

The MCU responds correctly, but the device does not. When MQTT is configured I can Send/Receive commands and status from the MCU, but the device itself does nothing at all. As soon as I unconfigure MQTT the device works fine when toggeling both Relays.
If this somehow helps, I can also make a video about it.
I'm happy about any help!

Thanks!

benjaminnestler, 2024-02-02:
Hey @JustH4ppy 

After a thorough analysis, I would claim that the standard Tuya Wifi Serial Protocol is not implemented in your Tuya MCU. It could be the Wifi for Low Power Protocol or a Bluetooth Serial Protocol (which would, of course, not be compatible with the TYWE1S).

As I interpret it, Tasmota supports the standard Wifi Serial Protocol (@barbudor, would you agree with me on this?)

The following observations led me to this assumption:

- The responses from your Tuya MCU always start with the header 0x55 AA 00 (the standard Tuya Wifi Protocol Response would start with the header 0x55 AA 03).
- The heartbeats are not answered correctly —> In the Wifi Low Power Protocol, there are no heartbeats, but in Bluetooth Wifi Serial, there are, and the MCU's response would at least correspond to the protocol.
- The reporting of the Network States that Tasmota tries to send to the MCU is completely misinterpreted by the Tuya MCU:

Tasmota reports the Network State in several stages. When Wifi is activated, Tasmota sends Network State 3 (0x02 = The Wi-Fi network is set up, but the device is not connected to the router).

`00:00:00.168 TYA: Send "55aa000300010205"`

When Tasmota is connected to Wifi, it sends an incremented Network State 4 (0x03 = The Wi-Fi network is set up, and the device is connected to the router).

`13:24:56.011 TYA: Send "55aa000300010306"`

The next stage would be a connection with MQTT. Then Tasmota would send Network State 5 (0x04 = The device is connected to the cloud). I would have liked to see this in your first LOG (with MQTT enabled).

Regardless of which Network State Tasmota sends, your TuyaMCU always responds with 55 AA 00 03 00 00 02.

In the normal Wifi Serial Protocol, this response does not exist. It is invalid!

However, assuming that your Tuya MCU responds to the Wifi Low Power Protocol, it would repeatedly signal the Wifi Reset command in its response.

Check here: https://developer.tuya.com/en/docs/iot/wifie1smodulemcu?id=K9hhi0xp301u1
and here: https://developer.tuya.com/en/docs/iot/tuyacloudlowpoweruniversalserialaccessprotocol?id=K95afs9h4tjjh

JustH4ppy, 2024-02-02:
Hey @benjaminnestler,

thank you very much for your deep research!
That would totally make sense!
What would be your suggestion on solving that? Can I somehow suppress sending the network state? Or is there a way to program the MCU to adjust the firmware?

Meanwhile I quickly want to post my workaround, how I currently managed to get the device controlled by Home Assistant without MQTT. Maybe it's useful for somebody dealing with the same.

```
command_line:
  - switch:
      name: Luftentfeuchter Reinigung
      command_on: >
        curl -X POST "http://192.168.178.44/cm?cmnd=Power1%20On"
      command_off: >
        curl -X POST "http://192.168.178.44/cm?cmnd=Power1%20Off"
      command_state: curl -X GET "http://192.168.178.44/cm?cmnd=Power1"
      value_template: >
        {{value_json.POWER1 == "ON" }}
  - switch:
      name: Luftentfeuchter Entfeuchtung
      command_on: >
        curl -X POST "http://192.168.178.44/cm?cmnd=Power2%20On"
      command_off: >
        curl -X POST "http://192.168.178.44/cm?cmnd=Power2%20Off"
      command_state: curl -X GET "http://192.168.178.44/cm?cmnd=Power2"
      value_template: >
        {{value_json.POWER2 == "ON" }}
```

barbudor, 2024-02-02:
Great analysis @benjaminnestler 

Actually I always refer to the "Tuya Wifi Serial Protocol" documentation and I even didn't noticed that "Wifi for Low Power Protocol"

At least we see possibility for improvements as it should be possible to indentify which of those 2 protocols the MCU is using and adapt the dialog accordingly.
But some more work and I don't have any of such devices in hand

JustH4ppy, 2024-02-02:
Hey @barbudor,

I don't know if this is helpful, but I would like to provide any help remotely needed if this somehow helps.
I can do photos of the pcb and also could set up a notebook in my network connected to the system, which you can remotely access if this somehow helps. Just let me know :)

benjaminnestler, 2024-02-02:
Hi @JustH4ppy,

> Can I somehow suppress sending the network state? Or is there a way to program the MCU to adjust the firmware?

Currently, there is no SetOption or similar functionality to prevent reporting the network state. For the standard Tuya serial protocol, suppressing it does not seem practical.

Tasmota is designed to adhere to the protocol by reporting the network state. The following lines are extracted from the Tuya serial protocol description:

>- When the module detects that the MCU is restarted or reconnected, it will proactively send the current Wi-Fi status to the MCU.
>- When the Wi-Fi status changes, the module will proactively send the current status to the MCU.

Perhaps we could consider implementing a configuration option to allow users to switch between different Tuya protocols, enabling them to decide which one to use.

@barbudor, I'm not sure if we can identify the protocol. The only difference lies in the third byte of the protocol versions (in some cases, 0x00 is used, and in others, 0x03), as well as in the available commands.

JustH4ppy, 2024-02-28:
Issue still exists, please keep it open.

JustH4ppy, 2024-03-24:
Please keep issue open.

JustH4ppy, 2024-04-18:
Please keep open

JustH4ppy, 2024-05-13:
please keep open

closed 2024-06-12 by github-actions[bot]