arendst/Tasmota #20639: MQTT stat messages way too often

cschwarz007, 2024-02-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
On my Shelly Duo RGBW, MQTT transmits stat messages way too often. Telemtry Period is set to 300 (default), but stat messages are sent every second. This is an issue for me as I want to use a Rule to trigger on DIMMER or CT changes only if they really occur (value change).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:40:20.678 MQT: stat/JonasLEDLamp/RESULT = {"NAME":"Shelly Duo RGBW","GPIO":[0,0,0,0,0,419,0,0,417,416,418,0,0,0],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
17:40:54.843 MQT: stat/JonasLEDLamp/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:41:19.754 MQT: stat/JonasLEDLamp/STATUS = {"Status":{"Module":0,"DeviceName":"JonasLEDLamp","FriendlyName":["LED Lampe"],"Topic":"JonasLEDLamp","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:41:19.760 MQT: stat/JonasLEDLamp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:16","StartupUTC":"2024-02-01T16:37:03","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2024-01-11T15:59:49","SaveCount":96,"SaveAddress":"FA000"}}
17:41:19.765 MQT: stat/JonasLEDLamp/STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota)","BuildDateTime":"2024-01-09T21:43:12","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"402/699"}}
17:41:19.770 MQT: stat/JonasLEDLamp/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FritzHOME_DE","Aqua"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["0A208009","0405C80001000600003C5A0A002800000000","08000080","01806400","00004000","00000000"]}}
17:41:19.783 MQT: stat/JonasLEDLamp/STATUS4 = {"StatusMEM":{"ProgramSize":647,"Free":356,"Heap":18,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","9F80C787","04368001","000000CF","010013C0","C000F181","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,6,8,9,10,11,12,16,18,19,20,21,22,24,26,27,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
17:41:19.789 MQT: stat/JonasLEDLamp/STATUS5 = {"StatusNET":{"Hostname":"jonasledlamp","IPAddress":"192.168.101.65","Gateway":"192.168.101.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.101.23","DNSServer2":"0.0.0.0","Mac":"48:55:19:DB:2E:99","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:41:19.795 MQT: stat/JonasLEDLamp/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.janettundchristian.de","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DB2E99","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:41:19.802 MQT: stat/JonasLEDLamp/STATUS7 = {"StatusTIM":{"UTC":"2024-02-01T16:41:19","Local":"2024-02-01T17:41:19","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:20","Sunset":"17:46"}}
17:41:19.809 MQT: stat/JonasLEDLamp/STATUS10 = {"StatusSNS":{"Time":"2024-02-01T17:41:19"}}
17:41:19.818 MQT: stat/JonasLEDLamp/STATUS11 = {"StatusSTS":{"Time":"2024-02-01T17:41:19","Uptime":"0T00:04:16","UptimeSec":256,"Heap":19,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"FritzHOME_DE","BSSId":"1C:ED:6F:26:F4:7A","Channel":13,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
17:42:18.164 SRC: MQTT
17:42:18.167 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:18.174 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
17:42:18.215 SRC: MQTT
17:42:18.218 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:18.226 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
17:42:18.245 UDP: Packet (119)
17:42:18.247 UDP: HUE
17:42:18.265 SRC: MQTT
17:42:18.268 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:18.276 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
17:42:18.317 SRC: MQTT
17:42:18.319 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:18.327 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
17:42:19.124 SRC: MQTT
17:42:19.127 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:19.136 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
17:42:19.177 SRC: MQTT
17:42:19.180 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:19.188 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
17:42:19.218 SRC: MQTT
17:42:19.221 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:19.230 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
[... 17 lines cut ...]
17:42:20.269 SRC: MQTT
17:42:20.272 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:20.280 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
17:42:20.320 SRC: MQTT
17:42:20.323 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
17:42:20.331 MQT: stat/JonasLEDLamp/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000B847","HSBColor":"0,0,0","White":100,"CT":250,"Channel":[0,0,0,72,28]}
17:42:21.128 SRC: MQTT

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Settings according to above

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
MQTT Stat messages are trasmitted when value changes (or telemetry period is reached)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-02-01:
When you issue commands all the time (publishing `Dimmer 100` commands via MQTT), you should expect each command you issue to generate a response. As you did not tell from where those commands are coming, you did not post steps to reproduce. Seeing `SRC: MQTT` indicates that a command is incoming via MQTT, but the source is not visible.

If you do not want the same commands being published incessantly, you should do something about the source of those commands.

cschwarz007, 2024-02-01:
Thank you - sometimes the problem sits right in front of the screen... My apologies

closed 2024-02-01 by cschwarz007