arendst/Tasmota #20687: BMP 280 temperature and pressure reading too high

jdschuitemaker, 2024-02-08:
### PROBLEM DESCRIPTION
I have a new BMP280 temperature and pressure sensor connected to a Wemos S2 MINI V1 TYPE-C. Both the readings for temperature and pressure are too high. The Tasmota UI is showing 188.5 degrees Celsius (and does not change) for the temperature and current pressure is around 1214 hPa and does fluctuate a little bit.

The BMP280 is powered from the 3.3V pin, SDA to GPIO8 and SCL to GPIO9. I have set the module settings in the configuration accordingly. Tasmota is obviously showing readings, but they are incorrect. This is not my first soldering job. I have verified my wiring and all seems correct. I do not have another ESP32 or BMP280 to test with.

[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Wemos S2 Mini V1 Type-C
- [X ] Tasmota binary firmware version number used: 13.3.0 (tasmota32
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: Web installer
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:08:58.879 CMD: Backlog Template; Module; GPIO 255
15:08:58.903 RSL: RESULT = {"NAME":"ESP32S2","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}
15:08:59.128 RSL: RESULT = {"Module":{"1":"ESP32S2"}}
15:08:59.337 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"640":"I2C SDA1"},"GPIO9":{"608":"I2C SCL1"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"},"GPIO40":{"0":"None"},"GPIO41":{"0":"None"},"GPIO42":{"0":"None"},"GPIO43":{"0":"None"},"GPIO44":{"0":"None"},"GPIO45":{"0":"None"},"GPIO46":{"0":"None"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
15:29:15.484 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:29:15.720 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:29:15.970 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [X ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:29:51.882 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_49941C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:29:51.916 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32s2.bin","RestartReason":"Software reset digital core","Uptime":"0T00:01:59","StartupUTC":"2024-02-08T14:27:52","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2024-02-06T21:18:55","SaveCount":24}}
15:29:51.950 RSL: STATUS2 = {"StatusFWR":{"Version":"13.3.0(tasmota32)","BuildDateTime":"2023-12-12T14:31:54","Core":"2_0_14","SDK":"4.4.6.231122","CpuFrequency":240,"Hardware":"ESP32-S2FNR2 v1.0","CR":"408/699"}}
15:29:51.963 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot-network",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000001"]}}
15:29:52.002 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1776,"Free":1103,"Heap":97,"StackLowMark":4,"PsrMax":2044,"PsrFree":2024,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","080840D2","20200000","D4BC482D","810A80B1","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
15:29:52.058 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-49941C-5148","IPAddress":"192.168.1.51","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.3","DNSServer2":"0.0.0.0","Mac":"80:65:99:49:94:1C","IP6Global":"","IP6Local":"fe80::8265:99ff:fe49:941c%st1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":15.0}}
15:29:52.092 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_49941C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:29:52.116 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-02-08T14:29:52","Local":"2024-02-08T15:29:52","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:10","Sunset":"17:57"}}
15:29:52.135 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-02-08T15:29:52","BMP280":{"Temperature":188.5,"Pressure":1214.3},"ESP32":{"Temperature":37.4},"PressureUnit":"hPa","TempUnit":"C"}}
15:29:52.151 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-02-08T15:29:52","Uptime":"0T00:02:00","UptimeSec":120,"Heap":97,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":45},"Wifi":{"AP":1,"SSId":"iot-network","BSSId":"14:59:C0:3A:25:B6","Channel":7,"Mode":"11n","RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
16:34:37.382 CFG: Saved, Count 25, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
I am expecting to see a temperature that is around 20-22 degrees Celsius. Weather sites provide information that current pressure is around 994 hPa and not above 1200, which would be pretty high anyway.

### SCREENSHOTS
See description

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-02-08:
Try a better power supply

jdschuitemaker, 2024-02-08:
Thanks! I should have added that to my initial post. I tried multiple chargers and usb-c cables.

sfromis, 2024-02-08:
I've seen cases of trouble with clones of the Wemos S2 Mini. Is your board a *branded* original?