arendst/Tasmota #20702: Restore fails on define USE_SCRIPT firmware

Subnum12, 2024-02-11:
### PROBLEM DESCRIPTION
It is not possible to restore backup dump files if firmware was compiled with USE_SCRIPT.

I did create 2 firmware files based on tasmota-4M (13.3.0)
Firmware A was done with no modification except some sensors removals (to reduce firmware size - just in case)
Firmware B was done with same settings as firmware A with additional define
```
#ifndef USE_SCRIPT
#define USE_SCRIPT
  #ifdef USE_RULES
    #undef USE_RULES
  #endif
#endif
```

I can create backup dump on both versions.
I can restore both dump files on Firmware A.
I cannot restore both dump files on Firmware B.
I'm getting "Not enough space" while I guess in fact real issue belongs on parsing dump file

console output (level 4)
```
[10:57:40]10:57:39.802 HTP: Restore Configuration
[10:57:52]10:57:51.704 CFG: Saved to flash at 3FB, Count 202, Bytes 4096
[10:57:52]10:57:51.704 UPL: File Config_tasmota.dmp
[10:57:52]10:57:51.725 APP: Not enough space
[10:57:52]10:57:51.734 UPP: Multicast (re)joined
[10:57:52]10:57:51.737 HTP: Upload done
```
I did compare dumps with [decode-config](https://github.com/tasmota/decode-config) and noticed on firmware B parameter **rules** got removed while **script** where added beside I guess not relevant changes on crc and save_flag modification.

As the script itself it not part of the backup (saved in script.txt on UFS) I do not understand the behavior.

Going back to older versions it looks like behavior did occur on Tasmota 12.1 until now.

memory on Firmware A
```
Flash Size	4096 KB
Program Flash Size	4096 KB
Program Size	611 KB
Free Program Space	1436 KB
Free Memory	26.6 KB
File System Size	2024 KB
```
memory on Firmware B
```
Flash Size	4096 KB
Program Flash Size	4096 KB
Program Size	625 KB
Free Program Space	1420 KB
Free Memory	17.9 KB 
File System Size	2024 KB
```

arendst, 2024-02-11:
Pls enable heap logging with ``SO130 1`` and watch available space when trying to upload config.dmp.

Subnum12, 2024-02-11:
no additional output...
````
[12:07:51]12:07:49.616-016 CMD: SO130 1
[12:07:51]12:07:49.619-016 SRC: WebConsole from 192.168.10.101
[12:07:51]12:07:49.620-016 CMD: Grp 0, Cmd 'SO', Idx 130, Len 1, Pld 1, Data '1'
[12:07:51]12:07:49.622-016 RSL: RESULT = {"SetOption130":"ON"}
[12:07:52]12:07:50.664-017 WIF: Checking connection...
[12:08:03]12:08:01.759-016 HTP: Main Menu
[12:08:04]12:08:03.141-016 HTP: Configuration
[12:08:06]12:08:04.561-017 WIF: Sending Gratuitous ARP
[12:08:06]12:08:04.968-016 HTP: Restore Configuration
[12:08:11]12:08:09.792-012 UPP: Multicast disabled
[12:08:11]12:08:09.987-009 CFG: Saved to flash at 3FB, Count 225, Bytes 4096
[12:08:11]12:08:09.988-009 UPL: File Config_tasmota.dmp
[12:08:11]12:08:09.008-010 APP: Not enough space
[12:08:11]12:08:10.004-012 UPP: Multicast (re)joined
[12:08:11]12:08:10.007-014 HTP: Upload done
[12:08:12]12:08:10.676-017 WIF: Checking connection...
````

arendst, 2024-02-11:
Yes. Additional output is lack of heap space to safely hold the restored dump file. As your heap is down to 9k it won't allow a restore of the config. The config backup and restore has changed to support additional filesystem configs (hidden .xdrvsetxxx files) into the .dmp file. To accomodate this the configs have to be loaded in RAM (using heap) before being able to save/load the .dmp file.
So time to freeup heap by removing drivers.
A workaround might be to lower the lowest heap mark chk in file ``settings.ino`` line 413 which currently reads like:
```
    if ((mem - upload_size) < (8 * 1024)) {
```
from 8k to 6k adding instability but might fix your case.

Subnum12, 2024-02-11:
I've done a small script:

````
>S
print heap:%0heap%
print ram:%0ram%
````
with that I did change option 130 and also try to upload dump again.

console output:
````
13:09:30.407 heap:17168
13:09:30.409 ram:8390
13:09:30.511 CMD: so130 0
13:09:30.515 RSL: RESULT = {"SetOption130":"OFF"}
13:09:31.412 heap:17144
13:09:31.414 ram:8390
13:09:32.416 heap:17144
13:09:32.418 ram:8390
13:09:33.404 heap:17168
13:09:33.406 ram:8390
13:09:34.415 heap:17064
13:09:34.417 ram:8390
13:09:34.867 CMD: so130 1
13:09:34.871-016 RSL: RESULT = {"SetOption130":"ON"}
13:09:35.481-016 heap:17144
13:09:35.483-016 ram:8390
13:09:36.400-016 heap:17144
13:09:36.402-016 ram:8390
13:09:37.406-016 heap:17144
13:09:37.408-016 ram:8390
13:09:38.087-009 UPL: File Config_tasmota.dmp
13:09:38.118-013 UPP: Multicast (re)joined
13:09:40.063-016 heap:17056
13:09:40.065-016 ram:8390
13:09:40.420-016 heap:17192
13:09:40.422-016 ram:8390
13:09:41.405-016 heap:17168
13:09:41.407-016 ram:8390
````

with free heap > 16k I'm not below 8.192 limitation.

beside this I've done several undef  already to reduce memory I found [here](// https://tasmota.github.io/docs/BUILDS/#available-features-and-sensors).
````
#undef USE_SONOFF_RF
#undef SE_SONOFF_SC
#undef USE_TUYA_MCU
#undef USE_PS_16_DZ
#undef USE_SONOFF_IFAN
#undef USE_BUZZER
#undef USE_ARILUX_RF
#undef USE_SHUTTER
#undef USE_EXS_DIMMER
#undef USE_DEVICE_GROUPS
#undef USE_PWM_DIMMER
#undef USE_AC_ZERO_CROSS_DIMMER

#undef USE_WS2812
#undef USE_MY92X1
#undef USE_SM16716
#undef USE_SM2135
#undef USE_SM2335
#undef USE_BP5758D
#undef USE_BP1658CJ
#undef USE_ELECTRIQ_MOODL
#undef USE_PZEM004T
#undef USE_PZEM_AC
#undef USE_PZEM_DC
#undef USE_MCP39F501
#undef USE_BL09XX
#undef USE_DS18x20
#undef USE_ADE7953

#undef USE_IR_REMOTE
#undef USE_IR_RECEIVE
````

Subnum12, 2024-02-11:
I just noticed I cannot even upload a 1byte file to Ufs. The file just gets not created, while creating a new file via web console works
````
Manage File system
UFS: UfsEditor GET
UFS: UfsEditor: file=//newfile.txt, ffs_type=3, TfsFileExist=0
WIF: Checking connection...
UFS: UfsEditor: file upload
UFS: UfsEditor: file '/newfile.txt'
HTP: Manage File system
````

closed 2024-03-12 by github-actions[bot]

CWempe, 2024-10-13:
I am experiencing the same issue.

I am using a firmware compiled with TasmoCompiler v12.2.0 fpr https://tasmota.github.io/docs/Smart-Meter-Interface/

Sp my `user_config_override.h` looks just like this:

```c
#ifndef USE_SCRIPT
#define USE_SCRIPT
#endif
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif
```

I can backup my configuration, but restoring into a new device (only Wifi configured) it does not work: `Not enough space`

Not a deal breaker for me, but it looks more like a bug than a restriction.
If this is expected there should be a warning during the backup process.

wvdv2002, 2025-01-05:
I can confirm this is still a bug with using the USE_SCRIPT define.

Gemikro1, 2025-01-08:
I have the same problem with 14.4.0.1, even with USE_SCRIPT disabled and most of the drivers disabled, heap is around 18k.
The issue seems to have been closed already though, we need to open a new issue,

Gemikro1, 2025-01-08:
In my configuration (Custom firmware based on 14.4.0.1 with `USE_MQTT TLS` and `USE_SHUTTER` enabled and #undef all unnecessary drivers) the restore is not working any more as soon as the heap is going below 19k.
I tried to reset the flashed firmware and initially there are around 20k heap free. I can save and restore the configuration at this time.
If i then activate a custom profile (Shelly 2.5) the heap is reduced to 18k and restore is not working any more with the `Not enough space` message.
Saving and restoring has been working with the 12.4.0 version with the same configuration without any problems.
`USE_SCRIPT` was not defined in my case so the problem is not related to this define but more to the 19k heap limit.