arendst/Tasmota #20703: tasmota32-webcam on ESP32-D0WD-V3 v3.0

tobiknue, 2024-02-11:
### PROBLEM DESCRIPTION
As described in ct 06/2023 page 32, tasmota32-webcam has problems with boars rev. "D0WD-V3 rev.3". With tasmota32-webcam the device nearly freezes, the webcam has max 1 fps. It is hard to get a connetion to the device. It is not possible to switch to console.
I tried with my ESP32-D0WD-V3 v3.0 and actual tasmota32-webcam v13.3.0 and 13.3.0.5. With both versions the problem still exists.

Tasmota32 13.3.0 works properly as expected. It is possible to turn on the LED an adjust their brightness but no camera.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____

https://templates.blakadder.com/ai-thinker_ESP32-CAM.html

- [x] Tasmota binary firmware version number used: 13.3.0 and 13.3.0.5
  - [x] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: WebInstaller
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
not possible. Can't open console, because device freezes, see above.

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
not possible. Can't open console, because device freezes, see above.

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
not possible. Can't open console, because device freezes, see above.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
load tasmota32-webcam 13.3.0 or 13.3.0.5 on an ESP32-D0WD-V3 v3.0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Simple usage of tasmota32-webcam possible.
tasmota32 works.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-02-11:
What happens if you switch of mqtt support in the Web config page of Tasmota?
(Power off / on after switching mqtt off)

tobiknue, 2024-02-12:
Hello Jason2866,

thanks for your support! I switched off MQTT in web-frontend and saved. After this, the CAM was reachable only after a few restarts with power off and on. When i tried to go to console, it freezes again.

But directly after flashing again i could connect to console:
```
00:00:00.002 HDW: ESP32-D0WD-V3 v3.0 (PSRAM)
00:00:00.017 UFS: FlashFS mounted with 308 kB free
00:00:00.031 CFG: Loaded from File, Count 5
00:00:00.039 QPC: Count 1
00:00:00.090 BRY: Berry initialized, RAM used 3599 bytes
00:00:00.103 Project tasmota - Tasmota Version 13.3.0(webcam)-2_0_14(2023-12-12T14:31:23)
00:00:00.139 CAM: InitErr 0x105 try 1
00:00:00.173 CAM: InitErr 0x105 try 2
00:00:00.206 CAM: InitErr 0x105 try 3
00:00:00.206 CAM: InitErr 0x105
00:00:00.928 WIF: Connecting to AP1 XXX in mode 11n as tasmota-3CFDCC-7628...
00:00:04.704 WIF: Connected
00:00:04.961 HTP: Web server active on tasmota-3CFDCC-7628 with IP address 192.168.179.7
14:18:05.642 RSL: INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"13.3.0(webcam)","FallbackTopic":"cmnd/DVES_3CFDCC_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:18:05.656 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-3CFDCC-7628","IPAddress":"192.168.179.7","IP6Global":"2a02:3100:7048:8001:ab6:1fff:fe3c:fdcc","IP6Local":"fe80::ab6:1fff:fe3c:fdcc%st1"}}
14:18:05.668 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":2}}
14:18:05.947 QPC: Reset
14:18:08.965 RSL: STATE = {"Time":"2024-02-11T14:18:08","Uptime":"0T00:00:09","UptimeSec":9,"Heap":173,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":32},"Wifi":{"AP":1,"SSId":"XXX","BSSId":"YYY","Channel":6,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}
14:18:13.429 CAM: InitErr 0x105 try 1
14:18:13.463 CAM: InitErr 0x105 try 2
14:18:13.497 CAM: InitErr 0x105 try 3
14:18:13.498 CAM: InitErr 0x105
```

The Wifi-options i can try tomorrow

Jason2866, 2024-02-12:
Have you tried the Cam with a other firmware? The error `0x105` looks like bad hardware. 
It Is a different issue as c`t encountered. I have one of the problematic c`t cams. Direct from the article author. The cam does not fail with error `0x105`!

My guess, you have one of the many bad esp32 cams.

tobiknue, 2024-02-12:
I have tried AI-on-the-edge v15.6.0 with the same device. The Log has nor errors (except the missing SD-Card), espacially no ram-error.
But of course I can't rule out the possibility that I got a bad example.

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7664
load:0x40078000,len:15360
load:0x40080400,len:3744
entry 0x4008067c
I (27) boot: ESP-IDF 4.3.2 2nd stage bootloader
I (27) boot: compile time 20:02:47
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 ota_0            OTA app          00 10 00010000 001db000
I (97) boot:  4 ota_1            OTA app          00 11 001f0000 001db000
I (104) boot: End of partition table
I (109) boot: No factory image, trying OTA 0
I (114) boot_comm: chip revision: 3, min. application chip revision: 0
I (121) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4071ch (263964) map
I (230) esp_image: segment 1: paddr=00050744 vaddr=3ffb0000 size=040e8h ( 16616) load
I (238) esp_image: segment 2: paddr=00054834 vaddr=40080000 size=0b7e4h ( 47076) load
I (258) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=1103b0h (1115056) map
I (685) esp_image: segment 4: paddr=001703d8 vaddr=4008b7e4 size=0eae0h ( 60128) load
I (724) boot: Loaded app from partition at offset 0x10000
I (767) boot: Set actual ota_seq=1 in otadata[0]
I (767) boot: Disabling RNG early entropy source...
I (778) cpu_start: Multicore app
I (778) quad_psram: This chip is ESP32-D0WD
I (779) esp_psram: Found 8MB PSRAM device
I (779) esp_psram: Speed: 40MHz
I (782) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (789) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (797) cpu_start: Pro cpu up.
I (800) cpu_start: Starting app cpu, entry point is 0x40082e0c
I (0) cpu_start: App cpu up.
I (1685) esp_psram: SPI SRAM memory test OK
I (1694) cpu_start: Pro cpu start user code
I (1694) cpu_start: cpu freq: 160000000 Hz
I (1694) cpu_start: Application information:
I (1694) cpu_start: Project name:     AI-on-the-edge
I (1699) cpu_start: App version:      v15.6.0
I (1703) cpu_start: Compile time:     Feb  9 2024 13:01:46
I (1708) cpu_start: ELF file SHA256:  8455cc9b8820fc12...
I (1713) cpu_start: ESP-IDF:          5.1.2
I (1717) cpu_start: Min chip rev:     v0.0
I (1721) cpu_start: Max chip rev:     v3.99 
I (1725) cpu_start: Chip rev:         v3.0
I (1729) heap_init: Initializing. RAM available for dynamic allocation:
I (1735) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1740) heap_init: At 3FFBC668 len 00023998 (142 KiB): DRAM
I (1746) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1751) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1757) heap_init: At 4009A2C4 len 00005D3C (23 KiB): IRAM
I (1763) esp_psram: Adding pool of 4082K of PSRAM memory to heap allocator
I (1770) spi_flash: detected chip: generic
I (1773) spi_flash: flash io: dio
I (1787) app_start: Starting scheduler on CPU0
I (1787) app_start: Starting scheduler on CPU1
I (1787) main_task: Started on CPU0
I (1797) esp_psram: Reserving pool of 40K of internal memory for DMA/internal allocations
I (1797) main_task: Calling app_main()
I (1797) MAIN: 

================ Start app_main =================
E (1847) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (1847) sdcard_init: sdmmc_card_init failed (0x107).
E (1847) MAIN: SD card init failed. Check if SD card is properly inserted into SD card slot or try another card
E (1857) MAIN: Device init aborted!
I (1867) main_task: Returned from app_main()

```

Mierbach, 2024-02-14:
Ich habe fast genau den gleichen Fehler mit dem ESP32"-D0WD-v3. Hast du schon eine Lösung für das Problem gefunden?

12:44:28.006 -> ================ Start app_main =================
12:44:28.006 -> E (1631) partition: Partition table MD5 mismatch
12:44:28.006 -> E (1641) partition: load_partitions returned 0x103
12:44:28.038 -> W (1681) sdcard_init: failed to mount card (13)
12:44:28.038 -> E (1691) sdcard_init: mount_to_vfs failed (0xffffffff).
12:44:28.038 -> E (1691) MAIN: Failed to mount FAT filesystem on SD card. Check SD card filesystem (only FAT supported) or try another card
12:44:28.084 -> E (1691) MAIN: Device init aborted!
12:44:28.084 -> I (1701) main_task: Returned from app_main()

tobiknue, 2024-02-14:
> Ich habe fast genau den gleichen Fehler mit dem ESP32"-D0WD-v3. Hast du schon eine Lösung für das Problem gefunden?
> 

Nein, ich bin noch nicht weiter. Ich wollte als nächstes versuchen, den https://github.com/easytarget/esp32-cam-webserver zum laufen zu bringen. Aber der hat keine binaries verfügbar, so dass ich mich da erst einarbeiten muss. Das geht nur am Wochenende.

Staars, 2024-03-14:
Today I got a cam module for testing, which seems to be not exactly the same model, because its `v3.1` and not `v3.0`.

```
00:00:00.001 HDW: ESP32-D0WDQ6-V3 v3.1 (PSRAM)
00:00:00.031 UFS: FlashFS mounted with 300 kB free
00:00:00.044 CFG: Loaded from File, Count 17
00:00:00.051 QPC: Count 1
00:00:00.057 SPI: Hardware using GPIO14(CLK), GPIO15(MOSI) and GPIO02(MISO)
00:00:10.617 UFS: SDCard mounted (SPI mode) with 31154656 kB free
00:00:10.669 BRY: Berry initialized, RAM used 4478 bytes
00:00:10.700 Project tasmota - Tasmota Version 13.4.0.2(arduino30)-3_0_0(2024-03-14T20:34:14)
00:00:12.199 CAM: OV2640 Initialized
```

I can not reproduce the problems here. 20 fps with `wcresolution 7`.

Build environment:
```
; *** ESP32 rev.0/1 with PSRAM needs a fix for faulty PSRAM ROM code
[env:tasmota32-psramfix_ard3]
extends                 = env:arduino30
board                   = esp32-fix
board_build.f_cpu       = 240000000L
build_unflags           = ${env:arduino30.build_unflags}
build_flags             = ${env:arduino30.build_flags}
                          -DFIRMWARE_ARDUINO30
                          -DUSE_WEBCAM
monitor_filters         = esp32_exception_decoder
lib_ignore              = ${env:arduino30.lib_ignore}
```

Then used *Autoconf* with `AITHINKER CAM`.

Sadly it is more or less impossible to find the exact model on AliExpress in the order process.

Can you try to build a firmware with Arduino 3?

closed 2024-04-13 by github-actions[bot]