arendst/Tasmota #20717: Display RA8876 not working anymore after Update from Tasmota 13.3.0 to  Pre 13.4.0

helmar74, 2024-02-13:
### PROBLEM DESCRIPTION
I tried, in order to prepare 13.4.0 upgrade, to compile and flash Pre 13.4.0 version.
Flashing works successfull but after that the display is only black.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32 and RA8876 Display
- [X] Tasmota binary firmware version number used: PRE 13.4.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:30:35.924 CMD: Backlog Template; Module; GPIO 255
13:30:35.949 MQT: stat/esp32_1/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
13:30:36.179 MQT: stat/esp32_1/RESULT = {"Module":{"1":"ESP32-DevKit"}}
13:30:36.393 MQT: stat/esp32_1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"6560":"RA8876 CS"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"992":"Backlight"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"736":"SPI CLK1"},"GPIO19":{"672":"SPI MISO1"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"704":"SPI MOSI1"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"6720":"SDCard CS"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:31:18.026 CMD: Status 0
13:31:18.034 MQT: stat/esp32_1/STATUS = {"Status":{"Module":1,"DeviceName":"ESP32-2 Infodisplay Büro","FriendlyName":["ESP32-2 Infodisplay Büro"],"Topic":"esp32_1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
13:31:18.072 MQT: stat/esp32_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:02:12","StartupUTC":"2024-02-13T12:29:06","Sleep":50,"CfgHolder":4617,"BootCount":50,"BCResetTime":"2020-12-02T09:01:42","SaveCount":3071}}
13:31:18.112 MQT: stat/esp32_1/STATUS2 = {"StatusFWR":{"Version":"13.4.0(tasmota32)","BuildDateTime":"2024.02.13 13:05:37","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":240,"Hardware":"ESP32-D0WD v1.0","CR":"418/699"}}
13:31:18.128 MQT: stat/esp32_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hs.net_EG",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","000000C0","00006000","00004000","00000000"]}}
13:31:18.166 MQT: stat/esp32_1/STATUS4 = {"StatusMEM":{"ProgramSize":2001,"Free":878,"Heap":148,"StackLowMark":3,"PsrMax":4093,"PsrFree":4051,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["0407","97AAD5DF","000DABA1","B7FFBFCF","05FABBC4","E0360CC7","480840F2","202C0000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,!8,9,10,11,12,13,!14,!16,!17,!24,26,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,31,34,35,37,39,40,41,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"3,4,5,7,8,9,10,11,12,13,14,15,16,17,18,20,24,28,29,31,36,38,39,41,42,44,46,47,48,58,62,65,69,76,77,82"}}
13:31:18.240 MQT: stat/esp32_1/STATUS5 = {"StatusNET":{"Hostname":"ESP32-2","IPAddress":"192.168.1.207","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"A8:03:2A:74:A6:A8","IP6Global":"2003:f1:af0d:fb00:aa03:2aff:fe74:a6a8","IP6Local":"fe80::aa03:2aff:fe74:a6a8%st2","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:31:18.290 MQT: stat/esp32_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"ESP32_2","MqttClient":"ESP32_2","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":15}}
13:31:18.309 MQT: stat/esp32_1/STATUS7 = {"StatusTIM":{"UTC":"2024-02-13T12:31:18Z","Local":"2024-02-13T13:31:18","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"07:20","Sunset":"17:25"}}
13:31:18.342 MQT: stat/esp32_1/STATUS10 = {"StatusSNS":{"Time":"2024-02-13T13:31:18"}}
13:31:18.354 MQT: stat/esp32_1/STATUS11 = {"StatusSTS":{"Time":"2024-02-13T13:31:18","Uptime":"0T00:02:12","UptimeSec":132,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":38},"POWER":"ON","Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:10"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
3:32:01.597 SRC: Backlog
13:32:01.599 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 21, Pld -99, Data '[x180y75f2p-5Ci2]10,0'
13:32:01.667 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x180y75f2p-5Ci2]10,0"}
13:32:01.836 SRC: Backlog
13:32:01.837 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 15, Pld -99, Data '[x265y70f1Ci2]o'
13:32:01.846 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x265y70f1Ci2]o"}
13:32:02.088 SRC: Backlog
13:32:02.090 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 15, Pld -99, Data '[x275y75f2Ci2]C'
13:32:02.110 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x275y75f2Ci2]C"}
13:32:02.339 SRC: Backlog
13:32:02.341 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 21, Pld -99, Data '[x180y125f2p-5Ci4]2,9'
13:32:02.408 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x180y125f2p-5Ci4]2,9"}
13:32:02.474 HTP: Werkzeuge
13:32:02.570 SRC: Backlog
13:32:02.572 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 16, Pld -99, Data '[x265y120f1Ci4]o'
13:32:02.582 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x265y120f1Ci4]o"}
13:32:02.805 SRC: Backlog
13:32:02.807 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 16, Pld -99, Data '[x275y125f2Ci4]C'
13:32:02.827 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x275y125f2Ci4]C"}
13:32:03.007 SRC: Backlog
[... 91 lines cut ...]
13:32:09.844 SRC: Backlog
13:32:09.846 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 36, Pld -99, Data '[x700y100P/zisterne/zisterne90.jpg:]'
13:32:10.083 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x700y100P/zisterne/zisterne90.jpg"}
13:32:10.111 SRC: Backlog
13:32:10.113 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 27, Pld -99, Data '[x750y250f2p-6Bi4Ci1]258 cm'
13:32:10.192 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x750y250f2p-6Bi4Ci1]258 cm"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash Firmware from version 13.3.0 to prerelease 13.4.0

### EXPECTED BEHAVIOUR
RA8876 Display should work as of version 13.3.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[configfiles.zip](https://github.com/arendst/Tasmota/files/14261617/configfiles.zip)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-02-13:
I doubt it's linked to LVGL since you don't use LVGL

s-hadinger, 2024-02-13:
You don't see any splash screen at boot?

s-hadinger, 2024-02-13:
Also, are you using Universal Display? Or specific driver?

arendst, 2024-02-13:
``Power 1``

helmar74, 2024-02-13:
14:23:59.521 CMD: power 1
14:23:59.535 MQT: stat/esp32_1/RESULT = {"POWER":"ON"}
14:23:59.537 MQT: stat/esp32_1/POWER = ON

Doesn't help at all

helmar74, 2024-02-13:
[image]

This is the splashscreen, afterwards display is only black
Reverting back to 13.3.0 brings Display up working again

helmar74, 2024-02-13:
I'm using these settings in platform_override.ini:
```
; Aktivierung von PSRAM mit build Flag -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -lc-psram-workaround -lm-psram-workaround
; Bis Tasmota v 9.5.0 nur -DBOARD_HAS_PSRAM
; Ab Tasmota v 9.5.0.1 neue platformio_override.ini
; damit auch große Bilder angezeigt werden können
; WEitere Infos siehe
; https://github.com/arendst/Tasmota/issues/10392
; https://github.com/arendst/Tasmota/issues/12390
build_flags = ${esp32_defaults.build_flags} -DBOARD_HAS_
```

Maybe one of these are causing the issue?

arendst, 2024-02-13:
Nope. See bootlog.

s-hadinger, 2024-02-13:
You are still using Core 2. Can you paste the boot logs? I also assume that the splash screen is not normal

helmar74, 2024-02-13:
OK really strange. My overrides where working for a long time, 13.4.0 breaks the RA8876 Display for me.

helmar74, 2024-02-13:
You mean this bootlog?
```
00:00:00.002 HDW: ESP32-D0WD v1.0 (PSRAM)
00:00:00.032 UFS: FlashFS mounted with 304 kB free
00:00:00.049 CFG: Loaded from File, Anzahl 3093
00:00:00.057 QPC: Count 1
00:00:00.063 SPI: Hardware using GPIO18(CLK), GPIO23(MOSI) and GPIO19(MISO)
00:00:00.077 UFS: SDCard mounted (SPI mode) with 31138432 kB free
00:00:00.077 SCR: ufilesystem found
00:00:00.085 SCR: nv=0, tv=0, vns=0, vmem=16, smem=8192
00:00:00.091 TFS: File 'mcp23x.dat' not found
00:00:00.160 BRY: Berry initialized, RAM used 4223 bytes
00:00:01.976 DSP: RA8876
00:00:01.990 Projekt sonoffStandard - ESP32-2 Infodisplay Büro Version 13.4.0(tasmota32)-2_0_14(2024.02.13 14:53:17)
00:00:09.065 QPC: Reset
00:00:10.000 WIF: verbinden mit AP1 hs.net_EG Channel 6 BSSId 9C:C7:A6:F7:72:A2 in Modus 11n als ESP32-2...
00:00:11.954 WIF: verbunden
15:01:28.407 HTP: Webserver aktiv bei ESP32-2 mit IP-Adresse 192.168.1.207
15:01:29.829 MQT: Verbindungsversuch…
15:01:29.899 MQT: verbunden
15:01:29.902 MQT: tele/esp32_1/LWT = Online (beibehalten)
15:01:29.904 MQT: cmnd/esp32_1/POWER = 
15:01:29.909 MQT: tele/esp32_1/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"13.4.0(tasmota32)","FallbackTopic":"cmnd/ESP32_2_fb/","GroupTopic":"cmnd/tasmotas/"}}
15:01:29.924 MQT: tele/esp32_1/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"ESP32-2","IPAddress":"192.168.1.207","IP6Global":"2003:f1:af0d:fb00:aa03:2aff:fe74:a6a8","IP6Local":"fe80::aa03:2aff:fe74:a6a8%st2"}}
15:01:29.940 MQT: tele/esp32_1/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":59}}
15:01:29.953 MQT: stat/esp32_1/RESULT = {"POWER":"ON"}
15:01:29.955 MQT: stat/esp32_1/POWER = ON
15:01:30.660 MQT: stat/esp32_1/RESULT = {"DisplayText":"[z]"}
15:01:30.966 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x0y0Bi1Ci1R1024:600]"}
15:01:31.124 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x200y50f2Ci0]Aussenwerte"}
15:01:31.298 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x750y50f2Ci0]Zisterne"}
15:01:31.673 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x70y100f2p-5Ci0]9,1"}
15:01:31.853 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x155y95f1]o"}
15:01:32.065 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x165y100f2]C"}
15:01:32.317 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x180y75f2p-5Ci2]10,0"}
15:01:32.495 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x265y70f1Ci2]o"}
15:01:32.707 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x275y75f2Ci2]C"}
15:01:32.959 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x180y125f2p-5Ci4]2,9"}
15:01:33.279 MQT: stat/esp32_1/RESULT = {"DisplayText":"[x265y120f1Ci4]o"}
```

s-hadinger, 2024-02-13:
Unfortunately I don't have this device. Logs don't seem to indicate anything wrong. 

What would be useful is to bisect the github commit history to find which commit broke the device. That would be a big hint

helmar74, 2024-02-13:
I have 2 of this display. Here is the bootlog of the working Display on version 13.3.0 : 
```
00:00:00.003 HDW: ESP32-D0WD v1.0 (PSRAM)
00:00:00.025 UFS: FlashFS mounted with 40 kB free
00:00:00.037 CFG: Loaded from File, Anzahl 7258
00:00:00.045 QPC: Count 1
00:00:00.052 SPI: Hardware using GPIO18(CLK), GPIO23(MOSI) and GPIO19(MISO)
00:00:00.076 UFS: SDCard mounted (SPI mode) with 31131872 kB free
00:00:00.077 SCR: ufilesystem found
00:00:00.104 SCR: nv=0, tv=0, vns=0, vmem=16, smem=8192
00:00:00.110 TFS: File 'mcp23x.dat' not found
00:00:00.168 BRY: Berry initialized, RAM used 4207 bytes
00:00:01.255 DSP: RA8876
00:00:01.281 Projekt sonoffStandard - ESP32-1 Infodisplay Esszimmer Version 13.3.0(tasmota32)-2_0_14(2023.12.12 16:22:15)
00:00:07.915 QPC: Reset
00:00:08.944 WIF: verbinden mit AP1 hs.net_EG Channel 6 BSSId 0C:72:74:C5:AB:12 in Modus 11n als ESP32-1...
00:00:10.973 WIF: verbunden
15:04:20.502 HTP: Webserver aktiv bei ESP32-1 mit IP-Adresse 192.168.1.237
15:04:21.757 MQT: Verbindungsversuch…
15:04:21.804 MQT: verbunden
15:04:21.808 MQT: tele/esp32_1/LWT = Online (beibehalten)
15:04:21.809 MQT: cmnd/esp32_1/POWER = 
15:04:21.815 MQT: tele/esp32_1/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"13.3.0(tasmota32)","FallbackTopic":"cmnd/ESP32_1_fb/","GroupTopic":"cmnd/tasmotas/"}}
15:04:21.830 MQT: tele/esp32_1/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"ESP32-1","IPAddress":"192.168.1.237","IP6Global":"","IP6Local":"fe80::ae67:b2ff:fef9:b03c%st2"}}
15:04:21.843 MQT: tele/esp32_1/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":105}}
15:04:21.860 MQT: stat/esp32_1/RESULT = {"POWER":"ON"}
```

helmar74, 2024-02-13:
> Unfortunately I don't have this device. Logs don't seem to indicate anything wrong.
> 
> What would be useful is to bisect the github commit history to find which commit broke the device. That would be a big hint

OK I can do.
Unfourtnately I don't understand what I should do.

Could you please give a hint which steps I should do?
That would help me a lot

helmar74, 2024-02-13:
@s-hadinger : Is it possible to download sources e.g. for development version 13.3.0.3 to figure out the version which breaks the display?

arendst, 2024-02-13:
As @gemu2015 changed the driver three weeks ago he might be able to help out.

helmar74, 2024-02-13:
OK. If I can try something, just tell me and I will help to sort things out.

helmar74, 2024-02-13:
> As @gemu2015 changed the driver three weeks ago he might be able to help out.

Do you have the commit number for me please?

arendst, 2024-02-13:
https://github.com/arendst/Tasmota/pull/20562

helmar74, 2024-02-13:
Thanks
@gemu2015 : Could you please have a look on it. I'll help you, with testing.

gemu2015, 2024-02-13:
i only changed the touch driver for this display. my version still works with 13.3.03
will update to latest dev version tomorrow and report again

helmar74, 2024-02-13:
> i only changed the touch driver for this display. my version still works with 13.3.03 will update to latest dev version tomorrow and report again

As far as I can see, the commit was merged in version 13.3.0.4. So maybe 13.3.0.4 is breaking the Display. 
Do I have any possibility to compile and test 13.3.0.3 ?

gemu2015, 2024-02-13:
https://github.com/gemu2015/Sonoff-Tasmota/tree/RA8876

helmar74, 2024-02-13:
> https://github.com/gemu2015/Sonoff-Tasmota/tree/RA8876

I compiled this version. Has the same issue.
Is it possible to try a version before? for example 13.3.0.2 ?

gemu2015, 2024-02-13:
ok, try this version, which is just before my last commit for this display
https://github.com/gemu2015/Sonoff-Tasmota/tree/RA8876

helmar74, 2024-02-13:
> ok, try this version, which is just before my last commit for this display https://github.com/gemu2015/Sonoff-Tasmota/tree/RA8876

isn't it the same version? both links show the same commit...

gemu2015, 2024-02-13:
yes i used the same name, but it is just before the commit
however after rechecking the commit if don't think, the display was affected by my touch driver changes

helmar74, 2024-02-13:
This version isn't also working for me.
Maybe this is breaking issue?
[image]

as the display is correct initialized but black?
Do you have a version before this?

helmar74, 2024-02-13:
btw:
[image]

gemu2015, 2024-02-13:
ok, as my display is working with these version do i can't help directly.
try going back step by step and find out which pull request breaks your display.

1. go to tasmota repository click on commits 
2. navigate back in history until a point which you want to test
3. click on '<>' to browse repository at this point
4. navigate to tasmota/include and check the file tasmota_version.h
5. now check out this version, compile and test

helmar74, 2024-02-13:
17:45:54.850 CMD: displaydimmer
17:45:54.857 MQT: stat/esp32_1/RESULT = {"DisplayDimmer":100}

helmar74, 2024-02-13:
I found a working version:
18:08:15.867 MQT: stat/esp32_1/STATUS2 = {"StatusFWR":{"Version":"13.3.0.3(tasmota32)","BuildDateTime":"2024.02.13 17:57:20","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":240,"Hardware":"ESP32-D0WD v1.0","CR":"418/699"}}

[image]

[https://github.com/arendst/Tasmota/tree/3e3bfccb584fcae3a274e35c1f461b1c6e8a94f1](https://github.com/arendst/Tasmota/tree/3e3bfccb584fcae3a274e35c1f461b1c6e8a94f1)

# Changelog
All notable changes to this project will be documented in this file.

## [Unreleased] - Development

## [13.3.0.3]
### Added
- Berry `debug.caller` (#20470)
- GPIO Viewer user selection of assets website now defaults to `https://ota.tasmota.com/tasmota|tasmota32/gpio_viewer/assets`

### Breaking Changed

### Changed
- SML update (#20474)

### Fixed
- Scripter memory leak in `>w x` (#20473)
- ESP8266 GPIO Viewer exception 9 on reading Analog GPIO

### Removed

The breaking changes must be somewhere afterwards!

helmar74, 2024-02-13:
@gemu2015 : Something is wrong with versions afterwards. MAybe you can have a look on it? I think it would be good to fix it before releasing 13.4.0?

gemu2015, 2024-02-13:
Possible problem
Add this to user config
#define RA8876_SPI_SPEED 25000000

I changed this to 4000000 default
Maybe that is to much for your wiring

helmar74, 2024-02-13:
> 25000000
Is higher than 4000000

Do you mean 2500000  (one 0 less)?

gemu2015, 2024-02-13:
25000000 
25 Mhz 6 zeros

helmar74, 2024-02-13:
OK, added this in user_config_override.h:
[image]

That solved the issue.
19:40:52.493 MQT: stat/esp32_1/STATUS2 = {"StatusFWR":{"Version":"13.4.0(tasmota32)","BuildDateTime":"2024.02.13 19:30:50","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":240,"Hardware":"ESP32-D0WD v1.0","CR":"418/699"}}

Thank you very much for your assistance

gemu2015, 2024-02-13:
ok will restore default to 25 Mhz.
the display may even work with 50 Mhz but that depends on wiring length.

helmar74, 2024-02-13:
OK great. Thank you so much.
Closing the issue

closed 2024-02-13 by helmar74

arendst, 2024-02-14:
I'll change the default back to 25000000 now.

gemu2015, 2024-02-14:
yes, seems to be saver

helmar74, 2024-02-14:
Hi, just upgraded to master 13.4.0 . Works like a charm
Thank you!