arendst/Tasmota #20719: Multiple rules triggered unexpectedly with temperature conditions

sxx1314, 2024-02-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Description:

I have configured several rules based on temperature ranges in Tasmota, expecting them to be mutually exclusive. However, the issue is that multiple rules are being triggered simultaneously, even when the temperature only satisfies one condition.

Steps to reproduce:
- Configure multiple rules with temperature ranges as conditions, in ascending order from low to high.
- Ensure there are no conflicting rules
- Observe the behavior when the temperature fluctuates within the configured ranges.
```
Rule1
  ON Tele-DHT11#Temperature<20 DO Dimmer 0 ENDON
  ON Tele-DHT11#Temperature>=20 AND Tele-DHT11#Temperature<25 DO Dimmer 1 ENDON
  ON Tele-DHT11#Temperature>=25 AND Tele-DHT11#Temperature<28 DO Dimmer 2 ENDON
  ON Tele-DHT11#Temperature>=28 AND Tele-DHT11#Temperature<31 DO Dimmer 3 ENDON
  ON Tele-DHT11#Temperature>=31 AND Tele-DHT11#Temperature<34 DO Dimmer 4 ENDON
  ON Tele-DHT11#Temperature>=34 AND Tele-DHT11#Temperature<37 DO Dimmer 5 ENDON
  ON Tele-DHT11#Temperature>=37 AND Tele-DHT11#Temperature<40 DO Dimmer 6 ENDON
  ON Tele-DHT11#Temperature>=40 AND Tele-DHT11#Temperature<43 DO Dimmer 7 ENDON
  ON Tele-DHT11#Temperature>=43 AND Tele-DHT11#Temperature<46 DO Dimmer 10 ENDON
  ON Tele-DHT11#Temperature>=46 AND Tele-DHT11#Temperature<50 DO Dimmer 20 ENDON
  ON Tele-DHT11#Temperature>=55 DO Dimmer 100 ENDON
ELSE DO ENDON
```
L O G
```
23:45:22.321 MQT: tele/tasmota_xxxx/SENSOR = {"Time":"2024-02-13T23:45:22","ANALOG":{"A0":816},"DHT11":{"Temperature":26.0,"Humidity":39.0,"DewPoint":11.0},"TempUnit":"C"}
23:45:22.347 RUL: TELE-DHT11#TEMPERATURE>=20 AND TELE-DHT11#TEMPERATURE<25 performs "Dimmer 1"
23:45:22.355 MQT: stat/tasmota_xxxx/RESULT = {"POWER":"ON","Dimmer":1}
23:45:22.371 RUL: TELE-DHT11#TEMPERATURE>=25 AND TELE-DHT11#TEMPERATURE<28 performs "Dimmer 2"
23:45:22.378 MQT: stat/tasmota_xxxx/RESULT = {"POWER":"ON","Dimmer":2}
```
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ *] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ *] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ *] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ *] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ___esp12f__
- [ ] Tasmota binary firmware version number used: __13.3.0.5___
  - [ ] Pre-compiled
  - [ *] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Rule1
  ON Tele-DHT11#Temperature<20 DO Dimmer 0 ENDON
  ON Tele-DHT11#Temperature>=20 AND Tele-DHT11#Temperature<25 DO Dimmer 1 ENDON
  ON Tele-DHT11#Temperature>=25 AND Tele-DHT11#Temperature<28 DO Dimmer 2 ENDON
  ON Tele-DHT11#Temperature>=28 AND Tele-DHT11#Temperature<31 DO Dimmer 3 ENDON
  ON Tele-DHT11#Temperature>=31 AND Tele-DHT11#Temperature<34 DO Dimmer 4 ENDON
  ON Tele-DHT11#Temperature>=34 AND Tele-DHT11#Temperature<37 DO Dimmer 5 ENDON
  ON Tele-DHT11#Temperature>=37 AND Tele-DHT11#Temperature<40 DO Dimmer 6 ENDON
  ON Tele-DHT11#Temperature>=40 AND Tele-DHT11#Temperature<43 DO Dimmer 7 ENDON
  ON Tele-DHT11#Temperature>=43 AND Tele-DHT11#Temperature<46 DO Dimmer 10 ENDON
  ON Tele-DHT11#Temperature>=46 AND Tele-DHT11#Temperature<50 DO Dimmer 20 ENDON
  ON Tele-DHT11#Temperature>=55 DO Dimmer 100 ENDON
ELSE DO ENDON

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
23:45:22.321 MQT: tele/tasmota_xxxx/SENSOR = {"Time":"2024-02-13T23:45:22","ANALOG":{"A0":816},"DHT11":{"Temperature":26.0,"Humidity":39.0,"DewPoint":11.0},"TempUnit":"C"}
23:45:22.347 RUL: TELE-DHT11#TEMPERATURE>=20 AND TELE-DHT11#TEMPERATURE<25 performs "Dimmer 1"
23:45:22.355 MQT: stat/tasmota_xxxx/RESULT = {"POWER":"ON","Dimmer":1}
23:45:22.371 RUL: TELE-DHT11#TEMPERATURE>=25 AND TELE-DHT11#TEMPERATURE<28 performs "Dimmer 2"
23:45:22.378 MQT: stat/tasmota_xxxx/RESULT = {"POWER":"ON","Dimmer":2}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-02-13:
Please do not be so creative as to assuming that rules supports things you'd like to have, but not part of how it works. Specifically that `AND` in the trigger part. Have you being getting horrible advice from ChatGPT? That "ELSE" part at the end looks like the type of hallucinations it is liable to have.

In your case, it might work to drop the AND part, and then use `break` instead of `endon`

sxx1314, 2024-02-14:
> Please do not be so creative as to assuming that rules supports things you'd like to have, but not part of how it works. Specifically that `AND` in the trigger part. Have you being getting horrible advice from ChatGPT? That "ELSE" part at the end looks like the type of hallucinations it is liable to have.
> 
> In your case, it might work to drop the AND part, and then use `break` instead of `endon`
> 
> And **do** read the docs. https://tasmota.github.io/docs/Rules/
If I use the following rules, none of the conditions will be triggered anymore.
```
Rule1
ON Tele-DHT11#Temperature>=55 DO Dimmer 100; BREAK ENDON
ON Tele-DHT11#Temperature>=46 DO Dimmer 20; BREAK ENDON
ON Tele-DHT11#Temperature>=43 DO Dimmer 10; BREAK ENDON
ON Tele-DHT11#Temperature>=40 DO Dimmer 7; BREAK ENDON
ON Tele-DHT11#Temperature>=37 DO Dimmer 6; BREAK ENDON
ON Tele-DHT11#Temperature>=34 DO Dimmer 5; BREAK ENDON
ON Tele-DHT11#Temperature>=31 DO Dimmer 4; BREAK ENDON
ON Tele-DHT11#Temperature>=28 DO Dimmer 3; BREAK ENDON
ON Tele-DHT11#Temperature>=25 DO Dimmer 2; BREAK ENDON
ON Tele-DHT11#Temperature>=20 DO Dimmer 1; BREAK ENDON
ON Tele-DHT11#Temperature<20 DO Dimmer 0; BREAK ENDON
```

barbudor, 2024-02-14:
Read again the syntax around BREAK

You have extra `;` and `ENDON`

sxx1314, 2024-02-14:
> Rule1
> ON Tele-DHT11#Temperature>=55 DO Dimmer 100; BREAK ENDON
> ON Tele-DHT11#Temperature>=46 DO Dimmer 20; BREAK ENDON
> ON Tele-DHT11#Temperature>=43 DO Dimmer 10; BREAK ENDON
> ON [...]

Thank you for your help, it is working now.

closed 2024-02-14 by sxx1314