arendst/Tasmota #20732: Buzzer remains active when pin unconfigured (ESP32) (minor issue)

vic42, 2024-02-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 DevKitC V4 by Espressif with WROOM module
- [x] Tasmota binary firmware version number used:  Tasmota 13.4.0.1(tasmota32)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:16:02.311 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
18:16:02.533 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
18:16:02.791 RSL: RESULT = {"GPIO0":{"480":"Buzzer"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

N/A

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

18:16:36.488 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_661178","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:16:36.523 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:07:48","StartupUTC":"2024-02-15T17:08:48","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2024-02-15T17:49:20","SaveCount":23}}
18:16:36.558 RSL: STATUS2 = {"StatusFWR":{"Version":"13.4.0.1(tasmota32)","BuildDateTime":"2024-02-15T09:44:35","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"406/699"}}
18:16:36.572 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["attingo_2ghz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","20006000","00004000","00000000"]}}
18:16:36.609 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1908,"Free":971,"Heap":147,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
18:16:36.678 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-661178-4472","IPAddress":"172.23.0.170","Gateway":"172.23.0.254","Subnetmask":"255.255.255.0","DNSServer1":"31.12.7.254","DNSServer2":"1.1.1.1","Mac":"94:B9:7E:66:11:78","IP6Global":"","IP6Local":"fe80::96b9:7eff:fe66:1178%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"31.12.7.254","DNSServer2":"1.1.1.1","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
18:16:36.724 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_661178","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:16:36.742 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-02-15T17:16:36Z","Local":"2024-02-15T18:16:36","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:59","Sunset":"18:09"}}
18:16:36.775 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-02-15T18:16:36"}}
18:16:36.785 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-02-15T18:16:36","Uptime":"0T00:07:48","UptimeSec":468,"Heap":149,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":46},"Wifi":{"AP":1,"SSId":"attingo_2ghz","BSSId":"B4:FB:E4:11:20:B7","Channel":1,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

18:17:37.606 CMD: buzzer 2,3,4
18:17:37.608 SRC: WebConsole from 172.23.0.254
18:17:37.610 CMD: Grp 0, Cmd 'BUZZER', Idx 1, Len 5, Pld 2, Data '2,3,4'
18:17:37.613 BUZ: Count 2(4), Time 3/4, Tune 0x00000000(0x00000000), Size 0, Mode 1
18:17:37.615 RSL: RESULT = {"Buzzer":"Done"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1) connect a speaker (and/or Oscilloscope) to GPIO27
2) configure GPIO27 as "Buzzer"
3) `SO111 1`
4) configure GPIO27 as "none"
5) configure GPIO0 (or probably any other GPIO) as "Buzzer"
6) Issue command 'buzzer 2,3,4'

RESULT: The buzzer waveform appears **both** on GPIO27 **and** GPIO0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The buzzer waveform should **only** appear on GPIO0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-02-16:
Testing with a LED, I'm also seeing this with ESP32-D0WDQ6 v1.0, but not on ESP32-C3.

arendst, 2024-02-16:
Is it only GPIO27 as original GPIO and then any other GPIO or does it happen with any two GPIOs?

You might want to try GPIOViewer and see if it also pops up with unexpected changing GPIO's.

vic42, 2024-02-16:
Tried it with GPIO25 and GPIO5 -> same result.
GPIO Viewer does not seem to show any signs of what's happening
[image]

sfromis, 2024-02-16:
My testing was with yet another pair of gpio pins (12,13), and the GPIO viewer only shows the pin configured as Buzzer, with both outputs blinking.

Not sure what "see if it also pops up with unexpected changing GPIO's" means.

vic42, 2024-02-16:
In my case (25,5) the unconfigured pin did not show any signs of activity in GPIO viewer.

arendst, 2024-02-16:
> Not sure what "see if it also pops up with unexpected changing GPIO's" means.

GPIO Viewer only shows changing GPIO's. If they do not change between smpling time (default 100mS) the won't pop up as active GPIO.

arendst, 2024-02-16:
Ah verified! Great fun. What's happening here. For some reason the PWM channel is not released.

arendst, 2024-02-16:
It's also a problem with moved PWM channels; The GPIO's, once a PWM now set to None are still performing like PWM GPIO's.

arendst, 2024-02-16:
Thie may lead to a solution https://github.com/micropython/micropython/issues/5737 ...

arendst, 2024-02-16:
OK. I have a fix. Need to further test where to use and what the impact will be.

```
void analogDetach(void) {
  for (uint32_t pin = 0; pin < SOC_GPIO_PIN_COUNT; pin++) { 
    if (pin_to_channel[pin] > 0) {
      ledcDetachPin(pin);
    }
  }
}
```

arendst, 2024-02-16:
This will fix the GUI Module Config. Still need to fix Template Config and command line GPIO config changes.

Give it a try.

sfromis, 2024-02-16:
Now the Buzzer from the previous boot seems to be gone for good.

sfromis, 2024-02-16:
Just tested on the updated precompiled ESP32-S3 build, and am seeing the previous Buzzer pin joining the party. `"Version":"13.4.0.1(tasmota32)","BuildDateTime":"2024-02-16T19:16:03"`

arendst, 2024-02-17:
Thx. The fix only solved the issue on esp32. I'll make it work for all AND on every restart.

closed 2024-02-18 by arendst