arendst/Tasmota #20753: Translation issue

AleXSR700, 2024-02-18:
Hi everyone and thank you @SteWers for making the eQ-3 TRV visible in the WebUI.

There is a small translation issue:
The `posn` information supplied by the TRV refers to the valve state (percentage of opening of the valve).
In the new code for the WebUI as well as the translations ( https://github.com/arendst/Tasmota/blob/920967506b5177b0a55563c408829c8620a3bf6e/tasmota/language/en_GB.h#L551C1-L556C51 ) it is referred to as "Duty Cycle".

I do not know what "Duty Cycle" is supposed to mean for the other TRV drivers, so I do not want to simply create a PR and change this translation.
For the eQ-3, as mentioned, this is not a "Duty Cycle" but would simply be "Valve" or maybe "Valve Opening" (in German probably correct "Ventilöffnungsgrad").
To not make it needlessly long I would simply go with "Valve" or "Ventil".

Please let me know if this means something specific for the `xdrv_39_thermostat.ino`. Maybe it needs to be relabelled in the driver itself to prevent misuse?

Thank you again and best regards
Alex

SteWers, 2024-02-18:
Hi @AleXSR700,
in general `posn` and `Duty cycle` is the same, but I agree with you to add a separate label for the valve position. This will not fit with thermostat driver. So I'll do a PR to add a different one.

SteWers, 2024-02-18:
[image]

AleXSR700, 2024-02-18:
> Hi @AleXSR700,
> in general `posn` and `Duty cycle` is the same, but I agree with you to add a separate label for the valve position. This will not fit with thermostat driver. So I'll do a PR to add a different one. 

Thank you for handling it :)
German translation sounds good to me :)

I think they are actually not the same, btw. I think duty cycle is time based, i.e. 50% means "half of the time the valve is fully open and half of the time fully closed". So "state per time/period".

With a TRV (at least the eQ-3) it is how far the valve is open. 
So 100 % means fully open. 50 % means half open.

SteWers, 2024-02-18:
Yes, you're right. At least the result should be the same. 😉