arendst/Tasmota #20770: Sonoff SPM PowerDelta Command Unknown

clau-bucur, 2024-02-20:
### PROBLEM DESCRIPTION
Running Tasmota on a Sonoff SPM, which supports power monitoring.
I try to change the reporting of the active power load by using `PowerDelta` but I get `Command Unknown`.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff SPM
- [x] Tasmota binary firmware version number used: 13.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:35:09.074 MQT: stat/spmworkshop/RESULT = {"NAME":"Sonoff SPM","GPIO":[0,0,0,0,3200,5536,0,0,672,704,736,0,3232,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,544,0,0,32,0,0,0,0],"FLAG":0,"BASE":1}
20:35:09.306 MQT: stat/spmworkshop/RESULT = {"Module":{"0":"Sonoff SPM"}}
20:35:09.566 MQT: stat/spmworkshop/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"3200":"Serial Tx"},"GPIO5":{"5536":"ETH POWER"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"672":"SPI MISO1"},"GPIO13":{"704":"SPI MOSI1"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"0":"None"},"GPIO16":{"3232":"Serial Rx"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"544":"LedLink"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"32":"Button1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:36:06.217 MQT: stat/spmworkshop/STATUS = {"Status":{"Module":0,"DeviceName":"SPM Breaker","FriendlyName":["Switch 1","Switch 2","Switch 3","Switch 4","Switch 5","Switch 6","Switch 7","Switch 8"],"Topic":"spmworkshop","ButtonTopic":"0","Power":16252927,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:36:06.225 MQT: stat/spmworkshop/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T02:44:04","StartupUTC":"2024-02-20T15:52:02","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2024-02-07T20:51:13","SaveCount":269}}
20:36:06.229 MQT: stat/spmworkshop/STATUS2 = {"StatusFWR":{"Version":"13.4.0(tasmota32)","BuildDateTime":"2024-02-14T16:17:40","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"455/699"}}
20:36:06.233 MQT: stat/spmworkshop/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"syslog.gme","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"548B00C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00004000","00004000","00000000"]}}
20:36:06.248 MQT: stat/spmworkshop/STATUS4 = {"StatusMEM":{"ProgramSize":1907,"Free":972,"Heap":151,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
20:36:06.257 MQT: stat/spmworkshop/STATUS5 = {"StatusNET":{"Hostname":"spmworkshop","IPAddress":"0.0.0.0","Gateway":"xxx98.254","Subnetmask":"255.255.255.0","DNSServer1":"xxx0.1","DNSServer2":"0.0.0.0","Mac":"xxxx","IP6Global":"","IP6Local":"","Ethernet":{"Hostname":"spmworkshop-eth","IPAddress":"xxx2.14","Gateway":"xxx2.254","Subnetmask":"255.255.255.0","DNSServer1":"xxx0.1","DNSServer2":"0.0.0.0","Mac":"xxx","IP6Global":"","IP6Local":"xxx"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
20:36:06.261 MQT: stat/spmworkshop/STATUS6 = {"StatusMQT":{"MqttHost":"xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E0648","MqttUser":"xxx","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:36:06.266 MQT: stat/spmworkshop/STATUS7 = {"StatusTIM":{"UTC":"2024-02-20T18:36:06Z","Local":"2024-02-20T20:36:06","StartDST":"2024-03-31T03:00:00","EndDST":"2024-10-27T04:00:00","Timezone":99,"Sunrise":"08:50","Sunset":"19:17"}}
20:36:06.292 MQT: stat/spmworkshop/STATUS10 = {"StatusSNS":{"Time":"2024-02-20T20:36:06","SPM":{"Energy":[0.2,0.0,0.0,0.0,0.0,0.1,0.0,0.0,0.0,0.2,0.0,0.3,0.0,0.1,0.0,0.0,0.1,0.0,0.5,0.0],"Yesterday":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Today":[0.2,0.0,0.0,0.0,0.0,0.1,0.0,0.0,0.0,0.2,0.0,0.3,0.0,0.1,0.0,0.0,0.1,0.0,0.5,0.0],"ActivePower":[100.0,11.6,31.6,5.2,0.0,0.0,4.7,6.4,0.0,13.1,9.8,169.1,1.5,16.8,17.9,4.4,0.6,2.7,2.2,0.0],"ApparentPower":[149.1,23.4,47.8,9.2,0.0,8.4,27.4,6.8,4.2,82.9,13.0,179.9,1.5,26.9,28.2,5.0,5.1,5.3,34.4,0.0],"ReactivePower":[111.6,19.8,35.5,7.3,0.0,8.3,26.7,1.7,4.1,81.7,7.4,59.7,0.0,20.8,21.8,2.5,4.9,4.1,34.3,0.0],"Factor":[0.67,0.50,0.66,0.56,0.00,0.00,0.17,0.94,0.01,0.16,0.75,0.94,1.00,0.62,0.64,0.90,0.12,0.51,0.06,0.00],"Voltage":[240.7,240.7,240.7,240.8,240.7,240.6,240.6,240.5,240.6,240.5,240.6,240.5,240.3,240.0,240.1,240.1,239.9,240.1,240.4,0.0],"Current":[0.64,0.10,0.20,0.04,0.00,0.03,0.11,0.02,0.02,0.34,0.05,0.74,0.00,0.11,0.11,0.01,0.02,0.02,0.14,0.00]}}}
20:36:06.301 MQT: stat/spmworkshop/STATUS11 = {"StatusSTS":{"Time":"2024-02-20T20:36:06","Uptime":"0T02:44:04","UptimeSec":9844,"Heap":154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Berry":{"HeapUsed":4,"Objects":43},"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","POWER5":"ON","POWER6":"ON","POWER7":"ON","POWER8":"ON","POWER9":"ON","POWER10":"ON","POWER11":"ON","POWER12":"ON","POWER13":"ON","POWER14":"ON","POWER15":"ON","POWER16":"ON","POWER17":"ON","POWER18":"ON","POWER19":"ON","POWER20":"OFF"}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Go to Console and enter command: `PowerDelta`.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
`PowerDelta` should be accepted since this is a power monitoring device.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-02-20:
Not supported on SPM because this is a "special" power monitoring device with too many power monitoring devices.

clau-bucur, 2024-02-20:
Understood. I use Tasmota integrated in Home Assistant.
I've come across this issue because I noticed the power usage reported in Home Assistant is quite different to the one shown in the Tasmota interface.
More exactly, in the Tasmota interface the power consumtion values change more frequently and pretty accurately, whereas in Home Assistant the readings update much less frequent (only experienced this behavior with SPM).

When I turn on a relay and there is a load on it, Tasmota shows power consumption for it, but in Home Assistant the readings (A, W, etc.) remain to 0 for a long time.

sfromis, 2024-02-20:
When it comes to updates in HA, there is simply no issue, other than the PowerDelta feature (which would trigger frequent updates) is not supported with SPM. You could create Berry code to keep track of values and do something similar to PowerDelta.